introlab/rtabmap #5: How to provide odometry?

Tobias-Fischer, 2015-03-15:
Hi,

Thanks for your great software package! I use the RGBD C++ version (https://github.com/introlab/rtabmap/blob/master/examples/RGBDMapping/main.cpp) slightly modified. This works great, however now I want to provide the odometry information provided by our robot. Could you please advise how to do that?

I suppose I have to delete the pipe from the camera thread to the odometry thread. Furthermore, I have to somehow update the pose (in my case, it will be 2D, so only x, y and the angle) of the robot provided by the wheels, and pass it to the odometry thread. However, I am stuck on how to do that.

It would be great if you hint me towards the right direction.

Thanks, Tobias

matlabbe, 2015-03-16:
You should create your own odometry thread, which will publish OdometryEvent. These odometry events are already catched by RtabmapThread (so no need to use pipes). OdometryEvent requires the pose + RGB-D images. You set the pose with x,y,yaw from your robot and fill the RGB and depth images. For convenience, you could use a class similar to CameraThread to do that. 

For example, you could change [this line in CameraThread.cpp](https://github.com/introlab/rtabmap/blob/721de2e76be0c41bf194ef2081df31bba4f9e436/corelib/src/CameraThread.cpp#L128):
```cpp
this->post(new CameraEvent(rgb, depth, fx, fy, cx, cy, _cameraRGBD->getLocalTransform(), ++_seq, UTimer::now()));
```
to
```cpp
float x = //... (from your robot's odometry)
float y = //... 
float theta = //...
Transform pose(x,y,0,0,0,theta);
SensorData data(rgb,
		  depth,
		  fx, fy, cx, cy,
		  _cameraRGBD->getLocalTransform(),
		  pose,
		  1, // rotational variance
		  1, // transitional variance
		  ++_seq,
		 UTimer::now());
this->post(new OdometryEvent(data));
```

Tobias-Fischer, 2015-03-16:
Hi @matlabbe,
Thanks for your fast reply! It seems to be straightforward to change it to use the robot's odometry, I'll give it a try in the next couple of days.

Thanks,
Tobias

Tobias-Fischer, 2015-05-22:
Hi,
just a quick update: I am using the ROS version of rtabmap by now, so I don't have this issue anymore. Thanks again for your help :+1: 

Best,
Tobias

closed 2015-05-22 by Tobias-Fischer