introlab/rtabmap #10: Problem about results reproduction

ErnstHowie, 2015-04-28:
Hello:

Thanks for you great package! It is very excellent. I installed it without any problem. After installation, I want to test this package, so I run the results reproduction following your instruction (https://github.com/introlab/rtabmap/wiki/Benchmark).
I tested the loop closure detection on UdeS 1Hz dataset using the same command as you:
$ ./rtabmap-console -Rtabmap/StatisticLogged true -Rtabmap/StatisticLoggedHeaders false -Rtabmap/PublishStats false -Mem/RehearsalIdUpdatedToNewOne true -Mem/BadSignaturesIgnored true -Rtabmap/LoopRatio 0.0 -Bayes/FullPredictionUpdate true -Kp/TfIdfLikelihoodUsed false -Mem/STMSize 30 -Mem/RehearsalSimilarity 0.2 -Mem/RecentWmRatio 0.2 -SURF/Extended false -Kp/WordsPerImage 400 -Kp/BadSignRatio 0.25 -SURF/HessianThreshold 150 -Rtabmap/TimeThr 700 UdeS_1Hz 

But my result is not good as yours:
GT_total_positives =
        2624
False positive! id=2927 with old=85 (p=0.78605)
False positive accepted! id=2927 with old=85 (p=0.78605)
False positive! id=3221 with old=507 (p=0.27742)
False positive accepted! id=3221 with old=507 (p=0.27742)
False positive! id=3303 with old=3261 (p=0.20176)
False positive accepted! id=3303 with old=3261 (p=0.20176)
False positive! id=3304 with old=3261 (p=0.18977)
False positive accepted! id=3304 with old=3261 (p=0.18977)
False positive! id=3305 with old=3261 (p=0.1824)
False positive accepted! id=3305 with old=3261 (p=0.1824)
False positive! id=508 with old=388 (p=0.18231)
False positive accepted! id=508 with old=388 (p=0.18231)
False positive! id=3306 with old=3261 (p=0.17308)
False positive accepted! id=3306 with old=3261 (p=0.17308)
False positive! id=3222 with old=507 (p=0.12043)
False positive accepted! id=3222 with old=507 (p=0.12043)
False positive! id=509 with old=384 (p=0.10367)
False positive! id=5007 with old=2423 (p=0.096849)
False positive! id=63 with old=15 (p=0.09674)
False positive! id=62 with old=4 (p=0.095517)
False positive! id=85 with old=25 (p=0.093663)
False positive! id=61 with old=4 (p=0.092973)
False positive! id=511 with old=398 (p=0.092742)
False positive! id=4127 with old=1392 (p=0.090118)
False positive! id=5006 with old=2423 (p=0.088463)
False positive! id=60 with old=4 (p=0.08766)
False positive! id=153 with old=47 (p=0.083117)
False positive! id=4270 with old=1554 (p=0.081888)
False positive! id=157 with old=34 (p=0.081526)
False positive! id=2622 with old=90 (p=0.081042)
False positive! id=148 with old=47 (p=0.078752)
False positive! id=152 with old=47 (p=0.078644)
False positive! id=156 with old=34 (p=0.078002)
False positive! id=59 with old=4 (p=0.076487)
False positive! id=3932 with old=1219 (p=0.075674)
False positive! id=149 with old=31 (p=0.075637)
False positive! id=147 with old=35 (p=0.074969)
False positive! id=154 with old=46 (p=0.074458)
False positive! id=5005 with old=2423 (p=0.074371)
False positive! id=1415 with old=190 (p=0.074223)
False positive! id=3836 with old=1178 (p=0.073854)
False positive! id=195 with old=20 (p=0.072997)
False positive! id=186 with old=90 (p=0.072248)
False positive! id=5341 with old=2732 (p=0.072152)
False positive! id=155 with old=46 (p=0.071894)
False positive! id=90 with old=37 (p=0.07122)
False positive! id=2800 with old=2718 (p=0.071138)
False positive! id=3223 with old=507 (p=0.07082)
False positive! id=5339 with old=2731 (p=0.070738)
False positive! id=179 with old=90 (p=0.070554)
False positive! id=151 with old=42 (p=0.070286)
False positive! id=2623 with old=91 (p=0.070054)
Recall max (Precision=100%) = 20.0457% (p=0.78693), accepted=526
Recall max accepted (Precision=100%) = 20.0457% (p=0.78693), accepted=526
ignored = 81

Did I miss something? Please help me find out the problem. Thanks in advance!

matlabbe, 2015-04-29:
The first false positives detected in the log above are not false. I updated the Ground Truth (see attachement). Just re-run the showlogs.m again with this ground truth. 

Note that RTAB-Map performance depends on how many locations can be kept in Working Memory (i.e. the computation power) under the real-time limit set. In particular, for this dataset, the Recall varies a lot, between 34% and 51% at 100% precision. See table table III of this [paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/b/bc/TRO2013.pdf). Well, the code has changed the last years so it may have an effect on the performance. I processed again two times with the latest code and I have a Recall of 47% and 42%. You should get similar results.

[image]

ErnstHowie, 2015-04-29:
Thanks for your update! With your new ground truth, I run this experiment again. Then I got a results as following:
GT_total_positives =
        2628
False positive! id=63 with old=15 (p=0.09674)
False positive! id=62 with old=4 (p=0.095517)
False positive! id=85 with old=25 (p=0.093663)
False positive! id=61 with old=4 (p=0.092973)
False positive! id=511 with old=398 (p=0.092742)
False positive! id=60 with old=4 (p=0.08766)
False positive! id=153 with old=47 (p=0.083117)
False positive! id=1012 with old=576 (p=0.082137)
False positive! id=157 with old=34 (p=0.081526)
False positive! id=148 with old=47 (p=0.078752)
False positive! id=152 with old=47 (p=0.078644)
False positive! id=156 with old=34 (p=0.078002)
False positive! id=59 with old=4 (p=0.076487)
False positive! id=149 with old=31 (p=0.075637)
False positive! id=147 with old=35 (p=0.074969)
False positive! id=154 with old=46 (p=0.074458)
False positive! id=1415 with old=190 (p=0.074028)
False positive! id=195 with old=20 (p=0.072997)
False positive! id=1011 with old=576 (p=0.072401)
False positive! id=186 with old=90 (p=0.072248)
False positive! id=155 with old=46 (p=0.071894)
False positive! id=2726 with old=2616 (p=0.071505)
False positive! id=90 with old=37 (p=0.07122)
False positive! id=179 with old=90 (p=0.070554)
False positive! id=151 with old=42 (p=0.070286)
Recall max (Precision=100%) = 45.3196% (p=0.096821), accepted=1191
Recall max accepted (Precision=100%) = 43.1126% (p=0.11003), accepted=1133

I think it is close to your result. Any way, thanks very much!

matlabbe, 2015-04-29:
It looks good to me!

ErnstHowie, 2015-04-29:
Hi:

Thanks for help me! I think your package is great. I like it, especially its framework.
I also want to add my algorithms (about loop closure detection) into the framework to run some experiments. Your code is good but it is too much and complex to understand for me.
And I also cannot find any documents to explain the structure of your source code. So can you give me some instruction?

Thanks very much!

mryellow, 2015-05-01:
> Your code is good but it is too much and complex to understand for me.

Just passing by, not involved in the research.

“Memory management for real-time appearance-based loop closure detection”
https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/f/f0/Labbe11memory.pdf

The "long-term memory", "working-memory" concepts from that paper might help clear up a lot of the complexity in there.

ErnstHowie, 2015-05-02:
@mryellow  Thanks!

matlabbe, 2015-05-04:
Well there is no official C++ API to actually modify the code, only some C++ examples ([RGB-D mapping](https://github.com/introlab/rtabmap/wiki/Cplusplus-RGBD-Mapping) or [loop closure detection-only](https://github.com/introlab/rtabmap/wiki/Cplusplus-Loop-Closure-Detection)) to use the basic interface of the rtabmap library.

As @mryellow pointed out, the papers are a good place to start to understand all the concepts used in the code. Note that the loop closure detection approach inside RTAB-Map is tightly coupled to RTAB-Map's memory management approach, which would make difficult to replace it with your own approach. 

I was thinking recently to create an interface to let people use their own loop closure detection approach inside the RTAB-Map's framework, like using [FAB-MAP](http://docs.opencv.org/modules/contrib/doc/openfabmap.html) inside to compare, but it is still in future works.

mryellow, 2015-05-04:
> let people use their own loop closure detection approach 

I took an interesting approach in a little experiment, something which might be worth adding as a feature.

Created a "whitelist" of loop closure signatures in RTABMap, this was then checked in the rejection filtering section and loops only passed once they had been externally approved and existed in this array.

https://github.com/introlab/rtabmap_ros/compare/master...mryellow:ratcheck

It has to wait a cycle for approval and didn't suit my needs, but might be handy.

wengxiuling36, 2016-03-26:
I download dataset UdeS_1Hz, I unzip UdeS_1Hz.part.rar,I did/t get right image  `UdeS_1Hz/946.jpg `and `UdeS_1Hz/3443.jpg`.Is my computer problem?
[image]

wengxiuling36, 2016-03-28:
matlab showlogs output peoblem:
The ground truth size doesn't match the log files (LogI=5394, LogF=5394, GT=5395)
I tested the loop closure detection on UdeS 1Hz,But I did't get iteration<5395>
[image]

matlabbe, 2016-03-29:
There was a bug (introduced by 0.11) where the first image in the directory was skipped. Either use argument "-start_at 0" when using `rtabmap-console` or update to have the commit above. The commit also fixes the warning above.

In your screenshot, the time used at each iteration is high, like if the time threshold is not set (memory management is disabled). If you use the commands on the [Benchmark](https://github.com/introlab/rtabmap/wiki/Benchmark#udes-1hz) page, you should have something like this (time would be around 700 ms):

```bash
 iteration(5387) loop(2803) hyp(0.53) time=0.741512s/0.753868s *
 iteration(5388) loop(2804) hyp(0.70) time=0.800624s/0.814004s *
 iteration(5389) loop(2804) hyp(0.81) time=0.745913s/0.759609s *
 iteration(5390) loop(2806) hyp(0.93) time=0.864439s/0.877369s *
 iteration(5391) loop(2806) hyp(0.97) time=0.770983s/0.783522s *
 iteration(5392) loop(2807) hyp(0.94) time=0.803866s/0.817033s *
 iteration(5393) loop(2809) hyp(0.88) time=0.925232s/0.940438s *
 iteration(5394) loop(2810) hyp(0.94) time=0.946947s/0.959859s *
 iteration(5395) loop(2811) hyp(0.95) time=0.833488s/0.833910s *
Processing images completed. Loop closures found = 1228
 Total time = 3141.841821s
 Cleanup time = 14.017228s
Database ("rtabmapconsole.db") and log files saved to current directory.
```

closed 2016-03-29 by matlabbe