introlab/rtabmap #12: make error after cmake succeed~~

lligen, 2015-05-26:
hi,
   i have got a problem when i followed the tutorial-- installation:Build from source.
   cmake info is :
-- Info :
--   CMAKE_INSTALL_PREFIX = /home/f2/catkin_ws_h/devel
--   CMAKE_BUILD_TYPE = Release
--   BUILD_SHARED_LIBS = ON
--   With OpenCV nonfree module (SIFT/SURF) = YES
--   With Freenect = YES
--   With OpenNI2  = YES
--   With Freenect2 = NO (libfreenect2 not found)
--   With dc1394  = NO (dc1394 not found)
--   With FlyCapture2/Triclops  = NO (Point Grey SDK not found)
--   With g2o  = YES
--   With Qt  = YES (version 4)
   make error is :
[  8%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/CameraThread.cpp.o
In file included from /home/f2/cmake/rtabmap/corelib/src/CameraThread.cpp:30:0:
/home/f2/cmake/rtabmap/corelib/src/../include/rtabmap/core/CameraRGBD.h:146:31: error: ‘openni_wrapper’ was not declared in this scope
/home/f2/cmake/rtabmap/corelib/src/../include/rtabmap/core/CameraRGBD.h:146:52: error: template argument 1 is invalid
/home/f2/cmake/rtabmap/corelib/src/../include/rtabmap/core/CameraRGBD.h:147:28: error: ‘openni_wrapper’ was not declared in this scope
/home/f2/cmake/rtabmap/corelib/src/../include/rtabmap/core/CameraRGBD.h:147:54: error: template argument 1 is invalid
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/CameraThread.cpp.o] Error 1
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [all] Error 2

how can i solve this problem? thanks for your help~~

matlabbe, 2015-05-26:
openni_wrapper is declared in the [included](https://github.com/introlab/rtabmap/blob/master/corelib/include/rtabmap/core/CameraRGBD.h#L42-L43) PCL headers [pcl/io/openni_camera/openni_depth_image.h](https://github.com/PointCloudLibrary/pcl/blob/master/io/include/pcl/io/openni_camera/openni_depth_image.h) and [pcl/io/openni_camera/openni_image.h](https://github.com/PointCloudLibrary/pcl/blob/master/io/include/pcl/io/openni_camera/openni_image.h).

What is your system?

lligen, 2015-05-28:
ubuntu12.04   
i can find the PCL headers  pcl/io/openni_camera/openni_depth_image.h and pcl/io/openni_camera/openni_image.h in /usr/include/pcl-1.7/pcl...
but it didn't work.

lligen, 2015-05-28:
I moved to another computer and then it works .

392210346, 2015-09-09:
I meet the same question ,my system is Ubantu 14.04,zh_CN,
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/CameraThread.cpp.o] Error 1
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [all] Error 2
 I don't know how to solve it,Have you solved it?

392210346, 2015-09-09:
what's your system ,12.04zh_CN or en_US? Is it important to use zh_CN or en _US?

matlabbe, 2015-09-09:
If it is the same errror as `‘openni_wrapper’ was not declared in this scope`, verify that your PCL installation has OpenNI. I don't think it is a language problem here.

392210346, 2015-09-09:
when I run cmake ,I meet the next question:
CMake Warning at CMakeLists.txt:175 (FIND_PACKAGE):
  By not providing "FindGTSAM.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "GTSAM", but
  CMake did not find one.

  Could not find a package configuration file provided by "GTSAM" with any of
  the following names:

    GTSAMConfig.cmake
    gtsam-config.cmake

  Add the installation prefix of "GTSAM" to CMAKE_PREFIX_PATH or set
  "GTSAM_DIR" to a directory containing one of the above files.  If "GTSAM"
  provides a separate development package or SDK, be sure it has been
  installed.

CMake Warning at CMakeLists.txt:183 (FIND_PACKAGE):
  By not providing "Findcvsba.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "cvsba", but
  CMake did not find one.

  Could not find a package configuration file provided by "cvsba" with any of
  the following names:

    cvsbaConfig.cmake
    cvsba-config.cmake

  Add the installation prefix of "cvsba" to CMAKE_PREFIX_PATH or set
  "cvsba_DIR" to a directory containing one of the above files.  If "cvsba"
  provides a separate development package or SDK, be sure it has been
  installed.
 With GTSAM                = NO (GTSAM not found)
--   With cvsba                = NO (cvsba not found)

I don't know whether  GTSAM and cvsba  are necessary?

matlabbe, 2015-09-09:
They are optional.

cheers

392210346, 2015-10-21:
I meet the same question ,my system is Ubantu 14.04,zh_CN,
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/CameraThread.cpp.o] Error 1
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [all] Error 2

When I don't install ros-indigo-desk-full ,I can compile rtabmap,but  after I installed ros-indigo-desk-full,
I met last question ,I don't know where the question occured.

I hope you can help me ,thanks.

matlabbe, 2015-10-21:
Are there more information about this error? or it is just these two lines?

lifeisfunny, 2016-02-06:
I am running to the same issue with mac os x 10.11.3   i brew installed pcl   and i got fatal error: 
      'pcl/io/openni_camera/openni_depth_image.h' file not found

lifeisfunny, 2016-02-06:
And wow  this is one of the earliest problems that not been fixed

matlabbe, 2016-02-06:
That error is PCL specific, that means that PCL is not installed with openni support. You should install PCL from brew with [openni](https://github.com/Homebrew/homebrew-science/blob/be74762728cf17680aaf671995a0e889642ef301/pcl.rb#L95-L96) option (you can also set option openni2).

For info with [macport](https://www.macports.org/ports.php?by=name&substr=libpcl), pcl is installed with openni by default. Packages I use:
```
vtk5 @5.10.1_3+qt4_mac
libpcl @1.7.2_0
```

lligen, 2016-02-24:
problem about /usr/include/ni and PCL specific are solved after uninstall and reinstall **libopenni-dev**, also may be need to install **libvtk-java** and **python-vtk**.

closed 2016-02-24 by matlabbe