introlab/rtabmap #26: double free or corruption - rtabmap compilation on ARM Ubuntu

tanmayshankar, 2015-06-09:
Hello Mathieu, 

I ran rtabmap successfully on Ubuntu 12.04 Ros Hydro, for my laptop. I am now trying to compile rtabmap on an Odroid XU3, ARM based, though also with Ubuntu 12.04 ROS Hydro. 

The cmake command for rtabmap runs fine - these are the last few lines of the output. 

-- --------------------------------------------
-- Info :
--   CMAKE_INSTALL_PREFIX = /home/odroid/catkin_ws/devel
--   CMAKE_BUILD_TYPE = Release
--   BUILD_SHARED_LIBS = ON
--   With OpenCV nonfree module (SIFT/SURF) = YES
--   With Freenect = YES
--   With OpenNI2  = NO (OpenNI2 not found)
--   With Freenect2 = NO (libfreenect2 not found)
--   With dc1394  = YES
--   With FlyCapture2/Triclops  = NO (Point Grey SDK not found)
--   With g2o  = NO (g2o not found)
--   With Qt  = YES (version 4)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/odroid/rtabmap/build

_________________________________________________
Now when I run make, there is a memory error somewhere - 
_________________________________________________
Scanning dependencies of target rtabmap_utilite
[  0%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UEventsManager.cpp.o
In file included from /home/odroid/rtabmap/utilite/src/../include/rtabmap/utilite/UEventsManager.h:27:0,
                 from /home/odroid/rtabmap/utilite/src/UEventsManager.cpp:20:
/home/odroid/rtabmap/utilite/src/../include/rtabmap/utilite/ULogger.h:458:18: note: the mangling of 'va_list' has changed in GCC 4.4
[  1%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UEventsHandler.cpp.o
In file included from /home/odroid/rtabmap/utilite/src/../include/rtabmap/utilite/UEventsManager.h:27:0,
                 from /home/odroid/rtabmap/utilite/src/UEventsHandler.cpp:21:
/home/odroid/rtabmap/utilite/src/../include/rtabmap/utilite/ULogger.h:458:18: note: the mangling of 'va_list' has changed in GCC 4.4
[  2%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UEventsSender.cpp.o
In file included from /home/odroid/rtabmap/utilite/src/../include/rtabmap/utilite/UEventsManager.h:27:0,
                 from /home/odroid/rtabmap/utilite/src/UEventsSender.cpp:21:
/home/odroid/rtabmap/utilite/src/../include/rtabmap/utilite/ULogger.h:458:18: note: the mangling of 'va_list' has changed in GCC 4.4
[  2%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UFile.cpp.o
[  3%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UDirectory.cpp.o
In file included from /home/odroid/rtabmap/utilite/src/UDirectory.cpp:43:0:
/home/odroid/rtabmap/utilite/src/../include/rtabmap/utilite/ULogger.h:458:18: note: the mangling of 'va_list' has changed in GCC 4.4
[  3%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UConversion.cpp.o
/home/odroid/rtabmap/utilite/src/UConversion.cpp:295:13: note: the mangling of 'va_list' has changed in GCC 4.4
[  4%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/ULogger.cpp.o
In file included from /home/odroid/rtabmap/utilite/src/ULogger.cpp:20:0:
/home/odroid/rtabmap/utilite/src/../include/rtabmap/utilite/ULogger.h:458:18: note: the mangling of 'va_list' has changed in GCC 4.4
[  5%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UThread.cpp.o
In file included from /home/odroid/rtabmap/utilite/src/UThread.cpp:21:0:
/home/odroid/rtabmap/utilite/src/../include/rtabmap/utilite/ULogger.h:458:18: note: the mangling of 'va_list' has changed in GCC 4.4
[  5%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UTimer.cpp.o
In file included from /home/odroid/rtabmap/utilite/src/UTimer.cpp:21:0:
/home/odroid/rtabmap/utilite/src/../include/rtabmap/utilite/ULogger.h:458:18: note: the mangling of 'va_list' has changed in GCC 4.4
[  6%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UProcessInfo.cpp.o
Linking CXX shared library ../../../bin/librtabmap_utilite.so
[  6%] Built target rtabmap_utilite
Scanning dependencies of target uresourcegenerator
[  7%] Building CXX object utilite/resource_generator/CMakeFiles/uresourcegenerator.dir/main.cpp.o
In file included from /home/odroid/rtabmap/utilite/resource_generator/../include/rtabmap/utilite/UtiLite.h:241:0,
                 from /home/odroid/rtabmap/utilite/resource_generator/main.cpp:20:
/home/odroid/rtabmap/utilite/resource_generator/../include/rtabmap/utilite/ULogger.h:458:18: note: the mangling of 'va_list' has changed in GCC 4.4
Linking CXX executable ../../../bin/uresourcegenerator
[  7%] Built target uresourcegenerator
[  8%] [Creating resources]
 Using namespace=rtabmap
 Using target directory=/home/odroid/rtabmap/build/corelib/src
Input file "/home/odroid/rtabmap/corelib/src/resources/DatabaseSchema.sql" size = 3695 bytes
Output file "/home/odroid/rtabmap/build/corelib/src/DatabaseSchema_sql.h" size = 7605 bytes
[... 70 lines cut ...]
Aborted (core dumped)
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/Rtabmap.cpp.o] Error 134
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [all] Error 2
==5928== 
==5928== HEAP SUMMARY:
==5928==     in use at exit: 55,364 bytes in 1,119 blocks
==5928==   total heap usage: 3,139 allocs, 2,020 frees, 144,901 bytes allocated
==5928== 
==5928== LEAK SUMMARY:
==5928==    definitely lost: 0 bytes in 0 blocks
==5928==    indirectly lost: 0 bytes in 0 blocks
==5928==      possibly lost: 0 bytes in 0 blocks
==5928==    still reachable: 55,364 bytes in 1,119 blocks
==5928==         suppressed: 0 bytes in 0 blocks
==5928== Rerun with --leak-check=full to see details of leaked memory
==5928== 
==5928== For counts of detected and suppressed errors, rerun with: -v
==5928== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)

______________________________________

Which wasn't really helpful. Would you have any idea as to what could be going wrong? 

Thanks!

matlabbe, 2015-06-09:
How much RAM do you have? Do you have a swap space? Maybe the compiler run out of memory when building Rtabmap.cpp. I've seen that on some compilers, some files of rtabmap require over 800 MB of RAM to compile. Can you check your memory while building?

tanmayshankar, 2015-06-09:
Hi Mathieu, 

The Odroid I was running it on has 2GB of RAM, but no swap. I didn't check how much memory it was consuming, but it's possible it's trying to allot the memory that doesn't exist. 
Is it possible for me to run this without swap? If not, I shall try realloting some space as swap and trying to run rtabmap.

matlabbe, 2015-06-09:
You can very how much memory is used with "top" command on console or using the system monitor.

To increase the swap without reinstalling Ubuntu, you may check this post: http://stackoverflow.com/questions/19761226/how-to-compile-ruby-with-rvm-on-a-low-memory-system

tanmayshankar, 2015-06-09:
Hi Mathieu, 

I set up a small swap section (256 Mb), and ran make clean && make, with my system monitor open (as well as my Chromium browser) - here is a screenshot. It doesn't appear like it's overflowing the memory. 

[image]

Is it possible I put on a pre-compiled version and change my CMakeCache or something similar to get this working? I assume it won't be that simple, since executables and library linkings will differ.

tanmayshankar, 2015-06-17:
Hi Mathieu, 

Does it matter that I am trying to compile rtabmap on an ARM based processor? I have been trying to find the binary for Ubuntu 12.04 / Hydro, but I am not sure whether you had released one for armhf architecture. Any suggestions on how I can solve this issue?

matlabbe, 2015-06-18:
ROS provides binaries for amd64 and i386. It looks like it is possible here to install the binaries : http://wiki.ros.org/hydro/Installation/UbuntuARM

I didn't try yet to build RTAB-Map on ARM, but if OpenCV and PCL can be built, RTAB-Map should too.

tanmayshankar, 2015-06-18:
I'm on the armhf architecture, and after spending quite a while updating sources and searching around, it doesn't look like there is a binary. 
Logically, it should be compiling - one stack overflow post suggested disabling Eigen vectorization to sort out a similar error to what I have. http://stackoverflow.com/questions/22825794/double-free-or-corruption-run-time-error-while-running-pclmovingleastsquares-c

The issue is this requires a source compilation of pcl libraries, which I had tried to install previously, but failed due to Boost version issues. As you can see, I'm in a soup.

tanmayshankar, 2015-06-19:
Hello Mathieu, 

Some good news! RTAB-Map compiled from source. Here is what i needed to do - 
0) Add a good 1Gb of swap just in case (not required in my opinion). 
1) Install Boost from source (version 1.47 or higher, required for PCL). 
2) Download PCL source package (version 1.7.2, latest release). 
3) Follow the steps in the above link, to disabled Eigen vectorization. 
4) Install PCL from source (I see what you mean when you said if it can compile PCL, it can do RTAB-Map. The PCL build took 4 hours). 
5) Modify RTAB-Map Cmake to find source PCL-1.7.2 instead of the binary installation. 
6) Clean make of RTAB-Map from source. 

Perhaps we can add this to the wiki as a heads up for ARM users?

closed 2015-06-19 by tanmayshankar

mryellow, 2015-06-19:
Got through a successful build on oDroid C1.

Used external USB for swap, didn't need PCL from source, no trouble with Eigen.

Took a few goes with USB falling asleep or something but got through in the end after 6 hours.

Now, same problem as I had with packages, crashing at startup. Might be I do need some of these other source packages, probably a bit of a stretch for C1 all the same.