introlab/rtabmap #31: can't install the package

Annaiyan, 2015-08-27:
hello,

I'm trying to install the rtabmap by following the instructions.Bu ti'm getting the following error
1.
CMake Error at src/rtabmap_ros/CMakeLists.txt:20 (find_package):
  Could not find a configuration file for package "RTABMap" that is
  compatible with requested version "0.10.5".

  The following configuration files were considered but not accepted:

    /opt/ros/indigo/lib/x86_64-linux-gnu/rtabmap-0.10/RTABMapConfig.cmake, version: 0.10.4
    /opt/ros/indigo/lib/x86_64-linux-gnu/rtabmap-0.10/RTABMapConfig.cmake, version: 0.10.4

2.CMake Error at /usr/share/cmake-2.8/Modules/FindPkgConfig.cmake:283 (message):
  A required package was not found

Can anyone help me to solve the issue?

Thanks in advance.

matlabbe, 2015-08-27:
If you are building `rtabmap_ros` from source, you should build `rtabmap` from source too (to match the version). Uninstall `ros-indigo-rtabmap`:
 ```
sudo apt-get remove ros-indigo-rtabmap
```

Then follow instructions from [here](https://github.com/introlab/rtabmap_ros#build-from-source). 

cheers

Annaiyan, 2015-08-27:
I tried the above the instructions,but still i'm getting the same error.Now i un-installed everything and i would like install by follwing below steps,
1.Install the RTAB standalone libraries in the catkin/devel folder.
2.Install the RTAB-ROS pkg in the catkin/src folder.
3.then catkin_make.

As per the above the comment,i installed the both the RTAB and RTAB-ROS pkg in the src folder,but it didn't work!

correct me,if i'm wrong!

Thanks in advance.

matlabbe, 2015-08-27:
It is not explicitly explained in the [instructions](https://github.com/introlab/rtabmap_ros#build-from-source), but rtabmap library should be built outside the catkin workspace because it is not a common catkin package (it is a plain cmake package). Cloning rtabmap in src folder of catkin will result in errors. 

Merging all directives in one walkthrough:
```bash
$ source /opt/ros/indigo/setup.bash

$ cd ~
$ git clone https://github.com/introlab/rtabmap.git rtabmap
$ cd rtabmap/build
$ cmake -DCMAKE_INSTALL_PREFIX=~/catkin_ws/devel ..
$ make -j4
$ make install

$ source ~/catkin_ws/devel/setup.bash

$ cd ~/catkin_ws
$ git clone https://github.com/introlab/rtabmap_ros.git src/rtabmap_ros
$ catkin_make
```

If you still have problems, a copy-paste of the catkin_make errors would be useful.

cheers

closed 2016-02-24 by matlabbe

walekarsayali, 2020-10-16:
I tried to implement above mentioned method with ROS-Melodic. Following error occurred :
Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Could NOT find move_base_msgs (missing: move_base_msgs_DIR)
-- Could not find the required component 'move_base_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "move_base_msgs"
  with any of the following names:

    move_base_msgsConfig.cmake
    move_base_msgs-config.cmake

  Add the installation prefix of "move_base_msgs" to CMAKE_PREFIX_PATH or set
  "move_base_msgs_DIR" to a directory containing one of the above files.  If
  "move_base_msgs" provides a separate development package or SDK, be sure it
  has been installed.

walekarsayali, 2020-10-16:
@matlabbe I tried to implement above mentioned method with ROS-Melodic. Following error occurred :
Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Could NOT find move_base_msgs (missing: move_base_msgs_DIR)
-- Could not find the required component 'move_base_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
Could not find a package configuration file provided by "move_base_msgs"
with any of the following names:

move_base_msgsConfig.cmake
move_base_msgs-config.cmake

Add the installation prefix of "move_base_msgs" to CMAKE_PREFIX_PATH or set
"move_base_msgs_DIR" to a directory containing one of the above files. If
"move_base_msgs" provides a separate development package or SDK, be sure it
has been installed.

matlabbe, 2020-10-18:
`sudo apt install ros-melodic-move-base-msgs`

zhaohaowu, 2021-02-25:
thanks to the answer of matlabbe, my question is 
`cmake -DCMAKE_INSTALL_PREFIX=~/catkin_ws/devel ..`
here my workspace is test, replace catkin_ws with test is ok

matlabbe, 2021-02-25:
This command:
```
cmake -DCMAKE_INSTALL_PREFIX=~/catkin_ws/devel ..
```
will install in your catkin workspace's devel directory.  It is more for preference, if you want to install rtabmap library like any linux libraries, just do:
```
cmake ..
```
and it will be installed in `/usr/local` prefix.

jakb2403, 2021-07-19:
@matlabbe Thanks for the walkthrough. After following the instructions and executing `catkin_make` I got the following errors:
```
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Warning at rtabmap_ros/CMakeLists.txt:28 (find_package):
  By not providing "Findapriltag_ros.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "apriltag_ros", but CMake did not find one.

  Could not find a package configuration file provided by "apriltag_ros" with
  any of the following names:

    apriltag_rosConfig.cmake
    apriltag_ros-config.cmake

  Add the installation prefix of "apriltag_ros" to CMAKE_PREFIX_PATH or set
  "apriltag_ros_DIR" to a directory containing one of the above files.  If
  "apriltag_ros" provides a separate development package or SDK, be sure it
  has been installed.

CMake Error at rtabmap_ros/CMakeLists.txt:34 (find_package):
  By not providing "FindRTABMap.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "RTABMap", but
  CMake did not find one.

  Could not find a package configuration file provided by "RTABMap"
  (requested version 0.20.13) with any of the following names:

    RTABMapConfig.cmake
    rtabmap-config.cmake

  Add the installation prefix of "RTABMap" to CMAKE_PREFIX_PATH or set
  "RTABMap_DIR" to a directory containing one of the above files.  If
  "RTABMap" provides a separate development package or SDK, be sure it has
  been installed.
```

Following the same logic as your answer to @walekarsayali, I tried:
`sudo apt install ros-noetic-rtabmap`
But then I got the same error message as in the original post on this issue:
```
CMake Error at rtabmap_ros/CMakeLists.txt:34 (find_package):
  Could not find a configuration file for package "RTABMap" that is
  compatible with requested version "0.20.13".

  The following configuration files were considered but not accepted:

    /opt/ros/noetic/lib/x86_64-linux-gnu/rtabmap-0.20/RTABMapConfig.cmake, version: 0.20.10
```
which makes sense because that just installs `rtabmap_ros` (not from source).

What should I do to solve the first error I quoted here?
Thanks for your help!

jakb2403, 2021-07-19:
Sorry the above (https://github.com/introlab/rtabmap/issues/31#issuecomment-882335533) is a duplicate.
The answer to this question can be found here: https://github.com/introlab/rtabmap_ros/issues/118#issuecomment-390713608