introlab/rtabmap #33: Error Exporting to a PLY file

kaifeichen, 2015-09-30:
I got this error when exporting to a PLY file.

```
** (rtabmap:6408): CRITICAL **: murrine_style_draw_focus: assertion 'height >= -1' failed

** (rtabmap:6408): CRITICAL **: murrine_style_draw_focus: assertion 'height >= -1' failed

** (rtabmap:6408): CRITICAL **: murrine_style_draw_focus: assertion 'height >= -1' failed
[ WARN] (2015-09-30 13:29:04.526) Memory.cpp:329::init() _vwd->getUnusedWordsSize() must be empty... size=55
[pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow.Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt. You must
reimplement QApplication::notify() and catch all exceptions there.

terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
```

matlabbe, 2015-09-30:
I can't reproduce the error with the binaries or the latest version from source. Are you using the default exporting parameters? For more information, you can set the Logger in Debug mode: Preferences->Logging->Logger level.

kaifeichen, 2015-09-30:
I cannot reproduce either.. I'll close this first and open it again next time I see it with log-level information. Thanks!

closed 2015-09-30 by kaifeichen