introlab/rtabmap #37: Multiple RGBD cameras input sources

kingfolk, 2015-10-19:
Hi Dear rtabmap,

The question is not technically a github issue. Sorry. But i don't know where to ask.
The practice we try to do is use two RGBD cameras. The additional camera's angle is fixed down to floor so as to make robot aware of the the environment of floor.
I looked into the SensorData.h https://github.com/introlab/rtabmap/blob/0.10.5/corelib/include/rtabmap/core/SensorData.h#L86
There is a constructor for multi rgbd cameras but no usage link to it. I am wondering if there is an example of using multi rgbd cameras as input sources.

many thanks

matlabbe, 2015-10-19:
You can ask this type of question here, no problems!

If you are using rtabmap_ros, there is a small example here: https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_two_kinects.launch

For the constructor of SensorData you are referring:
 * rgb: side-by-side RGB images (e.g., it would be 1280x480 for two images of 640x480)
 * depth: side-by-side depth images (e.g., it would be 1280x480 for two images of 640x480)
 * cameraModels: The [CameraModel](https://github.com/introlab/rtabmap/blob/0.10.5/corelib/include/rtabmap/core/CameraModel.h) for each camera. Make sure that [localTransform](https://github.com/introlab/rtabmap/blob/0.10.5/corelib/include/rtabmap/core/CameraModel.h#L61) of CameraModel represents exactly the transform from a fixed frame on the robot (e.g., /base_link on ROS). The localTransform should also contains the optical rotation R to put the images in x->forward, y->left, z->up referential. Example: localTransform = T*R, where T is the transformation of the camera from the fixed frame on the robot (e.g., your second camera may have pitch=PI/2 and some translation components) and R the optical rotation (roll=-PI/2, pitch=0, yaw=-PI/2).

cheers

kingfolk, 2015-10-19:
Super!
Thank you!

kingfolk, 2015-11-09:
Hi Mathieu,

i am wondering how you calibrate two rgb cameras?

thanks!

matlabbe, 2015-11-09:
The TFs should be set manually. An automated calibration while mapping with both cameras is not yet implemented.

cheers

closed 2016-02-24 by matlabbe

jacksonkr, 2016-10-28:
@kingfolk 

FYI there is the official rtabmap forum here: http://official-rtab-map-forum.67519.x6.nabble.com/

It's where I ask all my non-issue related questions.