introlab/rtabmap #38: Can the rtabmap be configured without using visual sensors?

somebodyus, 2015-10-21:
I have wheel encoder, hokuyo lidar and IMU. I would like to generate 2-D map.
Can the rtabmap be configured using above sensors without Kinect or stereo vision?

matlabbe, 2015-10-21:
No, rtabmap needs RGB images for loop closure detection. You may want to try [hector_mapping](http://wiki.ros.org/hector_mapping) or [gmapping](http://wiki.ros.org/slam_gmapping) packages.

cheers

somebodyus, 2015-10-21:
Thank you so much.
Can I use a web-cam instead of a RGB-D camera for loop closure? If yes, how can I configure the launch file?

matlabbe, 2015-10-21:
No, rtabmap needs a depth image too. This could be possible to do though with some coding and if the laser is long-range like >20 meters (like modifying the code to bypass the visual transformation estimation and go directly with ICP transformation estimation).

cheers

closed 2016-02-24 by matlabbe