introlab/rtabmap #39: Not found word #### with Mem/ReduceGraph=true

matlabbe, 2015-10-26:
Not linked nodes are not saved in database too.

```bash
[ERROR] (2015-10-26 00:22:47.593) DBDriverSqlite3.cpp:1789::loadWordsQuery() Query (50) doesn't match loaded words (45)
[ERROR] (2015-10-26 00:22:47.595) VWDictionary.cpp:736::addWordRef() Not found word 477313
[ERROR] (2015-10-26 00:22:47.595) VWDictionary.cpp:736::addWordRef() Not found word 477436
[ERROR] (2015-10-26 00:22:47.596) VWDictionary.cpp:736::addWordRef() Not found word 616983
[ERROR] (2015-10-26 00:22:47.596) VWDictionary.cpp:736::addWordRef() Not found word 659393
[ERROR] (2015-10-26 00:22:47.596) VWDictionary.cpp:736::addWordRef() Not found word 659393
[ERROR] (2015-10-26 00:22:47.596) VWDictionary.cpp:736::addWordRef() Not found word 659393
[ERROR] (2015-10-26 00:22:47.596) VWDictionary.cpp:736::addWordRef() Not found word 781890
```

phuicy, 2016-08-11:
Delete your database. In particular, ~/.ros/rtabmap.db

matlabbe, 2023-02-26:
Closing as error not happened during experiences of this paper (2021/2022) https://github.com/introlab/rtabmap/tree/master/archive/2022-IlluminationInvariant

closed 2023-02-26 by matlabbe