introlab/rtabmap #40: Handling removed obstacles

leftys, 2015-11-12:
Hello,

I have tried rtabmap on our Universitys robot and read the “Online Global Loop Closure Detection for Large-Scale Multi-Session Graph-Based SLAM” article, but I haven't found anything about how rtabmap handles changes in the map. Or doesn't it? 

I conducted an experiment with XTion rgb-d camera, rtabmap and rtabmapviz. I placed an obstacle into the scene and then removed it. It appeared in the map but sometimes was not removed. I suppose rtabmapviz select only some nodes to show...? 

And if rtabmap doesn't remove points from the map, the only changes happening in the map are loop closures. Are they propagated into the online octomap? I think it would require the whole octomap to be rebuilt. Does map_assembler node reflect loop closures in its map?

Thank you!

matlabbe, 2015-11-12:
Hi,

It handles dynamic objects implicitly when a new node is added in the map, but only for 2D mapping (2D maps). For example, the new laser scan will write over obstacle cells to clean objects removed. If the robot is not moving, the map is not updated though, so the removed object would be still there. For the 3D point cloud, there is no filtering (only temporary if you set `<param name="cloud_frustum_culling" type="bool" value="true"` for the `rtabmap/cloud_map` output topic). 

rtabmapviz and MapCloud plugin don't do frustum culling. They only keep the latest node in a fixed radius. In rtabmapviz, you can change this setting under 3D rendering panel in Preferences. For [MapCloud](http://wiki.ros.org/rtabmap_ros#Map_Cloud_Display) plugin in RVIZ, there are `Node filtering radius` and `Node filtering angle` parameters. For all map output topics from [rtabmap](http://wiki.ros.org/rtabmap_ros#rtabmap) node: see `map_filter_radius` and `map_filter_radius` parameters.

The octomap can be generated from a service request to rtabmap node: `/rtabmap/octomap_full` or `/rtabmap/octomap_binary` services. Or you can connect `/rtabmap/cloud_map` topic to an [octomap_server](http://wiki.ros.org/octomap_server), for which an octomap will be built again on each map update (so including graph optimization following a loop closure). You may want to tune the output `/rtabmap/cloud_map` topic to generate less points. See `cloud_map` related parameters of [rtabmap](http://wiki.ros.org/rtabmap_ros#rtabmap) node. Well, the [map_assembler](http://wiki.ros.org/rtabmap_ros#map_assembler) node is not required anymore to assemble a point cloud, the rtabmap node does it too as shown above.

cheers

leftys, 2015-11-17:
In that case, I will build my own octomap from sensor and location data and use raytracing to clean map from removed obstacles.  Loop closure will be probably very expensive and I may have to integrate only some nodes (discard similar ones), but I believe it should be feasible. Thank you!

closed 2015-11-17 by leftys