introlab/rtabmap #49: Running only using pointclouds

andre-nguyen, 2016-01-25:
Quick question, would it be possible to run rtabmap solely using point clouds generated from laser rangefinders (e.g. 2D Hokuyo lidar mounted on a slip ring or the new Velodyne vlp-16), admitting a few modifications.

Or is not possible because you depend on SURF features (or the like) for operation?

Thanks!

matlabbe, 2016-02-24:
Sorry I missed that question, better now than never...

Yes RTAB-Map depends on visual features for loop closure detection. If you add a camera to your setup it is possible to use these kinds of lidar.

cheers

chukcha2, 2016-02-24:
@matlabbe, how would you use RTAB-Map+camera in this setup? Only to get localization data to register lidar point clouds to get 3D model? Or would you use camera+RTAB-Map to get 3D model itself from the visual data (then it seems like lidar would not be needed)? What are the (dis)advantages of these methods? Thank you.

matlabbe, 2016-02-24:
Hi,

Here are some combinations:

- RTAB-Map + stereo camera = localization + 3D point clouds from the camera
- RTAB-Map + stereo camera + lidar = localization + 3D point clouds from the camera + 3D point clouds from the lidar
- RTAB-Map + RGB-D camera = localization + 3D point clouds from the camera
- RTAB-Map + RGB-D camera + lidar = localization + 3D point clouds from the camera + 3D point clouds from the lidar

With a single RGB camera and a 3D lidar, it would be possible using the lidar data to "fake" a depth image registered with the RGB camera. This would give the setup RGB-D+lidar above.

cheers,
Mathieu

andre-nguyen, 2016-02-24:
Merci Mathieu!

Super votre travail à Sherbrooke :smile:

closed 2016-02-24 by andre-nguyen

matlabbe, 2016-02-25:
Bienvenu!

chukcha2, 2016-03-02:
Thank you Mathieu, I'll try to use a RTAB-Map + stereo camera (ZED) + lidar (VLP-16). If you have any pointers on how to combine ldiar data with stereo camera's data, please let me know.

bj-neilson, 2020-02-25:
@chukcha2 
Hey, long time since your last post on this but if you get this message I’m wondering how you went?  I’m doing essentially the same now. Rtabmap has continued to evolve thanks to @matlabbe so I’m hoping for a great result!
Cheers, Ben