introlab/rtabmap #52: Problem about transplant rtabmap into android

diophantine86, 2016-02-23:


matlabbe, 2016-02-23:
Are there more details?

Note that rtabmap will only work on Google Tango as it requires a depth camera (or a stereo camera).

diophantine86, 2016-02-24:
Thanks! I develop a project something like tango but not  use project tango.I wanna transplant rtab into my project.now , i do not know how to compile opencv and pcl in android . can u give me the code on the google tango?

matlabbe, 2016-02-24:
Try to build OpenCV android examples before rtabmap. On google tango, I used [NVidia tegra OpenCV](http://docs.nvidia.com/gameworks/content/technologies/mobile/opencv_tegra_examples.htm) as a starting package.

cheers

diophantine86, 2016-02-25:
I have built OpenCV Android examples.but in this code,it also have point cloud library .can u tell me how to build pcl on android. thx.

sungwookson, 2016-05-10:
I was wondering if someone can help me port this repo to android. I have installed the prerequisites and trying to install it on Google Tango. When I try to open the repo using android studio, the android studio does not recognize the project

matlabbe, 2016-05-11:
See #74. This is a cmake project, you may be able to create an android project from this folder: https://github.com/introlab/rtabmap/tree/master/app/android

matlabbe, 2016-05-20:
Closed as duplicate of #74

closed 2016-05-20 by matlabbe