introlab/rtabmap #53: Not Enough Correspondences

omair18, 2016-03-01:
Hey. Thanks for this great package. I have tried it using rosbags and it worked great.
But when i tested my own data-set using rectified images it gave me warnings of "not enough correspondences found. I tried to find correspondence using MATLAB and it showed more than 40-50 per image ( 1 from left camera and 2nd from right camera. Both cameras are calibrated using ROS).  I have attached snapshots. Can you please suggest a solution regarding this matter? 
[image]
[image]

matlabbe, 2016-03-01:
Because the visual features are far (>3-4 meters), try 3D -> 2D motion estimation like in the screenshot below. If this doesn't help, set the Logger level to debug (Preferences->Logging) and copy/paste the output here.

[image]

omair18, 2016-03-02:
Thank you for the help. I have made the above mentioned changes but i'm still stuck with the same issue. I have tried changing odometry strategy and visual word type to see what changes it does with output. [Log file](https://drive.google.com/file/d/0BwVVozr6udESMDczQU9McUhFTmM/view?usp=sharing) for Rtabmap with SURF features and Optical flow is [here](https://drive.google.com/file/d/0BwVVozr6udESMDczQU9McUhFTmM/view?usp=sharing). [Log file ](https://drive.google.com/file/d/0BwVVozr6udESRUhxZEtRYUpVdGs/view?usp=sharing)with SIFT & BOW is [here.](https://drive.google.com/file/d/0BwVVozr6udESRUhxZEtRYUpVdGs/view?usp=sharing) 
[image]
[image]

matlabbe, 2016-03-02:
It looks like there is a problem with the calibration. Around 1200 features are extracted by the odometry but only ~100 can be triangulated in 3D. In these 100, maybe only 10 would be inliers so not enough to correctly find a transformation between the frames.

Can you do a small recording? Click Tools->"Data recorder...", start a new session, then on next "Stop", the recorded database will be saved. 

cheers

omair18, 2016-03-03:
I have done a small recording for 50-55 images. You can see the file here. 
[output.db.zip](https://github.com/introlab/rtabmap/files/156177/output.db.zip)

matlabbe, 2016-03-03:
Did you follow this [tutorial](http://wiki.ros.org/camera_calibration/Tutorials/StereoCalibration) to calibrate the cameras?

The stereo rectification is wrong, see this image (I opened the database with rtabmap-databaseViewer):
[image]
These are the left and right images superposed. Normally it should only have a disparity on x-axis, not both axes. 

Example of good stereo rectification:
[image]

Note that you can also use the Calibrate tool in RTAB-Map (Preferences->Source->Calibrate) to calibrate the cameras (it is mostly the same interface as the ROS tutorial). There is an example [here](https://github.com/introlab/rtabmap/wiki/Stereo-mapping#calibration-only-need-to-do-this-once-by-camera-setup) (you can do it with the live images of the camera, no need to use a video file).

cheers

closed 2016-11-09 by matlabbe