introlab/rtabmap #62: install from source code on windows

wengxiuling36, 2016-03-23:
hello,I install it according to https://github.com/introlab/rtabmap/wiki/Installation#windows.
I finished the first eight steps.But the next step I did't discover RTABMap.exe.So how can I get the RTABMap.exe.I will show you what I get now:
[image]
[image]

Wade5566, 2016-03-24:
Your project list is a little different from mine.
Here is my project list.
[image]
I think you miss some dependent libraries.
For example, you do not have rtabmap_gui.dll so I guess you didn't build Qt or something.

matlabbe, 2016-03-24:
Indeed, RTAB-Map.exe is built only if Qt is detected by CMake.

wengxiuling36, 2016-03-24:
[image]
But how can I deal with it ?

wengxiuling36, 2016-03-24:
I  add these three lines to solve it.But another error appears.I add 'D:\pcl\4.8.0\bin' to noEventsExampleProject -> Properties -> Configuration Properties -> VC++ Directories -> Executable Directories.It does't work.
[image]
[image]

Wade5566, 2016-03-24:
Set the "QT_MOC_EXECUTABLE" a full path for the executable file, moc.exe.

matlabbe, 2016-03-24:
You don't have to change the `CMakeLists.txt`. CMake tries already to find Qt [here](https://github.com/introlab/rtabmap/blob/18759e7197d627ec1b2d7e1ddb4cdcbf36131a02/CMakeLists.txt#L171-L175) (setting all Qt macros):
```cmake
IF("${RTABMAP_QT_VERSION}" STREQUAL "4")
        FIND_PACKAGE(Qt4 COMPONENTS QtCore QtGui QtSvg)
ELSE()
        FIND_PACKAGE(Qt5 COMPONENTS Widgets Core Gui Svg)
ENDIF()
```
Which Qt version do you have?
 - For Qt4, make sure the bin directory of Qt is in the `PATH` environment variable so that CMake can find `qmake` tool when searching for Qt.
 - For Qt5, you can put in the PATH the directory containing Qt5Config.cmake (somewhere inside the Qt5 installation directory) or execute CMake like this: `cmake -DRTABMAP_QT_VERSION=5 -DQt5_DIR=path/to/directory/of/Qt5Config.cmake`

cheers

wengxiuling36, 2016-03-25:
Thanks to you all.
@matlabbe I use qt4 and bin directory of Qt is in the `path` environment.I wonder there is no FindQT.cmake file in cmake_modules directory ,how can it find qt?
@matlabbe  I set a full path.I  can get release version,but can't get debug version.Is it related to PCL debug version.I did't compile version of PCL.

wengxiuling36, 2016-03-25:
Thanks to you all.
@matlabbe  I use qt4 and bin directory of Qt is in the path environment.I wonder there is no FindQT.cmake file in cmake_modules directory ,how can it find qt?
@Wade5566  I set a full path.It works.
Now,I have new qustion.I can get release version,but can't get debug version.
[image]

wengxiuling36, 2016-03-25:
Thanks to you all.
@matlabbe I use qt4 and bin directory of Qt is in the` path` environment.I wonder there is no FindQT.cmake file in cmake_modules directory ,how can it find qt?
@Wade5566 I set a full path.It works.

wengxiuling36, 2016-03-25:
Thanks to you all.
@matlabbe I use qt4 and bin directory of Qt is in the` path `environment.I wonder there is no `FindQT.cmake `file in cmake_modules directory ,how can it find qt?
@Wade5566 I set a full path.It works.

matlabbe, 2016-03-25:
`FindQt4.cmake` is already in CMake installation (cmake/share/modules/FindQt4.cmake). What is your CMake output?

wengxiuling36, 2016-03-26:
I find` FindQt4.cmake` in CMake installation.It is so interesting.I don't know why CMake output this:
Could NOT find Qt4 (missing:  `QT_MOC_EXECUTABLE`` QT_RCC_EXECUTABLE` `QT_UIC_EXECUTABLE QT_INCLUDE_DIR`) (found version "4.8.0")

wengxiuling36, 2016-03-26:
I find`FindQt4.cmake `in CMake installation.It is so interesting.I don't know why CMake output this:
Could NOT find Qt4 (missing:  `QT_MOC_EXECUTABLE ` `QT_RCC_EXECUTABLE ` `QT_UIC_EXECUTABLE`) (found version "4.8.0")
[image]

matlabbe, 2016-03-26:
It is like the development tools are not found by CMake. You could try to execute cmake-gui in Administrator mode. Otherwise, use the solution of @Wade5566 by setting manually the missing tool path.

wengxiuling36, 2016-03-26:
OK.Thank for your help.

closed 2016-03-30 by lynnsky