introlab/rtabmap #63: Regression error on ROS buildfarm

matlabbe, 2016-03-23:
The error happens when packaging the .deb:
```
dpkg-shlibdeps -Tdebian/ros-jade-rtabmap.substvars -l/tmp/binarydeb/ros-jade-rtabmap-0.10.10/debian/ros-jade-rtabmap//opt/ros/jade/lib/ debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-databaseViewer debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-dataRecorder debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-console debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-odometryViewer debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-extractObject debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-calibration debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-camera debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-rgbd_camera debian/ros-jade-rtabmap/opt/ros/jade/lib/x86_64-linux-gnu/librtabmap_utilite.so.0.10.10 debian/ros-jade-rtabmap/opt/ros/jade/lib/x86_64-linux-gnu/librtabmap_gui.so.0.10.10 debian/ros-jade-rtabmap/opt/ros/jade/lib/x86_64-linux-gnu/librtabmap_core.so.0.10.10
dpkg-shlibdeps: error: couldn't find library librtabmap_gui.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-databaseViewer (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_utilite.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-databaseViewer (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_core.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_gui.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_utilite.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_core.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-console (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_utilite.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-console (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_utilite.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-extractObject (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_core.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-dataRecorder (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_gui.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-dataRecorder (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_utilite.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-dataRecorder (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_utilite.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/lib/x86_64-linux-gnu/librtabmap_core.so.0.10.10 (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_core.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-camera (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_utilite.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-camera (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_core.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-rgbd_camera (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_utilite.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-rgbd_camera (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_core.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-calibration (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_gui.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-calibration (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_utilite.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-calibration (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_core.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/lib/x86_64-linux-gnu/librtabmap_gui.so.0.10.10 (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_utilite.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/lib/x86_64-linux-gnu/librtabmap_gui.so.0.10.10 (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_core.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-odometryViewer (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_gui.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-odometryViewer (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: error: couldn't find library librtabmap_utilite.so.0.10 needed by debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-odometryViewer (ELF format: 'elf64-x86-64'; RPATH: '')
dpkg-shlibdeps: warning: package could avoid a useless dependency if debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-extractObject debian/ros-jade-rtabmap/opt/ros/jade/lib/x86_64-linux-gnu/librtabmap_core.so.0.10.10 were not linked against libpcl_recognition.so.1.7 (they use none of the library's symbols)
dpkg-shlibdeps: error: cannot continue due to the errors listed above
Note: libraries are not searched in other binary packages that do not have any shlibs or symbols file.
To help dpkg-shlibdeps find private libraries, you might need to use -l.
dh_shlibdeps: dpkg-shlibdeps -Tdebian/ros-jade-rtabmap.substvars -l/tmp/binarydeb/ros-jade-rtabmap-0.10.10/debian/ros-jade-rtabmap//opt/ros/jade/lib/ debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-databaseViewer debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-dataRecorder debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-console debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-odometryViewer debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-extractObject debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-calibration debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-camera debian/ros-jade-rtabmap/opt/ros/jade/bin/rtabmap-rgbd_camera debian/ros-jade-rtabmap/opt/ros/jade/lib/x86_64-linux-gnu/librtabmap_utilite.so.0.10.10 debian/ros-jade-rtabmap/opt/ros/jade/lib/x86_64-linux-gnu/librtabmap_gui.so.0.10.10 debian/ros-jade-rtabmap/opt/ros/jade/lib/x86_64-linux-gnu/librtabmap_core.so.0.10.10 returned exit code 2
```
References: 
 - Indigo: http://build.ros.org/view/Ibin_uT64/job/Ibin_uT64__rtabmap__ubuntu_trusty_amd64__binary/6/console
 - Jade: http://build.ros.org/view/Jbin_uT64/job/Jbin_uT64__rtabmap__ubuntu_trusty_amd64__binary/8/console

matlabbe, 2016-03-23:
Trying locally (Indigo on Virtual machine):
```bash
$ cd rtabmap
$ bloom-generate rosdebian --os-name ubuntu --os-version trusty --ros-distro indigo
$ fakeroot debian/rules binary
```
and there is no error at the same place (though some warnings):
```
[... build/install with success]
dpkg-shlibdeps -Tdebian/ros-indigo-rtabmap.substvars -l/home/mathieu/workspace/rtabmap/debian/ros-indigo-rtabmap//opt/ros/indigo/lib/ debian/ros-indigo-rtabmap/opt/ros/indigo/lib/librtabmap_gui.so.0.10.10 debian/ros-indigo-rtabmap/opt/ros/indigo/lib/librtabmap_utilite.so.0.10.10 debian/ros-indigo-rtabmap/opt/ros/indigo/lib/librtabmap_core.so.0.10.10 debian/ros-indigo-rtabmap/opt/ros/indigo/bin/rtabmap-calibration debian/ros-indigo-rtabmap/opt/ros/indigo/bin/rtabmap-odometryViewer debian/ros-indigo-rtabmap/opt/ros/indigo/bin/rtabmap-dataRecorder debian/ros-indigo-rtabmap/opt/ros/indigo/bin/rtabmap-console debian/ros-indigo-rtabmap/opt/ros/indigo/bin/rtabmap-extractObject debian/ros-indigo-rtabmap/opt/ros/indigo/bin/rtabmap-camera debian/ros-indigo-rtabmap/opt/ros/indigo/bin/rtabmap-rgbd_camera debian/ros-indigo-rtabmap/opt/ros/indigo/bin/rtabmap-databaseViewer debian/ros-indigo-rtabmap/opt/ros/indigo/bin/rtabmap
dpkg-shlibdeps: warning: can't extract name and version from library name 'libg2o_types_slam3d.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'libg2o_types_slam3d.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'libg2o_stuff.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'libg2o_stuff.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'libg2o_types_slam2d.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'libg2o_types_slam2d.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'libg2o_core.so'
dpkg-shlibdeps: warning: can't extract name and version from library name 'libg2o_core.so'
dpkg-shlibdeps: warning: symbol glPopAttrib used by debian/ros-indigo-rtabmap/opt/ros/indigo/lib/librtabmap_core.so.0.10.10 found in none of the libraries
dpkg-shlibdeps: warning: symbol glVertex3f used by debian/ros-indigo-rtabmap/opt/ros/indigo/lib/librtabmap_core.so.0.10.10 found in none of the libraries
dpkg-shlibdeps: warning: symbol glColor3f used by debian/ros-indigo-rtabmap/opt/ros/indigo/lib/librtabmap_core.so.0.10.10 found in none of the libraries
dpkg-shlibdeps: warning: symbol glEnd used by debian/ros-indigo-rtabmap/opt/ros/indigo/lib/librtabmap_core.so.0.10.10 found in none of the libraries
dpkg-shlibdeps: warning: symbol glPushAttrib used by debian/ros-indigo-rtabmap/opt/ros/indigo/lib/librtabmap_core.so.0.10.10 found in none of the libraries
dpkg-shlibdeps: warning: symbol glDisable used by debian/ros-indigo-rtabmap/opt/ros/indigo/lib/librtabmap_core.so.0.10.10 found in none of the libraries
dpkg-shlibdeps: warning: symbol glBegin used by debian/ros-indigo-rtabmap/opt/ros/indigo/lib/librtabmap_core.so.0.10.10 found in none of the libraries
dpkg-shlibdeps: warning: package could avoid a useless dependency if debian/ros-indigo-rtabmap/opt/ros/indigo/lib/librtabmap_core.so.0.10.10 debian/ros-indigo-rtabmap/opt/ros/indigo/bin/rtabmap-extractObject were not linked against libpcl_recognition.so.1.7 (they use none of the library's symbols)
make[1]: Leaving directory `/home/mathieu/workspace/rtabmap'
   dh_installdeb
Unknown option: buildsystem
dh_installdeb: warning: ignored unknown options in DH_OPTIONS
	printf '#!/bin/sh
set -e
' > debian/ros-indigo-rtabmap/DEBIAN/postinst
	cat debian/ros-indigo-rtabmap.postinst.debhelper >> debian/ros-indigo-rtabmap/DEBIAN/postinst
	chown 0:0 debian/ros-indigo-rtabmap/DEBIAN/postinst
	chmod 755 debian/ros-indigo-rtabmap/DEBIAN/postinst
	printf '#!/bin/sh
set -e
' > debian/ros-indigo-rtabmap/DEBIAN/postrm
	cat debian/ros-indigo-rtabmap.postrm.debhelper >> debian/ros-indigo-rtabmap/DEBIAN/postrm
	chown 0:0 debian/ros-indigo-rtabmap/DEBIAN/postrm
	chmod 755 debian/ros-indigo-rtabmap/DEBIAN/postrm
   dh_gencontrol
Unknown option: buildsystem
dh_gencontrol: warning: ignored unknown options in DH_OPTIONS
	echo misc:Depends= >> debian/ros-indigo-rtabmap.substvars
	dpkg-gencontrol -ldebian/changelog -Tdebian/ros-indigo-rtabmap.substvars -Pdebian/ros-indigo-rtabmap
	chmod 644 debian/ros-indigo-rtabmap/DEBIAN/control
	chown 0:0 debian/ros-indigo-rtabmap/DEBIAN/control
   dh_md5sums
Unknown option: buildsystem
dh_md5sums: warning: ignored unknown options in DH_OPTIONS
	(cd debian/ros-indigo-rtabmap >/dev/null ; find . -type f  ! -regex './DEBIAN/.*' -printf '%P\0' | LC_ALL=C sort -z | xargs -r0 md5sum > DEBIAN/md5sums) >/dev/null
	chmod 644 debian/ros-indigo-rtabmap/DEBIAN/md5sums
	chown 0:0 debian/ros-indigo-rtabmap/DEBIAN/md5sums
   dh_builddeb
Unknown option: buildsystem
dh_builddeb: warning: ignored unknown options in DH_OPTIONS
	dpkg-deb --build debian/ros-indigo-rtabmap ..
dpkg-deb: building package `ros-indigo-rtabmap' in `../ros-indigo-rtabmap_0.10.10-0trusty_amd64.deb'.
```

wjwwood, 2016-03-23:
Could it be because of `GNUInstallDirs`?

This line (from: ):

```
dh_shlibdeps -l/tmp/binarydeb/ros-indigo-rtabmap-0.10.10/debian/ros-indigo-rtabmap//opt/ros/indigo/lib/
```

Points to `indigo/lib`, but maybe your libraries are getting put into a different folder, like `lib/x86_64` or something like that? I can't explain why it doesn't happen with fakeroot. You can try running the build farm's docker job locally, which might give you an opportunity to investigate it in more detail.

You could also change the above line to include the folder you would expect `GNUInstallDirs` to use. That line is in the rules file in your release repository and can be patched by editing it directly on GitHub and then rerunning bloom (double check you change is there in the release repository afterwards):

https://github.com/introlab/rtabmap-release/blob/debian/indigo/trusty/rtabmap/debian/rules#L55

matlabbe, 2016-03-23:
Thank you, yes [GNUInstallDirs](https://github.com/introlab/rtabmap/blob/1f999c4d2899ce4d5589dd9f2d166eda8c9dae2e/CMakeLists.txt#L13) is used. It seems effectively the problem looking at the install section of the log. For example:
```
-- Installing: /tmp/binarydeb/ros-indigo-rtabmap-0.10.10/debian/ros-indigo-rtabmap/opt/ros/indigo/lib/x86_64-linux-gnu/librtabmap_core.so.0.10.10
-- Installing: /tmp/binarydeb/ros-indigo-rtabmap-0.10.10/debian/ros-indigo-rtabmap/opt/ros/indigo/lib/x86_64-linux-gnu/librtabmap_core.so.0.10
-- Installing: /tmp/binarydeb/ros-indigo-rtabmap-0.10.10/debian/ros-indigo-rtabmap/opt/ros/indigo/lib/x86_64-linux-gnu/librtabmap_core.so
```
The libraries are installed in `indigo/lib/x86_64-linux-gnu/`. I've made the changes you suggested in debian/rules (well I think you just saw it for Indigo, I'm pushing changes on Jade now). Hope it will work!

wjwwood, 2016-03-23:
Ok, just make sure you try it out after the deb builds because I'm not sure whether or not our setup file will add `/opt/ros/<distro>/lib/x86_64-linux-gnu` to the `LD_LIBRARY_PATH` (or do something else to address the issue). See: https://github.com/ros/catkin/issues/545

Another option would be to disable GNUInstallDirs when building in our farm (by way of CMake variable passed in the rules files for example).

matlabbe, 2016-03-23:
The last successful build of the same version (February 1):
http://build.ros.org/view/Ibin_uT64/job/Ibin_uT64__rtabmap__ubuntu_trusty_amd64__binary/3/console
did work with GNUInstallDirs (libraries were also installed under lib/x86_64-linux-gnu). The `dh_shlibdeps` line is also exactly the same (without the need to explicitly add the path `x86_64-linux-gnu`).

I'll look to disable GNUInstallDirs for the buildfarm.

matlabbe, 2016-03-23:
I missed that GNUInstallDirs [is already disabled](https://github.com/introlab/rtabmap-release/blob/74a9dcdf123c3db34e5ebddd45967abd2f67f0ee/CMakeLists.txt) for catkin (similar to your suggestion on this [post](https://github.com/ros-infrastructure/bloom/pull/268#issuecomment-40855499)):
```cmake
# Catkin doesn't support multiarch library path, 
# fix to "lib" if not set by user.
IF(NOT DEFINED CMAKE_INSTALL_LIBDIR)
   set(CMAKE_INSTALL_LIBDIR "lib")
ENDIF(NOT DEFINED CMAKE_INSTALL_LIBDIR)

INCLUDE(GNUInstallDirs)
```
I tested on my side and `CMAKE_INSTALL_LIBDIR` is correctly overridden to "lib" by default. It seems that on the buildfarm `CMAKE_INSTALL_LIBDIR` is set before calling rtabmap cmake. Looking at the build history, it's happening (setting `lib/x86_64-linux-gnu` or `lib/i386-linux-gnu`) since the beginning on the new buildfarm (January 1).

wjwwood, 2016-03-23:
Hmm, not sure why that would be. @dirk-thomas @tfoote

matlabbe, 2016-03-24:
Or maybe it is related to cmake version (I used 2.8.12.2) and now CMake overrides `CMAKE_INSTALL_LIBDIR` in GNUInstallDirs if included after. I'll give a try with CMake 3.

tfoote, 2016-03-24:
I see a successful build of rtabmap. What did you change? http://build.ros.org/job/Ibin_uT32__rtabmap__ubuntu_trusty_i386__binary/8/

rtabmap_ros looks like a linking error with the new release of rviz. @wjwwood fyi http://build.ros.org/job/Ibin_uT32__rtabmap_ros__ubuntu_trusty_i386__binary/11/console

wjwwood, 2016-03-24:
Looks like they have an implicit dependency on rviz's default plugin, same issue here: https://github.com/ros-planning/moveit_ros/issues/654

matlabbe, 2016-03-24:
1) I added `lib/x86_64-linux-gnu` and `lib/i386-linux-gnu` to debian/rules as suggested by @wjwwood :
https://github.com/introlab/rtabmap-release/blob/debian/indigo/trusty/rtabmap/debian/rules#L55

2) I fixed rviz errors, I'm doing a new release shortly. I needed to explicitly add `find_package(rviz REQUIRED)` to get `rviz_DEFAULT_PLUGIN_LIBRARIES` for linking. I have some plugins inheriting rviz plugins.

3) However, there are new errors on some builds like this one (failing on` get sourcedeb` step):
http://build.ros.org/view/Jbin_uV32/job/Jbin_uV32__rtabmap__ubuntu_vivid_i386__binary/11/
Looks like similar to this post:
https://github.com/ros-infrastructure/ros_buildfarm/issues/248

tfoote, 2016-03-26:
It seems to be building on x86 but still has a problem on armhf linking: http://build.ros.org/view/Ibin_arm_uThf/job/Ibin_arm_uThf__rtabmap__ubuntu_trusty_armhf__binary/9/console

matlabbe, 2016-03-26:
Oh I missed that one, I was looking on http://repositories.ros.org/status_page/ros_indigo_default.html. I added `lib/arm-linux-gnueabihf` to debian/rules.

tfoote, 2016-03-26:
Great it looks to have rebuilt successfully on armhf too. There's a parallel build statue page for the arm builds. http://repositories.ros.org/status_page/ros_indigo_arm.html

matlabbe, 2016-03-29:
Everything is green (or blue), I guess I can close this issue. Thx for the help!

closed 2016-03-29 by matlabbe