introlab/rtabmap #66: about CameraRGBDImages()

Wade5566, 2016-03-28:
Hi,

Thanks for the wonderful project.
I want to use the CameraRGBDImages() by the images and depth maps captured by ASUS Xtion Pro Live.
But it said that **Rectified image required! Calibrate your camera!**
How can I do?
Thank you.

matlabbe, 2016-03-29:
A calibration file should be given to [CameraRGBDImages object on initialization](https://github.com/introlab/rtabmap/blob/bcd96a97a3fca3a0e419829598a16aafed11ddb9/corelib/include/rtabmap/core/CameraRGBD.h#L289).

Example of a minimal calibration file (assuming that images are rectified and depth is already registered to RGB) arbitrarly called `xtion.yaml` (**note that values are based on Kinect, you may have to change fx and fy**) in the same directory as the executable:
```yaml
%YAML:1.0
camera_name: xtion
image_width: 640
image_height: 480
camera_matrix:
   rows: 3
   cols: 3
   data: [ 525, 0., 525, 0., 319.5,
       239.5, 0., 0., 1. ]
```

```c++
CameraRGBDImages camera;
camera.init(".", "xtion");
```

Wade5566, 2016-03-30:
Hi,
Thanks for your explanation.
I had searched for calibration file of ASUS xtion pro live but I can not find it.
Do I need to calibrate it using RTABMAP by myself?

Wade5566, 2016-03-30:
Hi, 

Thanks for your answer.
The .yaml file in your answer consist of a few mistakes in the data term and it should be    
**data: [ 525, 0., 319.5, 0., 525, 239.5, 0., 0., 1. ]**
I had tried the this file and it works on xtion!
Thanks.

matlabbe, 2016-03-30:
Yes thx, I just corrected the answer. Doing a calibration of the RGB image could give you more precise `fx` anf `fy` values. The mapping still work with any focal distances (non-null), but the scale would not be the same as the environment.

closed 2016-03-30 by matlabbe