introlab/rtabmap #69: memory management problem

wengxiuling36, 2016-03-30:
I use the commands on the[ Benchmark](https://github.com/introlab/rtabmap/wiki/Benchmark#udes-1hz) page.But the time used at each iteration is high,I think memory management is enabled.
[image]
[image]

lynnsky, 2016-03-30:
In addition to the time is too long, the result is not good.
[image]
and matlab` ShowLogs `output:
-- getPrecisionRecall ---
Loading GroundTruth 'E:\vswork\rtabmap\build\tools\ConsoleApp\UdeS_1Hz.png' ...

GT_total_positives =

        2628

False positive! id=2978 with old=2831 (p=0.10797)
False positive! id=5375 with old=5201 (p=0.097505)
False positive! id=507 with old=399 (p=0.097035)
False positive! id=5376 with old=5201 (p=0.097008)
False positive! id=3851 with old=415 (p=0.0962)
False positive! id=3850 with old=415 (p=0.096099)
False positive! id=5377 with old=5201 (p=0.095318)
False positive! id=3501 with old=3378 (p=0.094431)
False positive! id=3849 with old=415 (p=0.09439)
False positive! id=4157 with old=507 (p=0.093311)
False positive! id=65 with old=4 (p=0.092973)
False positive! id=3852 with old=415 (p=0.092477)
False positive! id=3848 with old=415 (p=0.092203)
False positive! id=5374 with old=5201 (p=0.092158)
False positive! id=5283 with old=5175 (p=0.091495)
False positive! id=5373 with old=5192 (p=0.091009)
False positive! id=3500 with old=3378 (p=0.090536)
False positive! id=2976 with old=2829 (p=0.089209)
False positive! id=2703 with old=2590 (p=0.089098)
False positive! id=2704 with old=2590 (p=0.089098)
False positive! id=2702 with old=2590 (p=0.089096)
False positive! id=2701 with old=2590 (p=0.089094)
False positive! id=2700 with old=2590 (p=0.08909)
False positive! id=2699 with old=2590 (p=0.089081)
False positive! id=2698 with old=2590 (p=0.089065)
False positive! id=2697 with old=2590 (p=0.089034)
False positive! id=755 with old=590 (p=0.088978)
False positive! id=2696 with old=2590 (p=0.088971)
False positive! id=2695 with old=2590 (p=0.088844)
False positive! id=2678 with old=2590 (p=0.088723)
False positive! id=2694 with old=2590 (p=0.088577)
False positive! id=2982 with old=2840 (p=0.088211)
False positive! id=506 with old=399 (p=0.088136)
False positive! id=505 with old=399 (p=0.088073)
False positive! id=2693 with old=2590 (p=0.088005)
False positive! id=64 with old=4 (p=0.08766)
False positive! id=2677 with old=2590 (p=0.087298)
False positive! id=2691 with old=2590 (p=0.087079)
False positive! id=2690 with old=2590 (p=0.086798)
False positive! id=2692 with old=2590 (p=0.086747)
False positive! id=3499 with old=3378 (p=0.086701)
False positive! id=756 with old=590 (p=0.086687)
False positive! id=5372 with old=5192 (p=0.086635)
False positive! id=602 with old=527 (p=0.086583)
False positive! id=1192 with old=986 (p=0.086438)
False positive! id=252 with old=179 (p=0.0856)
False positive! id=3498 with old=3378 (p=0.085588)
False positive! id=2389 with old=1906 (p=0.08553)
False positive! id=2689 with old=2590 (p=0.085482)
False positive! id=2679 with old=2590 (p=0.085294)
False positive! id=3846 with old=3720 (p=0.084956)
False positive! id=2688 with old=2590 (p=0.084733)
False positive! id=3497 with old=3378 (p=0.084646)
False positive! id=3845 with old=3720 (p=0.084575)
False positive! id=2676 with old=2590 (p=0.08443)
False positive! id=3637 with old=3347 (p=0.084387)
False positive! id=754 with old=558 (p=0.084361)
False positive! id=3950 with old=387 (p=0.084354)
False positive! id=5201 with old=5162 (p=0.084115)
False positive! id=1894 with old=1597 (p=0.083921)
[... 437 lines cut ...]
False positive! id=744 with old=557 (p=0.070336)
False positive! id=5066 with old=4442 (p=0.070327)
False positive! id=1643 with old=1446 (p=0.070318)
False positive! id=2548 with old=418 (p=0.070318)
False positive! id=3780 with old=3504 (p=0.070277)
False positive! id=1691 with old=1628 (p=0.070268)
False positive! id=2820 with old=411 (p=0.07024)
False positive! id=1096 with old=425 (p=0.070226)
False positive! id=2818 with old=411 (p=0.070225)
False positive! id=594 with old=464 (p=0.070198)
False positive! id=1604 with old=1524 (p=0.070176)
False positive! id=3913 with old=406 (p=0.070155)
False positive! id=4491 with old=4332 (p=0.070149)
False positive! id=3654 with old=3376 (p=0.070129)
False positive! id=4471 with old=4331 (p=0.070111)
False positive! id=2710 with old=2572 (p=0.070083)
False positive! id=3024 with old=413 (p=0.070082)
False positive! id=3859 with old=463 (p=0.070063)
False positive! id=4755 with old=4192 (p=0.070044)
False positive! id=4600 with old=4329 (p=0.070016)
False positive! id=582 with old=520 (p=0.070013)
False positive! id=877 with old=731 (p=0.070012)
Recall max (Precision=100%) = 0
Recall max accepted (Precision=100%) = 0
ignored = 47

matlabbe, 2016-03-30:
What kind of computer you have? It seems that just SURF extraction take around 600/700 ms, thus triggering memory management even with an empty WM. The benchmark was done on a MacbookPro 2010 with i7 (Arrandale). As you can see on Table III of the [referred paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/b/bc/TRO2013.pdf), recall performance will vary depending on the maximum Working Memory size that can be processed online. On that laptop with 700 ms time threshold, the maximum size of WM is 221 locations. On your side the maximum size of WM seems stuck around 20 locations and processing time is already over 1 sec.

lynnsky, 2016-03-30:
My computer is Thinkpad T450 with i7-5500U，256G SSD.So I should improve time  threshold？

lynnsky, 2016-03-30:
Is it related to use debug mode rather than release mode？

matlabbe, 2016-03-30:
Try to build the code in Release mode instead of Debug, this generally increase a lot the performance, particularly on Windows.

lynnsky, 2016-03-31:
Thanks for your advice.I build the code in Release mode.Then I got a results as following:
[image]
--- getPrecisionRecall ---
Loading GroundTruth 'E:\vswork\rtabmaprelease\build\tools\ConsoleApp\UdeS_1Hz.png' ...

GT_total_positives =

        2628

False positive! id=3927 with old=1194 (p=0.13997)
False positive accepted! id=3927 with old=1194 (p=0.13997)
False positive! id=579 with old=309 (p=0.12179)
False positive accepted! id=579 with old=309 (p=0.12179)
False positive! id=1010 with old=310 (p=0.098755)
False positive! id=3928 with old=1184 (p=0.09676)
False positive! id=1009 with old=310 (p=0.095884)
False positive! id=1011 with old=310 (p=0.094518)
False positive! id=65 with old=4 (p=0.092973)
False positive! id=1012 with old=310 (p=0.092869)
False positive! id=2622 with old=90 (p=0.09209)
False positive! id=1008 with old=310 (p=0.090816)
False positive! id=3499 with old=743 (p=0.090761)
False positive! id=64 with old=4 (p=0.08766)
False positive! id=2778 with old=2726 (p=0.086346)
False positive! id=3498 with old=743 (p=0.083566)
False positive! id=85 with old=33 (p=0.083521)
False positive! id=511 with old=399 (p=0.082917)
False positive! id=1007 with old=310 (p=0.081828)
False positive! id=197 with old=15 (p=0.080056)
False positive! id=3895 with old=1178 (p=0.079923)
False positive! id=72 with old=21 (p=0.079816)
False positive! id=4193 with old=1509 (p=0.078492)
False positive! id=157 with old=42 (p=0.077738)
False positive! id=63 with old=4 (p=0.076487)
False positive! id=5354 with old=5192 (p=0.075888)
False positive! id=196 with old=15 (p=0.075511)
False positive! id=181 with old=139 (p=0.074966)
False positive! id=73 with old=24 (p=0.07481)
False positive! id=2777 with old=2726 (p=0.07398)
False positive! id=2623 with old=91 (p=0.073908)
False positive! id=3932 with old=1220 (p=0.073788)
False positive! id=3896 with old=1178 (p=0.073484)
False positive! id=4929 with old=2339 (p=0.072982)
False positive! id=79 with old=33 (p=0.072933)
False positive! id=183 with old=138 (p=0.072771)
False positive! id=201 with old=15 (p=0.072714)
False positive! id=3852 with old=1102 (p=0.072387)
False positive! id=3897 with old=1178 (p=0.072042)
False positive! id=3851 with old=1102 (p=0.071948)
False positive! id=3985 with old=1253 (p=0.07177)
False positive! id=3850 with old=1102 (p=0.071707)
False positive! id=172 with old=21 (p=0.071564)
False positive! id=186 with old=72 (p=0.071511)
False positive! id=3849 with old=1102 (p=0.07139)
False positive! id=202 with old=15 (p=0.071375)
False positive! id=184 with old=85 (p=0.071159)
False positive! id=3481 with old=691 (p=0.071008)
False positive! id=3848 with old=1102 (p=0.070946)
False positive! id=3847 with old=1102 (p=0.070281)
False positive! id=3497 with old=809 (p=0.070265)
False positive! id=185 with old=138 (p=0.070222)
False positive! id=178 with old=68 (p=0.070111)
Recall max (Precision=100%) = 45.0913% (p=0.14045), accepted=1185
Recall max accepted (Precision=100%) = 45.0913% (p=0.14045), accepted=1185
ignored = 74

lynnsky, 2016-03-31:
Can I ask a question?what can I do with this file ` rtabmapconsole.db`?

matlabbe, 2016-03-31:
Your results are now more similar to the paper. If you were only interested on statistics of loop closure detection, you can delete it. The database contains the graph of the created map (e.g., loop closure links). You can browse images and features saved in it with `rtabmap-databaseViewer.exe`.

lynnsky, 2016-04-01:
It's is cool.Thanks for your project.

lynnsky, 2016-04-01:
It's cool.Thanks for your project.

closed 2016-04-14 by lynnsky