introlab/rtabmap #71: librtabmap_gui.so links to wrong library on make install

OAkyildiz, 2016-04-05:
Hello,

I am  trying to link rtabmap against Qt5 and I compiled VTK and PCL with Qt5 as mentioned #6.

After I 'make install' to my workspace's devel folder, when I run rtabmap , I get a segfault. 

```
$ valgrind ..

I tried both [master](https://github.com/introlab/rtabmap/) and [indigo-devel](https://github.com/introlab/rtabmap/tree/indigo-devel) branches. I tried modifying `CMakeLists.txt` files:

For example,
QT_LIBRARIES

matlabbe, 2016-04-06:
Did you try to clear the build directory of rtabmap before rebuilding it with Qt5? You can also `make` rtabmap with `VERBOSE=TRUE`:
```bash
$ cd rtabmap/build
$ make VERBOSE=TRUE
```

Then look for libraries linked by rtabmap_gui:
```bash
...
Linking CXX shared library ../../../bin/librtabmap_gui.so
cd /home/mathieu/workspace/rtabmap/build/guilib/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/rtabmap_gui.dir/link.txt --verbose=TRUE
/usr/bin/c++  -fPIC  -fmessage-length=0  -Wno-deprecated -O3 -DNDEBUG    -shared -Wl,-soname,librtabmap_gui.so.0.10 -o ../../../bin/librtabmap_gui.so.0.10.10 CMakeFiles/rtabmap_gui.dir/MainWindow.cpp.o CMakeFiles/rtabmap_gui.dir/PreferencesDialog.cpp.o CMakeFiles/rtabmap_gui.dir/KeypointItem.cpp.o CMakeFiles/rtabmap_gui.dir/ImageView.cpp.o CMakeFiles/rtabmap_gui.dir/PdfPlot.cpp.o CMakeFiles/rtabmap_gui.dir/StatsToolBox.cpp.o CMakeFiles/rtabmap_gui.dir/ProgressDialog.cpp.o CMakeFiles/rtabmap_gui.dir/AboutDialog.cpp.o CMakeFiles/rtabmap_gui.dir/ConsoleWidget.cpp.o CMakeFiles/rtabmap_gui.dir/DatabaseViewer.cpp.o CMakeFiles/rtabmap_gui.dir/utilite/UPlot.cpp.o CMakeFiles/rtabmap_gui.dir/CloudViewer.cpp.o CMakeFiles/rtabmap_gui.dir/OdometryViewer.cpp.o CMakeFiles/rtabmap_gui.dir/LoopClosureViewer.cpp.o CMakeFiles/rtabmap_gui.dir/DataRecorder.cpp.o CMakeFiles/rtabmap_gui.dir/CameraViewer.cpp.o CMakeFiles/rtabmap_gui.dir/CalibrationDialog.cpp.o CMakeFiles/rtabmap_gui.dir/ExportDialog.cpp.o CMakeFiles/rtabmap_gui.dir/PostProcessingDialog.cpp.o CMakeFiles/rtabmap_gui.dir/ExportCloudsDialog.cpp.o CMakeFiles/rtabmap_gui.dir/MapVisibilityWidget.cpp.o CMakeFiles/rtabmap_gui.dir/GraphViewer.cpp.o CMakeFiles/rtabmap_gui.dir/CreateSimpleCalibrationDialog.cpp.o CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_MainWindow.cxx.o CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_PreferencesDialog.cxx.o CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_DatabaseViewer.cxx.o CMakeFiles/rtabmap_gui.dir/moc_AboutDialog.cxx.o CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_ConsoleWidget.cxx.o CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_ImageView.cxx.o CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_PdfPlot.cxx.o CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_StatsToolBox.cxx.o CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_ProgressDialog.cxx.o CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/utilite/moc_UPlot.cxx.o CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/utilite/moc_UImageView.cxx.o CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_CloudViewer.cxx.o CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_OdometryViewer.cxx.o CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_LoopClosureViewer.cxx.o CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_DataRecorder.cxx.o CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_CameraViewer.cxx.o CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_CalibrationDialog.cxx.o CMakeFiles/rtabmap_gui.dir/moc_ExportDialog.cxx.o CMakeFiles/rtabmap_gui.dir/moc_PostProcessingDialog.cxx.o CMakeFiles/rtabmap_gui.dir/moc_ExportCloudsDialog.cxx.o CMakeFiles/rtabmap_gui.dir/moc_MapVisibilityWidget.cxx.o CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_GraphViewer.cxx.o CMakeFiles/rtabmap_gui.dir/moc_CreateSimpleCalibrationDialog.cxx.o CMakeFiles/rtabmap_gui.dir/qrc_GuiLib.cxx.o ../../../bin/librtabmap_core.so.0.10.10 ../../../bin/librtabmap_utilite.so.0.10.10 -lQtSvg -lQtGui -lQtXml -lQtCore /usr/lib/x86_64-linux-gnu/libopencv_videostab.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_video.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_ts.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_superres.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_stitching.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_photo.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_ocl.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_objdetect.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_ml.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_legacy.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_gpu.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_flann.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_features2d.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_contrib.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.2.4.8 -lboost_system -lboost_filesystem -lboost_thread -lboost_date_time -lboost_iostreams -lboost_serialization -lpthread -lpcl_common -lpcl_octree -lOpenNI /usr/lib/libvtkCommon.so.5.8.0 /usr/lib/libvtkRendering.so.5.8.0 /usr/lib/libvtkHybrid.so.5.8.0 /usr/lib/libvtkCharts.so.5.8.0 -lpcl_io -Wl,-Bstatic -lflann_cpp_s -Wl,-Bdynamic -lpcl_kdtree -lpcl_search -lpcl_sample_consensus -lpcl_filters -lpcl_features -lpcl_keypoints -lpcl_segmentation -lpcl_visualization -lpcl_outofcore -lpcl_registration -lpcl_recognition -lqhull -lpcl_surface -lpcl_people -lpcl_tracking -lpcl_apps -lboost_system -lboost_filesystem -lboost_thread -lboost_date_time -lboost_iostreams -lboost_serialization -lpthread -lqhull -lOpenNI -Wl,-Bstatic -lflann_cpp_s -Wl,-Bdynamic /usr/lib/libvtkCommon.so.5.8.0 /usr/lib/libvtkRendering.so.5.8.0 /usr/lib/libvtkHybrid.so.5.8.0 /usr/lib/libvtkCharts.so.5.8.0 -lQVTK /usr/lib/x86_64-linux-gnu/libopencv_photo.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_legacy.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_video.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_objdetect.so.2.4.8 /usr/lib/x86_64- [... cut ...]
	libQtSql.so.4 => /usr/lib/x86_64-linux-gnu/libQtSql.so.4 (0x00007f89d687b000)
	libvtkQtChart.so.5.8 => /usr/lib/libvtkQtChart.so.5.8 (0x00007f89d6308000)
```
```bash
$ ldd /usr/lib/libQVTK.so.5.8 | grep Qt
	libQtGui.so.4 => /usr/lib/x86_64-linux-gnu/libQtGui.so.4 (0x00007f9e21aed000)
	libQtSql.so.4 => /usr/lib/x86_64-linux-gnu/libQtSql.so.4 (0x00007f9e218ae000)
	libQtCore.so.4 => /usr/lib/x86_64-linux-gnu/libQtCore.so.4 (0x00007f9e213c9000)
	libvtkQtChart.so.5.8 => /usr/lib/libvtkQtChart.so.5.8 (0x00007f9e1f81e000)
```

OAkyildiz, 2016-04-08:
I cleaned the build directory and even tried with a fresh clone too, and when I call `ldd` for the .so's in the building directory, I saw they were linked against Qt5, it also run without segfault.

closed 2016-11-09 by matlabbe