introlab/rtabmap #72: Magic matrix during exporting to Bundler output

egoist-sx, 2016-04-06:
I saw there is a magic matrix multiplied before export rotation as bundler format.

In [https://github.com/introlab/rtabmap/blob/d489cd48e969b38c8b004498d400a0b47a9bf637/guilib/src/MainWindow.cpp](this) line 5246. Why this matrix is needed?

matlabbe, 2016-04-06:
The magic transform is to transform camera poses in OpenGL coordinate system (x right, y up, z backward), which is the one used by MeshLab when importing bundler project. RTAB-Map uses ROS coordinate system (x forward, y left, z up). See this example (from this [tutorial](https://github.com/introlab/rtabmap/wiki/Export-Raster-Layers-to-MeshLab)):

[image]

egoist-sx, 2016-04-10:
@matlabbe Thanks a lot!

closed 2016-04-10 by matlabbe