introlab/rtabmap #74: Tango compilation

r0nn13, 2016-04-21:
Hi Mathieu

How is your environment set up to compile the Android code?

I have included the tango_client_api.h etc. but I'm getting an "android" not found error.

Thanks
Ronnie

matlabbe, 2016-04-24:
Hi,

I didn't take the time yet to write down all the build instructions for Tango. When you are able to cross-compile all dependencies, you may use these instructions for rtabmap:
```bash
$ export ANDROID_NDK="Path to android ndk"

$ export ANDROID_NATIVE_API_LEVEL=android-17

$ cd rtabmap/build

$ cmake -DCMAKE_TOOLCHAIN_FILE=../cmake_modules/android.toolchain.cmake -DBUILD_SHARED_LIBS=OFF -DBUILD_TOOLS=OFF -DCMAKE_BUILD_TYPE=Release ..

$ make

$ adb install -r app/android/bin/RTABMap-debug.apk

```

The longer/harder part is to get flann/eigen/boost/pcl/opencv/gtsam correctly built (use static libraries) for android.

cheers,

mushasun, 2016-05-10:
Hi Mathieu,

Do you use pcl-superbuild(pcl 1.6) for Tango ?

Thanks
Musha

matlabbe, 2016-05-10:
I used PCL 1.8 directly.

ilovepanda4, 2016-05-21:
Can you provide exact instructions on building apk for the tango? (the dependencies and instructions)
after following #74 for tango I do not see the apk file.
Thanks in advance

mushasun, 2016-05-23:
I just build the apk and can work now. My question: Is the gtsam must build ? I do quick test and it looks workable without gtsam .

mushasun, 2016-05-23:
I built it on Windows. Do you use Windows as well ?

>>>>> Original Message <<<<<
From: introlab/rtabmap
To: "introlab/rtabmap"
Cc: "Comment";"mushasun"
Sent Date: Sun,22 May 2016 03:21:10 Asia/Taipei
Subject: Re: [introlab/rtabmap] Tango compilation (#74)
 

Can you provide exact instructions on building apk for the tango? (the dependencies and instructions)
 after following #74 for tango I do not see the apk file.
 Thanks in advance
&mdash;

ilovepanda4, 2016-05-24:
I am using Ubuntu 14.04

matlabbe, 2016-05-25:
@mushasun [GTSAM](https://collab.cc.gatech.edu/borg/gtsam?destination=node%2F299) is optional, [TORO](https://www.openslam.org/toro.html) will be used instead. [GTSAM](https://collab.cc.gatech.edu/borg/gtsam?destination=node%2F299) would give slightly better graph optimizations though. Note also that [g2o](https://github.com/RainerKuemmerle/g2o) is required for "Post-Processing" -> "Bundle Adjustment".

@ilovepanda4 The instructions above were done on Mac OS X, but it should be the same for Ubuntu. I don't remember every steps for each dependency, but as mentioned above, you may use the same cmake instructions (for all dependencies using cmake):
```bash
$ cd "my_dependency_lib"
$ mkdir build
$ cd build
$ cmake -DCMAKE_TOOLCHAIN_FILE=~/rtabmap/cmake_modules/android.toolchain.cmake -DBUILD_SHARED_LIBS=OFF -DBUILD_TOOLS=OFF -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=~/android_install ..
$ make
$ make install
```

Eigen is headers only, so they can just be installed directly in `~/android_install`. For Boost, I did manually a cmake wrapper for convenience to only build the libs I needed (or that PCL needed):
[boost_cmake.zip](https://github.com/introlab/rtabmap/files/282233/boost_cmake.zip)

cheers

zhngling123, 2016-06-29:
@matlabbe @mushasun 
Hello Guys,

I just follow this script:
`$ export ANDROID_NDK="Path to android ndk"
$ export ANDROID_NATIVE_API_LEVEL=android-17
$ cd rtabmap/build
$ cmake -DCMAKE_TOOLCHAIN_FILE=../cmake_modules/android.toolchain.cmake -DBUILD_SHARED_LIBS=OFF -DBUILD_TOOLS=OFF -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=~/android_install ..`

With cmake version 3.2.2, following error display:
`CMake Error at /usr/share/OpenCV/OpenCVConfig.cmake:45 (include):
  include could not find load file:
    /usr/share/OpenCV/OpenCVModules_armeabi_v7a.cmake
Call Stack (most recent call first):
  CMakeLists.txt:141 (FIND_PACKAGE)
-- Could NOT find eigen (missing:  EIGEN_INCLUDE_DIRS) 
CMake Error at /usr/share/pcl-1.7/PCLConfig.cmake:42 (message):
  common is required but eigen was not found
Call Stack (most recent call first):
  /usr/share/pcl-1.7/PCLConfig.cmake:454 (pcl_report_not_found)
  /usr/share/pcl-1.7/PCLConfig.cmake:603 (find_external_library)
  CMakeLists.txt:142 (FIND_PACKAGE)
-- Configuring incomplete, errors occurred!`

While with cmake version 2.
`$ export ANDROID_NDK="Path to android ndk"
$ export ANDROID_NATIVE_API_LEVEL=android-17
$ cd rtabmap/build
$ cmake -DCMAKE_TOOLCHAIN_FILE=../cmake_modules/android.toolchain.cmake -DBUILD_SHARED_LIBS=OFF -DBUILD_TOOLS=OFF -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=~/android_install ..
$ make`
cmake and make run well. But no apk generated. 
Any comment or direction is very appreciate.

matlabbe, 2016-06-29:
PCL and its dependencies should be cross-compiled. You cannot use standard binaries. For OpenCV, you can use the one already built in [NVIDIA Tegra android sdk](https://developer.nvidia.com/codeworks-android). You may have to execute cmake with `-DOpenCV_DIR="path/to/OpenCVConfig.cmake/directory"`.

zhngling123, 2016-06-29:
Thanks for quick guide on OpenCV and cross-compiled. How about PCL one your Mac/Linux?

matlabbe, 2016-06-29:
PCL can be built in a similar way than RTAB-Map using `android.toolchain.cmake`.

zhngling123, 2016-06-29:
I saw sudo apt-get -y install libpcl-1.7-all libfreenect-dev in .travis.It's use standard binaries. I use Mac 10.11 and brew. It's very helpful to dig into PCL installation/compile from you. Thanks in advance.
[Official doc for PCL with Brew](http://www.pointclouds.org/documentation/tutorials/installing_homebrew.php)

hzqkit, 2017-01-12:
Hi, could someone help me how to compile RTAB-AndroidApp on Windows? I have been struggled for a couple of days but it cannot still find PCL. So I thought maybe I should cross compile "PCL for Android" at first ?

I have used "PCL Superbild" and after cross compiling I have got the Makefile. But then I got some problems with the make. As generator I have chosen Mingw32 make, but it seems that it cannot find the Boost_Root variable even after I set the path (I built the Boost lib with VS 2013). So I have been stuck on this step for a long time. Maybe I should use another generator (e.g cygwin or nmake)?

Really hoping that someone could give me some advises and sorry about my poor English. Thanks in advance!

liyou331, 2017-01-12:
Hi guys, thanks for the instructions!
Anybody know how to deal with the tango C libraries? I set -DTango_INCLUDE_DIR=...../path to head files  \
-DTango_LIBRARY=.....path..../libtango_client_api.so \
-DTango_support_LIBRARY=....path......../libtango_support_api.so

The apk can be successfully generated by following Mathieu's instructions. However,  when I click the app, it stopped. The error information is 

 java.lang.UnsatisfiedLinkError: dlopen failed: library "libtango_support_api.so" not found
            at java.lang.Runtime.loadLibrary(Runtime.java:372)
            at java.lang.System.loadLibrary(System.java:1076)
            at com.introlab.rtabmap.RTABMapLib.<clinit>(RTABMapLib.java:20)
            at com.introlab.rtabmap.RTABMapActivity.onCreate(RTABMapActivity.java:199)
            at android.app.Activity.performCreate(Activity.java:6285)
            at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1108)
            at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2370)
            at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2477)
            at android.app.ActivityThread.access$900(ActivityThread.java:150)
            at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1344)
            at android.os.Handler.dispatchMessage(Handler.java:102)
            at android.os.Looper.loop(Looper.java:148)
            at android.app.ActivityThread.main(ActivityThread.java:5418)
            at java.lang.reflect.Method.invoke(Native Method)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)

matlabbe, 2017-01-12:
Solution here for the missing *.so libraries: http://official-rtab-map-forum.67519.x6.nabble.com/RTABMap-for-Android-td2528.html#a2558

matlabbe, 2017-03-23:
I added some Dockerfile to project for convenience. In particular to close this issue, see how docker images are built for tango:
1. All android dependencies (Image: [introlab3it/rtabmap:android-deps](https://hub.docker.com/r/introlab3it/rtabmap/tags/)): 
    * https://github.com/introlab/rtabmap/tree/master/docker/xenial/android
2. RTAB-Map Tango (Image: [introlab3it/rtabmap:tango](https://hub.docker.com/r/introlab3it/rtabmap/tags/)): 
    * https://github.com/introlab/rtabmap/tree/master/docker/xenial/android/tango

In particular look at the file [install_deps.bash](https://github.com/introlab/rtabmap/blob/master/docker/xenial/android/install_deps.bash) and then [build.bash](https://github.com/introlab/rtabmap/blob/master/docker/xenial/android/tango/build.bash).

cheers,
Mathieu

closed 2017-03-23 by matlabbe