introlab/rtabmap #75: when i compile this code , i always occur this problem.

diophantine86, 2016-04-28:
../../../bin/librtabmap_core.so.0.11.4: undefined reference to `pcl::OrganizedFastMesh<pcl::PointXYZRGBNormal>::performReconstruction(pcl::PolygonMesh&)'
../../../bin/librtabmap_core.so.0.11.4: undefined reference to `pcl::OrganizedFastMesh<pcl::PointXYZRGBNormal>::performReconstruction(std::vector<pcl::Vertices, std::allocator<pcl::Vertices> >&)'
collect2: error: ld returned 1 exit status
make[2]: *** [../bin/rtabmap] Error 1
make[1]: *** [app/src/CMakeFiles/rtabmap.dir/all] Error 2
make: *** [all] Error 2

matlabbe, 2016-04-28:
Hi,
Do you have PCL 1.8 installed? if so, there is a workaround here: http://official-rtab-map-forum.67519.x6.nabble.com/Error-building-Rtabmap-on-RPi2-on-99-tp1234p1237.html

diophantine86, 2016-04-29:
tks.I fixed this error after your note.

closed 2016-04-29 by matlabbe