introlab/rtabmap #78: input to vocabularyComparison

lynnsky, 2016-05-05:
How can I get a dictionary file as the input to vocabularyComparison.exe?
By the way,where can I see information of UDEBUG("Time to create kdTreeIndex4 = %f s", timer.ticks());

matlabbe, 2016-05-05:
The format is the same as the file `DumpMemoryWordDesc.txt` when you dump the memory (Edit->Advanced->Dump the memory). To see the log messages, you should start the application from the console.

For now, `vocabularyComparison.exe` will only work with dictionary using float descriptors (SIFT or SURF). You may use 0.10.10 as I just fixed an [exporting crash bug in 0.11](https://github.com/introlab/rtabmap/issues/79).

For example, using the exported dictionary [DumpMemoryWordDesc.txt](https://github.com/introlab/rtabmap/files/250690/DumpMemoryWordDesc.txt):
```bash
$ vocabularyComparison.exe DumpMemoryWordDesc.txt
[DEBUG] (2016-05-05 11:32:35.667) main.cpp:79::main() Loading the dictionary from "DumpMemoryWordDesc.txt"
[DEBUG] (2016-05-05 11:32:51.260) main.cpp:144::main() Time loading dictionary = 15.593227s, dimension=64
[DEBUG] (2016-05-05 11:32:51.261) main.cpp:157::main() Creating data structures...
[DEBUG] (2016-05-05 11:32:51.262) main.cpp:185::main() descriptors.size()=2792, objectDescriptorsSize=400, copying data = 0.001221 s
[DEBUG] (2016-05-05 11:32:51.262) main.cpp:187::main() Creating indexes...
[DEBUG] (2016-05-05 11:32:51.265) main.cpp:189::main() Time to create linearIndex = 0.002998 s
[DEBUG] (2016-05-05 11:32:51.269) main.cpp:192::main() Time to create kdTreeIndex1 = 0.004752 s
[DEBUG] (2016-05-05 11:32:51.283) main.cpp:195::main() Time to create kdTreeIndex4 = 0.013089 s
[DEBUG] (2016-05-05 11:32:51.396) main.cpp:198::main() Time to create kMeansIndex = 0.113059 s
[DEBUG] (2016-05-05 11:32:51.501) main.cpp:201::main() Time to create compositeIndex = 0.105632 s
[DEBUG] (2016-05-05 11:32:51.501) main.cpp:207::main() Search indexes...
[DEBUG] (2016-05-05 11:32:51.563) main.cpp:215::main() Time to search linearIndex = 0.061898 s
[DEBUG] (2016-05-05 11:32:51.567) main.cpp:230::main() Time to search kdTreeIndex1 = 0.003748 s (size=400, dist error(k1,k2)=(0.000249,0.000333))
[DEBUG] (2016-05-05 11:32:51.573) main.cpp:245::main() Time to search kdTreeIndex4 = 0.005760 s (size=400, dist error(k1,k2)=(0.000236,0.000270))
[DEBUG] (2016-05-05 11:32:51.578) main.cpp:260::main() Time to search kMeansIndex = 0.005466 s (size=400, dist error(k1,k2)=(0.000089,0.000082))
[DEBUG] (2016-05-05 11:32:51.584) main.cpp:275::main() Time to search compositeIndex = 0.005525 s (size=400, dist error(k1,k2)=(0.000071,0.000080))
```

Dictionary Format is one line header with descriptor dimension (e.g., 64), then "`WordID d[0] d[1] ... d[63]`":
```
WordID Descriptors...64
1 0.000189 -0.003662 0.003686 0.005446 -0.013637 -0.015411 0.015540 0.022084 0.058712 -0.006057 0.062436 0.006169 0.000274 -0.000117 0.000280 0.000262 0.002735 -0.008792 0.003191 0.015704 -0.052450 -0.072957 0.061925 0.097124 0.412357 -0.064487 0.441145 0.065206 0.001510 -0.001717 0.001510 0.001921 0.003088 0.046359 0.004489 0.069307 0.017021 0.345333 0.030938 0.553175 0.203921 0.147982 0.205702 0.216885 0.002734 -0.005917 0.002734 0.005917 0.000149 -0.013798 0.000776 0.013902 0.021332 -0.046979 0.021799 0.046979 -0.017301 -0.041189 0.018357 0.041234 -0.000566 -0.014872 0.001689 0.015059 
2 -0.000646 0.006341 0.000656 0.006394 -0.006894 0.054988 0.006894 0.055223 -0.003616 0.066334 0.004897 0.066794 0.000544 0.008586 0.001192 0.008586 0.003731 -0.033549 0.006944 0.052717 0.008945 -0.327264 0.069178 0.401798 0.179218 -0.201747 0.186884 0.225109 -0.001410 0.003245 0.001410 0.003353 -0.002580 0.003034 0.005642 0.011606 -0.182668 0.006761 0.187635 0.050461 0.475801 0.082491 0.477092 0.083148 -0.001959 0.000471 0.001959 0.001079 0.001589 0.003220 0.002318 0.003360 -0.018043 0.009317 0.046160 0.015602 0.057782 0.010420 0.058523 0.010505 -0.000321 -0.000110 0.000321 0.000217 
...
```

cheers

lynnsky, 2016-05-06:
Thank your for your help.When I dump the memory ,RTABmap.exe  stop working,did't generate file like `DumpMemoryWordDesc.txt `.I use `settingsAppearanceDemo.ini `with` L6I `dataset.The dictionary size (about1.4W)is smaller than paper 5.2W.

matlabbe, 2016-05-06:
Are you compiling the latest code from source? (as you have vocabularyComparison.exe). Make sure you updated the code to have this fix https://github.com/introlab/rtabmap/issues/79. If it is still crashing after recompiling, let me know.

If I run the [0.10.10 Windows binaries](https://github.com/introlab/rtabmap/releases/tag/0.10.10), I can export and the vocabulary is around 1.5K words. The difference with the paper is the parameters. You should use parameters specified on the [Benchmark](https://github.com/introlab/rtabmap/wiki/Benchmark#lip6-indoor) page for each dataset if you want to compare with the paper. Well, the files are saved directly in `My Documents` (there is a little bug with the file name, which is also fixed in the commit above).

[image]

lynnsky, 2016-05-07:
I use the old version 0.11.2.You mean in version `0.11.2` and` 0.11.4`,vocabularyComparison.exe don't work with dictionary using binary descriptors .` 0.11.2`'s  Dump memory is broken and `0.11.4 ` fixed the exporting crash bug?
Now I compile the latest code from source.The problem is I can't generate ` rtabmap_core.lib`  success.Compiling  project  rtabmap_core don't have errors,just generate failed.
[image]
How can I solve this problem?

lynnsky, 2016-05-08:
I use the old version 0.11.2.You mean in version 0.11.2 and0.11.4,vocabularyComparison.exe can't work with dictionary using binary descriptors ,but 0.10.10 can?
Now I compile the latest code from source.I find in file Memory.cpp and Odometry.cpp
The follow statements have function overload problem,leading to generate` rtabmap_core.lib `failed.
double `log2value` = log(`preDecimation`)/log(2);
Now I can generate `DumpMemoryWordDesc.txt `successfully.
Thanks for your help.

lynnsky, 2016-05-08:
I use  version 0.11.2.Now I compile the latest code from source.I find in file `Memory.cpp` and `Odometry.cpp`
The follow statements have function overload problem,leading to generate` rtabmap_core.lib` failed.
double `log2value `= log(`preDecimation`)/log(2);
Now I can generate `DumpMemoryWordDesc.txt` successfully.
Thanks for your help.

closed 2016-05-08 by matlabbe