introlab/rtabmap #82: Tango app crashes with memory used over 1300 MB

matlabbe, 2016-05-18:
From comment on Tango app page:

- when the map get too large (about 1300 MB in ram :debug show) Rtab apk had been shooting down .

- even medium map can't export to obj. 3:Is there any way to post processing bundle adjustment work better or mesh stick together properly? Thanks for your consideration!

matlabbe, 2016-05-18:
Because of slow SD-Card access, the database is kept in RAM instead of hard drive like on the Desktop version. Indeed, the app may crash if the RAM is full.

For post-processing with more options and faster exporting to *.obj, save the database and open it in Desktop version:

Open Database
Edit -> Download all clouds (update cache)
File -> Export 3D clouds...
The Bundle Adjustment option is very new, so there are probably some improvements to code that can be done.

matlabbe, 2016-05-18:
I reproduced the problem on Tango:

1. Open RTAB-Map's app on Google's Project Tango (tablet used here).
2. Click on `Rendering Options...`, then on `Debug` to show debug overlay.
2. Map until `Memory used` in Debug view is around 1400MB (~600 nodes in Working Memory, total length of the trajectory is ~400 meters). A message "Tango disconnected!" will appear. The mapping is not working anymore. However I was able to save the database, hopefully! The resulting database is 103 MB and can be opened in Desktop version:
[image]
 
Looking at the idle memory used by the device, there is normally 2.6GB RAM free:
[image]

I'm not sure about if android apps have access to whole free memory or not. I'm using [Debug. getNativeHeapAllocatedSize()](https://github.com/introlab/rtabmap/blob/dde0e2611026dd6d764fc35f98955475086ea3ae/app/android/src/com/introlab/rtabmap/RTABMapActivity.java#L352) to get the memory usage. A more in deep analyse may be done to see how much memory is available for the app. We could warn the user that RAM limit is almost reached.

On other side, optimizations could be done to reduce the RAM memory used. For example, all point clouds are still kept in cache even if some nodes are transferred in LTM. Caching was used to avoid reconstructing the point clouds when bringing back to WM, thus saving computation time. Being able to change the memory threshold could limit the number of nodes in WM.

greymfm, 2017-02-10:
I'm trying to localize a robotic mower in an larger outdoor environment (30x30m) and experiencing the same (memory overflow). Actually, I don't need an accurate map (30cm grid size would be fine too). Do you think you could add some options for the next release that allows us to reduce memory? It will be also a great localization app, even with lower resolution :-)

matlabbe, 2017-02-10:
Are you referring to Tango App? or rtabmap_ros?

greymfm, 2017-02-10:
Tango App

matlabbe, 2017-02-10:
You can set already a voxel size of 20 cm with the current app when exporting the point cloud. I can indeed add more options if needed (e.g., 0.3, 0.4, 0.5 m). See "Voxel Size" in Export... options:
[image]

For the online rendering while you are scanning, you can reduce the memory used by showing only the point cloud and reducing the point cloud density:
[image]