introlab/rtabmap #83: compilation error of rtabmap_gui

nguyenduchoangha, 2016-05-23:
ubuntu@tegra-ubuntu:~/slam/rtabmap/build$ cmake -DCMAKE_INSTALL_PREFIX=/home/ubuntu/ros_ws/devel ..
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Found OpenMP
-- Found OpenCV: /usr/include/opencv;/usr/include
-- Found PCL: /usr/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/vtk-5.8
-- Found ZLIB: /usr/include
-- Found Freenect: /opt/ros/indigo/include/libfreenect
-- Found OpenNI2: /usr/include/openni2
-- Could NOT find CSPARSE (missing:  CSPARSE_INCLUDE_DIR CSPARSE_LIBRARY) 
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   Version : 0.11.5
--   CMAKE_INSTALL_PREFIX = /home/ubuntu/ros_ws/devel
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib/arm-linux-gnueabihf
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -Wno-deprecated -fopenmp
--   With OpenCV 2 nonfree module (SIFT/SURF) = NO (not found, License: BSD)
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = NO (libfreenect2 not found)
--   With dc1394               = NO (dc1394 not found)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = NO (g2o not found)
--   With GTSAM                = NO (GTSAM not found)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = NO (cvsba not found)
--   With Qt4                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/slam/rtabmap/build
ubuntu@tegra-ubuntu:~/slam/rtabmap/build$ make
Scanning dependencies of target rtabmap_utilite
[  0%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UEventsManager.cpp.o
[  1%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UEventsHandler.cpp.o
[  1%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UEventsSender.cpp.o
[  2%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UFile.cpp.o
[  2%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UDirectory.cpp.o
[  3%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UConversion.cpp.o
[  3%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/ULogger.cpp.o
[  4%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UThread.cpp.o
[  4%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UTimer.cpp.o
[  5%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UProcessInfo.cpp.o
[  5%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UVariant.cpp.o
Linking CXX shared library ../../../bin/librtabmap_utilite.so
[  5%] Built target rtabmap_utilite
Scanning dependencies of target res_tool
[  6%] Building CXX object utilite/resource_generator/CMakeFiles/res_tool.dir/main.cpp.o
Linking CXX executable ../../../bin/rtabmap-res_tool
[  6%] Built target res_tool
[  6%] [Creating resources]
 Using namespace=rtabmap
 Using target directory=/home/ubuntu/slam/rtabmap/build/corelib/src
Input file "/home/ubuntu/slam/rtabmap/corelib/src/resources/DatabaseSchema.sql" size = 3800 bytes
Output file "/home/ubuntu/slam/rtabmap/build/corelib/src/DatabaseSchema_sql.h" size = 7815 bytes
Scanning dependencies of target rtabmap_core
[  7%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Rtabmap.cpp.o
[  7%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/RtabmapThread.cpp.o
[  8%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Statistics.cpp.o
[... 90 lines cut ...]
[ 55%] Generating ui_consoleWidget.h
[ 56%] Generating ui_DatabaseViewer.h
[ 56%] Generating ui_loopClosureViewer.h
[ 57%] Generating ui_exportDialog.h
[ 57%] Generating ui_postProcessingDialog.h
[ 58%] Generating ui_exportCloudsDialog.h
[ 58%] Generating ui_exportScansDialog.h
[ 59%] Generating ui_calibrationDialog.h
[ 59%] Generating ui_createSimpleCalibrationDialog.h
Scanning dependencies of target rtabmap_gui
[ 59%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/MainWindow.cpp.o
In file included from /usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:52:0,
                 from /usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:39,
                 from /home/ubuntu/slam/rtabmap/guilib/src/MainWindow.cpp:104:
/usr/include/pcl-1.7/pcl/visualization/area_picking_event.h: In member function 'bool pcl::visualization::AreaPickingEvent::getPointsIndices(std::vector<int>&) const':
/usr/include/pcl-1.7/pcl/visualization/area_picking_event.h:64:30: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
           if (nb_points_ == -1)
                              ^
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [guilib/src/CMakeFiles/rtabmap_gui.dir/MainWindow.cpp.o] Error 4
make[1]: *** [guilib/src/CMakeFiles/rtabmap_gui.dir/all] Error 2
make: *** [all] Error 2

matlabbe, 2016-05-25:
Hi,

You may run out of memory, try to launch the compilation again or increase your swap space. For recent compilers, some files like MainWindow.cpp may require 1.5 GB RAM or more to compile.

cheers

nguyenduchoangha, 2016-05-26:
Thank you very much! It worked!

closed 2016-05-26 by nguyenduchoangha