introlab/rtabmap #85: rtabmap/corelib/src/VWDictionary.cpp:545:31: error: #if with no expression  #if HAVE_OPENCV_CUDAFEATURES2D

willdzeng, 2016-06-22:
Hey Mathieu,

I try to build rtabmap with OpenCV3, but I got this error while building it.
How can I fix this?

> error: #if with no expression
 #if HAVE_OPENCV_CUDAFEATURES2D

Also Zed updated the it's SDK today, seems like they changed some function input.
How to ask rtabmap to build without building ZED surport? Since I am using ROS and ZED are launched separately .

> In file included from /home/zeng/essential/rtabmap/utilite/include/rtabmap/utilite/UEventsManager.h:27:0,
                 from /home/zeng/essential/rtabmap/corelib/src/CameraStereo.cpp:33:
/home/zeng/essential/rtabmap/corelib/src/CameraStereo.cpp: In constructor ‘rtabmap::CameraStereoZed::CameraStereoZed(int, int, int, int, int, float, const rtabmap::Transform&)’:
/home/zeng/essential/rtabmap/corelib/src/CameraStereo.cpp:771:26: error: ‘FULL’ is not a member of ‘sl::zed’
  UASSERT(sensingMode_ >= sl::zed::FULL && sensingMode_ <=sl::zed::RAW);
                          ^
/home/zeng/essential/rtabmap/utilite/include/rtabmap/utilite/ULogger.h:64:33: note: in definition of macro ‘UASSERT’
 #define UASSERT(condition) if(!(condition)) ULogger::write(ULogger::kFatal, __FILE__, __LINE__, __FUNCTION__, "Condition (%s) not met!", #condition)
                                 ^
/home/zeng/essential/rtabmap/corelib/src/CameraStereo.cpp:771:58: error: ‘RAW’ is not a member of ‘sl::zed’
  UASSERT(sensingMode_ >= sl::zed::FULL && sensingMode_ <=sl::zed::RAW);
                                                          ^
/home/zeng/essential/rtabmap/utilite/include/rtabmap/utilite/ULogger.h:64:33: note: in definition of macro ‘UASSERT’
 #define UASSERT(condition) if(!(condition)) ULogger::write(ULogger::kFatal, __FILE__, __LINE__, __FUNCTION__, "Condition (%s) not met!", #condition)
                                 ^
/home/zeng/essential/rtabmap/corelib/src/CameraStereo.cpp: In constructor ‘rtabmap::CameraStereoZed::CameraStereoZed(const string&, int, int, int, float, const rtabmap::Transform&)’:
/home/zeng/essential/rtabmap/corelib/src/CameraStereo.cpp:796:26: error: ‘FULL’ is not a member of ‘sl::zed’
  UASSERT(sensingMode_ >= sl::zed::FULL && sensingMode_ <=sl::zed::RAW);
                          ^
/home/zeng/essential/rtabmap/utilite/include/rtabmap/utilite/ULogger.h:64:33: note: in definition of macro ‘UASSERT’
 #define UASSERT(condition) if(!(condition)) ULogger::write(ULogger::kFatal, __FILE__, __LINE__, __FUNCTION__, "Condition (%s) not met!", #condition)
                                 ^
/home/zeng/essential/rtabmap/corelib/src/CameraStereo.cpp:796:58: error: ‘RAW’ is not a member of ‘sl::zed’
  UASSERT(sensingMode_ >= sl::zed::FULL && sensingMode_ <=sl::zed::RAW);
                                                          ^
/home/zeng/essential/rtabmap/utilite/include/rtabmap/utilite/ULogger.h:64:33: note: in definition of macro ‘UASSERT’
 #define UASSERT(condition) if(!(condition)) ULogger::write(ULogger::kFatal, __FILE__, __LINE__, __FUNCTION__, "Condition (%s) not met!", #condition)
                                 ^
/home/zeng/essential/rtabmap/corelib/src/CameraStereo.cpp: In member function ‘virtual bool rtabmap::CameraStereoZed::init(const string&, const string&)’:
/home/zeng/essential/rtabmap/corelib/src/CameraStereo.cpp:842:22: error: no matching function for call to ‘sl::zed::Camera::init(sl::zed::MODE, int, bool, bool, bool)’
    true, false, false);
                      ^
/home/zeng/essential/rtabmap/corelib/src/CameraStereo.cpp:842:22: note: candidate is:
In file included from /home/zeng/essential/rtabmap/corelib/src/CameraStereo.cpp:52:0:
/usr/local/zed/include/zed/Camera.hpp:174:21: note: sl::zed::ERRCODE sl::zed::Camera::init(sl::zed::InitParams&)
             ERRCODE init(InitParams &parameters);

Thanks~

willdzeng, 2016-06-22:
I added this in CMakeList.txt
`set(WITH_ZED OFF)`
to cancel the zed support.

Seems like the error is caused by unable to detect Opencv CUDA 2d features
Here is my Cmake print

> -- Found OpenMP
-- Found OpenCV: /usr/local/include;/usr/local/include/opencv
-- Found PCL: /usr/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/openni2;/usr/include/vtk-5.8
-- Found ZLIB: /usr/include
-- Found Freenect: /usr/include
-- Found freenect2: /usr/local/include
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- Found g2o: /opt/ros/jade/include;/usr/include/suitesparse;/usr/include/suitesparse
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/local/include/gtsam/3rdparty/Eigen/
-- Found cvsba: /usr/local/include
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   Version : 0.11.7
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib/x86_64-linux-gnu
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -Wno-deprecated -fopenmp -std=c++11
--   With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = YES (License: Non commercial)
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = YES (License: Apache v2 and/or GPLv2)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = YES (License: BSD)
--   With GTSAM                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = YES (License: GPLv2)
--   With ZED                  = NO (WITH_ZED=OFF)
--   With Qt4                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/zeng/essential/rtabmap/build

>

willdzeng, 2016-06-22:
fixed this issue by change

`#if HAVE_OPENCV_CUDAFEATURES2D`

to

`#ifdef HAVE_OPENCV_CUDAFEATURES2D`

matlabbe, 2016-06-22:
To build without ZED, you can run cmake-gui and uncheck "WITH_ZED" option or do:
```cmake
$ cd rtabmap/build
$ cmake -DWITH_ZED=OFF ..
```

I'll check tomorrow to fix the build with latest ZED sdk.

closed 2016-07-29 by willdzeng