introlab/rtabmap #86: Build with OpenCV3.1: /usr/bin/ld: warning: libopencv_core.so.2.4, needed by /usr/local/lib/libcvsba.so, may conflict with libopencv_core.so.3.1

willdzeng, 2016-06-22:
[100%] Built target databaseViewer
Linking CXX executable ../../../bin/rtabmap-calibration
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by /usr/local/lib/libcvsba.so, may conflict with libopencv_core.so.3.1
[100%] Built target dataRecorder
[100%] Built target rtabmap
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by /usr/local/lib/libcvsba.so, may conflict with libopencv_core.so.3.1
/usr/bin/ld: warning: libopencv_calib3d.so.2.4, needed by /usr/local/lib/libcvsba.so, may conflict with libopencv_calib3d.so.3.1
[100%] Built target epipolar_geometry
Linking CXX executable ../../../bin/rtabmap-odometryViewer
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by /usr/local/lib/libcvsba.so, may conflict with libopencv_core.so.3.1
[100%] Built target calibration
Linking CXX executable ../../../bin/rtabmap-rgbd_mapping
Linking CXX executable ../../../bin/rtabmap-wifi_mapping
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by /usr/local/lib/libcvsba.so, may conflict with libopencv_core.so.3.1
[100%] Built target odometryViewer
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by /usr/local/lib/libcvsba.so, may conflict with libopencv_core.so.3.1
[100%] Built target wifi_mapping
Linking CXX executable ../../../bin/rtabmap-noEventsExample
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by /usr/local/lib/libcvsba.so, may conflict with libopencv_core.so.3.1
[100%] Built target rgbd_mapping
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by /usr/local/lib/libcvsba.so, may conflict with libopencv_core.so.3.1
[100%] Built target noEventsExample

I got above linakge warning while build rtabmap from source with Opencv3.1

matlabbe, 2016-06-22:
It seems that you may build cvsba with OpenCV 2 before installing OpenCV 3. Try rebuild cvsba with OpenCV 3 instead.

willdzeng, 2016-06-23:
I am having problem buiding cvsba with Opencv3. Any idea why this is happening?

> /home/zeng/essential/cvsba-1.0.0/src/sba_levmar.c: In function ‘emalloc_’:
/home/zeng/essential/cvsba-1.0.0/src/sba_levmar.c:70:5: warning: format ‘%u’ expects argument of type ‘unsigned int’, but argument 3 has type ‘size_t’ [-Wformat=]
     fprintf(stderr, "SBA: memory allocation request for %u bytes failed in file %s, line %d, exiting", sz, file, line);
     ^
/home/zeng/essential/cvsba-1.0.0/src/sba_chkjac.c: In function ‘emalloc_’:
/home/zeng/essential/cvsba-1.0.0/src/sba_chkjac.c:41:5: warning: format ‘%u’ expects argument of type ‘unsigned int’, but argument 3 has type ‘size_t’ [-Wformat=]
     fprintf(stderr, "SBA: memory allocation request for %u bytes failed in file %s, line %d, exiting", sz, file, line);
     ^
/home/zeng/essential/cvsba-1.0.0/src/cvsba.cpp:479:10: warning: unused parameter ‘i’ [-Wunused-parameter]
     void Sba::img_projKDRTS ( int j, int i, double *aj, double *bi, double *xij, void *adata ) {
          ^
/home/zeng/essential/cvsba-1.0.0/src/cvsba.cpp:601:10: warning: unused parameter ‘i’ [-Wunused-parameter]
     void Sba::img_projKDS ( int j, int i, double *bi, double *xij, void *adata ) {
          ^
/home/zeng/essential/cvsba-1.0.0/src/cvsba.cpp:622:10: warning: unused parameter ‘i’ [-Wunused-parameter]
     void Sba::img_projKDS_jac ( int j, int i, double *bi, double *Bij, void *adata ) {
          ^
Linking CXX shared library libcvsba.so
[ 66%] Built target cvsba
Scanning dependencies of target generate_synthetic_data
Scanning dependencies of target test_cvsba
[ 77%] [ 88%] [100%] Building CXX object utils/CMakeFiles/test_cvsba.dir/test_cvsba.cpp.o
Building C object utils/CMakeFiles/test_cvsba.dir/readparams.c.o
Building CXX object utils/CMakeFiles/generate_synthetic_data.dir/generate_synthetic_data.cpp.o
/home/zeng/essential/cvsba-1.0.0/utils/test_cvsba.cpp: In function ‘void readDataCVFormat(char*, char*, std::vector<cv::Point3_<float> >&, std::vector<std::vector<cv::Point_<float> > >&, std::vector<std::vector<int> >&, std::vector<cv::Mat>&, std::vector<cv::Mat>&, std::vector<cv::Mat>&, std::vector<cv::Mat>&)’:
/home/zeng/essential/cvsba-1.0.0/utils/test_cvsba.cpp:144:47: error: ‘numeric_limits’ is not a member of ‘std’
       imagePoints[i][j].x=imagePoints[i][j].y=std::numeric_limits<float>::quiet_NaN();
                                               ^
/home/zeng/essential/cvsba-1.0.0/utils/test_cvsba.cpp:144:67: error: expected primary-expression before ‘float’
       imagePoints[i][j].x=imagePoints[i][j].y=std::numeric_limits<float>::quiet_NaN();
                                                                   ^
/home/zeng/essential/cvsba-1.0.0/utils/test_cvsba.cpp:144:67: error: expected ‘;’ before ‘float’
/home/zeng/essential/cvsba-1.0.0/utils/generate_synthetic_data.cpp: In function ‘void generateSyntheticData(std::vector<cv::Point3_<double> >&, std::vector<std::vector<cv::Point_<double> > >&, std::vector<std::vector<int> >&, std::vector<cv::Mat>&, std::vector<cv::Mat>&, std::vector<cv::Mat>&, std::vector<cv::Mat>&)’:
/home/zeng/essential/cvsba-1.0.0/utils/generate_synthetic_data.cpp:123:34: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
  for(int i=0; i<imagePoints.size(); i++) {
                                  ^
/home/zeng/essential/cvsba-1.0.0/utils/generate_synthetic_data.cpp:124:40: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for(int j=0; j<imagePoints[i].size(); j++) {
                                        ^
/home/zeng/essential/cvsba-1.0.0/utils/generate_synthetic_data.cpp:130:30: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for(int j=0; j<points.size(); j++) {
                              ^
/home/zeng/essential/cvsba-1.0.0/utils/generate_synthetic_data.cpp: In function ‘void saveToFile(char*, char*, std::vector<cv::Point3_<double> >&, std::vector<std::vector<cv::Point_<double> > >&, std::vector<std::vector<int> >&, std::vector<cv::Mat>&, std::vector<cv::Mat>&, std::vector<cv::Mat>&, std::vector<cv::Mat>&)’:
/home/zeng/essential/cvsba-1.0.0/utils/generate_synthetic_data.cpp:182:30: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for(int i=0; i<points.size(); i++) {

willdzeng, 2016-06-23:
Also I got this,

> /usr/bin/ld: warning: libopencv_imgproc.so.2.4, needed by /usr/local/lib/libopencv_calib3d.so.2.4, may conflict with libopencv_imgproc.so.3.1

Do I need to remove 2.4 in order to install 3.1?

matlabbe, 2016-06-23:
Did you clear the `CMakeCache.txt` of cvsba before rebuilding (so that it can use OpenCV 3 include directories)?

On Windows 10, I have cvsba built with OpenCV 3, so I don't think it is an OpenCV version issue. Normally, you can have both OpenCV installed at the same time, but you have to make sure that cmake is finding the right `OpenCVConfig.cmake` (because of the PATH, your OpenCV 3 installed in /usr/local/ would be found before OpenCV 2 installed in /usr). Using `cmake -DOpenCV_DIR="directory/where/OpenCVConfig.cmake/is"` is the manual way to build against a specific OpenCV.

If you plan to build rtabmap_ros against rtabmap built with OpenCV3, be aware of the related issues at [point 1](https://github.com/introlab/rtabmap_ros/#build-from-source) of rtabmap_ros build instructions:

> RTAB-Map can build with OpenCV3+xfeatures2d module, but rtabmap_ros package will have libraries conflict as cv-bridge is depending on OpenCV2. If you want OpenCV3, you should build ros vision-opencv package yourself (and all ros packages depending on it) so it can link on OpenCV3.

cheers

willdzeng, 2016-06-23:
I removed the build folder and build again, it's still having the same issue.
I built cv-bridge from sources.

cheers.

matlabbe, 2016-06-23:
I don't have have the setup to test cvsba on Ubuntu with OpenCV2 and OpenCV3, but you can disable cvsba in rtabmap build for now (cvsba is only used for optional Bundle Adjustement post-processing, not used in rtabmap_ros):
```bash 
cmake -DWITH_CVSBA=OFF ..
```

willdzeng, 2016-06-24:
Got it.

closed 2016-06-24 by willdzeng