introlab/rtabmap #88: Can't find the binary files

willdzeng, 2016-06-24:
After I re-build the rtabmap without the DCMAKE_INSTALL_PREFIX to ROS working space, seems like the symbolic ink doesn't work anymore. 

> rtabmap-databaseViewer: error while loading shared libraries: librtabmap_gui.so.0.11: cannot open shared object file: No such file or directory

How can I fix this?

matlabbe, 2016-06-24:
Depending on where you are installing, the libraries should be on the `PATH` or `LD_LIBRARY_PATH` when starting `rtabmap` or one of its tools.

By default with a clean build directory:
```bash
$ cd rtabmap/build
$ rm CMakeCache.txt
$ cmake ..
$ make -j4
$ sudo make install 
```
This installs in `/usr/local`, the libs should be in `/usr/local/lib` or` /usr/local/lib/x86_64-linux-gnu`, then the libraries should be found automatically when executing `rtabmap`.

Another example if you install in ROS workspace:
```bash
$ cd rtabmap/build
$ rm CMakeCache.txt
$ cmake -DCMAKE_INSTALL_PREFIX=~/catkin_ws/devel ..
$ make -j4
$ make install
```
This installs in `~/catkin_ws/devel`, the libs should be in `~/catkin_ws/devel/lib` or `~/catkin_ws/devel/lib/x86_64-linux-gnu`. To find the libraries, you should `$ source ~/catkin_ws/devel/setup.bash` **or** add `~/catkin_ws/devel/lib:~/catkin_ws/devel/lib/x86_64-linux-gnu`
to `LD_LIBRARY_PATH`.

willdzeng, 2016-06-24:
I build it in /usr/local

```
zeng@m2:/usr/local/lib/x86_64-linux-gnu$ ls -l
total 9912
lrwxrwxrwx 1 root root      23 Jun 24 16:04 librtabmap_core.so -> librtabmap_core.so.0.11
lrwxrwxrwx 1 root root      25 Jun 24 16:04 librtabmap_core.so.0.11 -> librtabmap_core.so.0.11.7
-rw-r--r-- 1 root root 6421300 Jun 24 15:50 librtabmap_core.so.0.11.7
lrwxrwxrwx 1 root root      22 Jun 24 16:04 librtabmap_gui.so -> librtabmap_gui.so.0.11
lrwxrwxrwx 1 root root      24 Jun 24 16:04 librtabmap_gui.so.0.11 -> librtabmap_gui.so.0.11.7
-rw-r--r-- 1 root root 3571263 Jun 24 15:51 librtabmap_gui.so.0.11.7
lrwxrwxrwx 1 root root      26 Jun 24 16:04 librtabmap_utilite.so -> librtabmap_utilite.so.0.11
lrwxrwxrwx 1 root root      28 Jun 24 16:04 librtabmap_utilite.so.0.11 -> librtabmap_utilite.so.0.11.7
-rw-r--r-- 1 root root  145626 Jun 24 15:49 librtabmap_utilite.so.0.11.7
drwxr-xr-x 2 root root    4096 Jun 23 15:20 pkgconfig
drwxr-xr-x 2 root root    4096 Jun 24 16:04 rtabmap-0.11
```
But I am still have issue with the command,

rtabmap: error while loading shared libraries: librtabmap_core.so.0.11: cannot open shared object file: No such file or directory

willdzeng, 2016-06-24:
But my rtabmap_ros can find the library, that's why I am saying there is a symbolic link issue

matlabbe, 2016-06-25:
I reproduced the error, I working on it.

matlabbe, 2016-06-25:
Ok, from this [post](http://askubuntu.com/questions/350068/where-does-ubuntu-look-for-shared-libraries), I did:
```bash
sudo ldconfig
```
and it worked.

willdzeng, 2016-06-25:
Really. I tried this, doesn't work for me. 
Don't worry about this dude, there is a pointer bug in the new commit.

willdzeng, 2016-07-09:
Fixed this by reinstall ubuntu.

closed 2016-07-09 by willdzeng