introlab/rtabmap #90: Project Tango Compilation

yhoussin, 2016-06-29:
Hi Mathieu,
I'm trying to compile rtabmap using the standalone toolchain from android-ndk-r12.
I have been following your recommendations using Code Works from a previous post. I'm getting the following message and I'm not sure how to address it. I can't find that res_tool anywhere.
Any suggestion? Thanks in advance!

CMake Error at corelib/src/CMakeLists.txt:263 (MESSAGE):
  RTABMAP_RES_TOOL is not defined (it is the path to "rtabmap-res_tool"
  application created by a non-Android build).

matlabbe, 2016-06-29:
I would need to make this step automatic  for easier compilation. If you don't have rtabmap already built for the host computer, you have to build it using standard instructions for the host to get `rtabmap-res_tool` build in `bin` folder. Then back to your rtabmap android build, you can do `cmake -DRTABMAP_RES_TOOL="path/to/rtabmap-res_tool" ..`.

cheers

yhoussin, 2016-06-29:
OK I'll give it a shot, thanks for your quick reply!

zhngling123, 2016-07-28:
@matlabbe
Hi,
I just go to the final step for Tango compile, rtabmap-res_tools and qt5/vtk/eigen/flann/pcl with NDK compile, also use OpenCV from NV Pack you suggested. Tango C client API download is only provide TangSKD-Qianru-C.zip so I can't find Mira version for compare. For boost, I tried 1.61 and 1.55 compiled with NDK and boost file you provide in #74 . Please take some time to look at following error message. I am very appreciated our advice.

Thanks

`[ 79%] Building CXX object app/android/jni/CMakeFiles/NativeRTABMap.dir/RTABMapApp.cpp.o
/Volumes/data/rtabmap/app/android/jni/RTABMapApp.cpp:60:1: warning: identifier 'constexpr' is a keyword in C++11 [-Wc++0x-compat]
 constexpr int kTangoCoreMinimumVersion = 9377;
 ^
In file included from /Volumes/data/rtabmap/app/android/jni/tango-gl/include/tango-gl/drawable_object.h:22:0,
                 from /Volumes/data/rtabmap/app/android/jni/tango-gl/include/tango-gl/line.h:20,
                 from /Volumes/data/rtabmap/app/android/jni/tango-gl/include/tango-gl/axis.h:20,
                 from /Volumes/data/rtabmap/app/android/jni/scene.h:25,
                 from /Volumes/data/rtabmap/app/android/jni/RTABMapApp.h:37,
                 from /Volumes/data/rtabmap/app/android/jni/RTABMapApp.cpp:30:
/Volumes/data/rtabmap/app/android/jni/tango-gl/include/tango-gl/color.h:25:25: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   Color(const Color&) = default;
                         ^
/Volumes/data/rtabmap/app/android/jni/tango-gl/include/tango-gl/color.h:26:36: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   Color& operator=(const Color&) = default;
                                    ^
In file included from /Volumes/data/rtabmap/app/android/jni/tango-gl/include/tango-gl/drawable_object.h:23:0,
                 from /Volumes/data/rtabmap/app/android/jni/tango-gl/include/tango-gl/line.h:20,
                 from /Volumes/data/rtabmap/app/android/jni/tango-gl/include/tango-gl/axis.h:20,
                 from /Volumes/data/rtabmap/app/android/jni/scene.h:25,
                 from /Volumes/data/rtabmap/app/android/jni/RTABMapApp.h:37,
                 from /Volumes/data/rtabmap/app/android/jni/RTABMapApp.cpp:30:
/Volumes/data/rtabmap/app/android/jni/tango-gl/include/tango-gl/transform.h:29:39: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   Transform(const Transform& other) = delete;
                                       ^
/Volumes/data/rtabmap/app/android/jni/tango-gl/include/tango-gl/transform.h:30:54: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   const Transform& operator=(const Transform& rhs) = delete;
                                                      ^
In file included from /Volumes/data/rtabmap/app/android/jni/tango-gl/include/tango-gl/line.h:20:0,
                 from /Volumes/data/rtabmap/app/android/jni/tango-gl/include/tango-gl/axis.h:20,
                 from /Volumes/data/rtabmap/app/android/jni/scene.h:25,
                 from /Volumes/data/rtabmap/app/android/jni/RTABMapApp.h:37,
                 from /Volumes/data/rtabmap/app/android/jni/RTABMapApp.cpp:30:
/Volumes/data/rtabmap/app/android/jni/tango-gl/include/tango-gl/drawable_object.h:30:49: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   DrawableObject(const DrawableObject& other) = delete;
                                                 ^
/Volumes/data/rtabmap/app/android/jni/tango-gl/include/tango-gl/drawable_object.h:31:60: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   const DrawableObject& operator=(const DrawableObject&) = delete;
                                                            ^
In file included from /Volumes/data/rtabmap/app/android/jni/scene.h:26:0,
                 from /Volumes/data/rtabmap/app/android/jni/RTABMapApp.h:37,
                 from /Volumes/data/rtabmap/app/android/jni/RTABMapApp.cpp:30:
/Volumes/data/rtabmap/app/android/jni/tango-gl/include/tango-gl/camera.h:26:33: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   Camera(const Camera& other) = delete;
                                 ^
/Volumes/data/rtabmap/app/android/jni/tango-gl/include/tango-gl/camera.h:27:38: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   Camera& operator=(const Camera&) = delete;
                                      ^
In file included from /Volumes/data/rtabmap/app/android/jni/tango-gl/include/tango-gl/gesture_camera.h:21:0,
                 from /Volumes/data/rtabmap/app/android/jni/scene.h:28,
                 from /Volumes/data/rtabmap/app/android/jni/RTABMapApp.h:37,
                 from /Volumes/data/rtabmap/app/android/jni/RTABMapApp.cpp:30:
/Volumes/data/rtabmap/app/android/jni/tango-gl/include/tango-gl/segment.h:28:29: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   Segment(const Segment&) = default;
                             ^
/Volumes/data/rtabmap/app/android/jni/tango-gl/include/tango-gl/segment.h:29:40: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   Segment& operator=(const Segment&) = default;
                                        ^
[... cut ...]
     static const error_category &  errno_ecat     = generic_category();
                                    ^
/Volumes/data/android_install/include/boost/system/error_code.hpp:224:36: warning: 'boost::system::native_ecat' defined but not used [-Wunused-variable]
     static const error_category &  native_ecat    = system_category();
                                    ^
make[2]: *** [app/android/jni/CMakeFiles/NativeRTABMap.dir/RTABMapApp.cpp.o] Error 1
make[1]: *** [app/android/jni/CMakeFiles/NativeRTABMap.dir/all] Error 2
make: *** [all] Error 2
'

zhngling123, 2016-07-28:
Locate to 2 errors,
1: [Line 60](https://github.com/introlab/rtabmap/blob/master/app/android/jni/RTABMapApp.cpp#L60),constexpr int kTangoCoreMinimumVersion = 9377;
Error description: 'constexpr' does not name a type
2:[Line 758](https://github.com/introlab/rtabmap/blob/master/app/android/jni/RTABMapApp.cpp#L761),pcl::PointCloud<pcl::PointXYZRGBNormal>::Ptr denseCloud(new pcl::PointCloud<pcl::PointXYZRGBNormal>);
Error description:no matching function for call to 'boost::shared_ptrpcl::PointCloud<pcl::PointXYZRGBNormal >::shared_ptr(const boost::shared_ptrpcl::PointCloud<pcl::Normal >&)'

Any ideas for this 2 lines?

zhngling123, 2016-07-30:
Update:
Tried pcl 1.7.2, the error remain.
1:comment the line to pass the compile
2:Line 758,pcl::PointCloudpcl::PointXYZRGBNormal::Ptr denseCloud(new pcl::PointCloudpcl::PointXYZRGBNormal);
Error description:no matching function for call to 'boost::shared_ptrpcl::PointCloud::shared_ptr(const boost::shared_ptrpcl::PointCloud&)'

matlabbe, 2016-08-01:
For the first error, it may be fixed with the above c++11 commit.

matlabbe, 2016-08-01:
The second commit would fix the second error.

cheers

zhngling123, 2016-08-17:
Hi @matlabbe 
Following error occurred when linking final shared object, I guess its related to rtabmap code and not PCL.
`[ 11%] Built target rtabmap_utilite
[ 12%] Built target res_tool
[ 79%] Built target rtabmap_core
Linking CXX shared library ../libs/armeabi-v7a/libNativeRTABMap.so
/home/zhngling/libs/android-ndk-r12b/toolchains/arm-linux-androideabi-4.9/prebuilt/linux-x86_64/bin/../
lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: error: /home/zhngling/android_install/
lib/libpcl_features.a(normal_3d.cpp.o): multiple definition of 'typeinfo name for pcl::NormalEstimationOMP<pcl::Poi
ntXYZRGB, pcl::Normal>'
/home/zhngling/libs/android-ndk-r12b/toolchains/arm-linux-androideabi-4.9/prebuilt/linux-x86_64/bin/../
lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: ../../../../lib/librtabmap_core.a(uti
l3d_surface.cpp.o): previous definition here
/home/zhngling/libs/android-ndk-r12b/toolchains/arm-linux-androideabi-4.9/prebuilt/linux-x86_64/bin/../
lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: error: /home/zhngling/android_install/
lib/libpcl_features.a(normal_3d.cpp.o): multiple definition of 'typeinfo for pcl::NormalEstimationOMP<pcl::PointXYZ
RGB, pcl::Normal>'
/home/zhngling/libs/android-ndk-r12b/toolchains/arm-linux-androideabi-4.9/prebuilt/linux-x86_64/bin/../
lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: ../../../../lib/librtabmap_core.a(uti
l3d_surface.cpp.o): previous definition here
/home/zhngling/android_install/lib/libpcl_io.a(ply_io.cpp.o):ply_io.cpp:function pcl::PLYReader::parse(std::string c
onst&): error: undefined reference to 'pcl::io::ply::ply_parser::parse(std::string const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [app/android/libs/armeabi-v7a/libNativeRTABMap.so] Error 1
make[1]: *** [app/android/jni/CMakeFiles/NativeRTABMap.dir/all] Error 2
make: *** [all] Error 2`

willdzeng, 2016-08-17:
@zhngling123 Are you using PCL1.7 ?

zhngling123, 2016-08-18:
@willdzeng 
pcl-1.8.0

willdzeng, 2016-08-18:
try use 1.7

zhngling123, 2016-08-18:
After try pcl 1.7.2, it seems that multiple definition gone. But undefined reference to 'pcl::io::ply::ply_parser::parse(std::string const&)' is still.

Linking CXX executable ../../../bin/rtabmap-bow_mapping
/home/zhngling/android_install/lib/libpcl_io.a(ply_io.cpp.o):ply_io.cpp:function pcl::PLYReader::parse(std::string c
onst&): error: undefined reference to 'pcl::io::ply::ply_parser::parse(std::string const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [../bin/rtabmap-bow_mapping] Error 1
make[1]: *** [examples/BOWMapping/CMakeFiles/bow_mapping.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Linking CXX shared library ../libs/armeabi-v7a/libNativeRTABMap.so
/home/zhngling/android_install/lib/libpcl_io.a(ply_io.cpp.o):ply_io.cpp:function pcl::PLYReader::parse(std::string c
onst&): error: undefined reference to 'pcl::io::ply::ply_parser::parse(std::string const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [app/android/libs/armeabi-v7a/libNativeRTABMap.so] Error 1
make[1]: *** [app/android/jni/CMakeFiles/NativeRTABMap.dir/all] Error 2
make: *** [all] Error 2

yhoussin, 2016-08-18:
Hi Mathieu,
I made some progress compiling rtgbmap, got PCL1.8 compiled, but when I try to compile rtgbmap for Tango, I am getting linking errors (see bellow)
Can you confirm you are using PCL1.8 , and maybe you will have an idea about the error we have?

../../../../lib/librtabmap_core.a(util3d_registration.cpp.o):util3d_registration.cpp:vtable for pcl::registration::CorrespondenceEstimationBase<pcl::PointNormal, pcl::PointNormal, float>: error: undefined reference to 'pcl::PCLBase<pcl::PointNormal>::setIndices(unsigned int, unsigned int, unsigned int, unsigned int)'
collect2: error: ld returned 1 exit status
make[2]: *** [app/android/libs/armeabi-v7a/libNativeRTABMap.so] Error 1
make[1]: *** [app/android/jni/CMakeFiles/NativeRTABMap.dir/all] Error 2
make: *** [all] Error 2

matlabbe, 2016-08-22:
The Tango app is built with PCL 1.7.2 on my computer. 

For `error: undefined reference to 'pcl::io::ply::ply_parser::parse(std::string const&)` error, I manually added the missing library "`libpcl_io_ply.a`" to **PCLConfig.cmake** installed (e.g., `/Users/mathieu/Documents/workspace/android_libs/install/share/pcl-1.7/PCLConfig.cmake`):
```cmake
  pcl_remove_duplicate_libraries(PCL_LIBRARIES PCL_DEDUP_LIBRARIES)
  set(PCL_LIBRARIES ${PCL_DEDUP_LIBRARIES})
+ set(PCL_LIBRARIES ${PCL_LIBRARIES} /Users/mathieu/Documents/workspace/android_libs/install/lib/libpcl_io_ply.a)
  # Add 3rd party libraries, as user code might include our .HPP implementations
  list(APPEND PCL_LIBRARIES ${BOOST_LIBRARIES} ${QHULL_LIBRARIES} ${OPENNI_LIBRARIES} ${OPENNI2_LIBRARIES} ${FLANN_LIBRARIES} ${VTK_LIBRARIES})
```

This is not related to the error, but I also I added writing OBJ/PLY optimizations from this [pull request](https://github.com/PointCloudLibrary/pcl/pull/1580).

cheers

matlabbe, 2017-03-23:
Closed by this post: https://github.com/introlab/rtabmap/issues/74#issuecomment-288602733

closed 2017-03-23 by matlabbe