introlab/rtabmap #91: fatal error occurred when using rtabmap.exe

lynnsky, 2016-07-01:
I use rtabmap.exe to test loop closure.When using dataset `Lip6 Indoor` and ` Lip6 Outdoor`,I get right results.But dataset `New College` occured error when processing image id is around 650.I don't know why this happen.My rtabmap version is 0.11.7.
[image]
[image]

matlabbe, 2016-07-04:
Bug reproduced on 0.11.7 Windows binaries:
```bash
[DEBUG] (2016-07-04 12:28:33) Memory.cpp:3824::rtabmap::Memory::reactivateSignatures()
[ INFO] (2016-07-04 12:28:33) Memory.cpp:3835::rtabmap::Memory::reactivateSignatures() Loading location 2 from database...
[ INFO] (2016-07-04 12:28:33) Memory.cpp:3835::rtabmap::Memory::reactivateSignatures() Loading location 1 from database...
[DEBUG] (2016-07-04 12:28:33) Memory.cpp:3840::rtabmap::Memory::reactivateSignatures() idsToLoad = 2
[DEBUG] (2016-07-04 12:28:33) DBDriver.cpp:419::rtabmap::DBDriver::loadSignatures()
[DEBUG] (2016-07-04 12:28:33) DBDriver.cpp:459::rtabmap::DBDriver::loadSignatures()
[DEBUG] (2016-07-04 12:28:33) DBDriverSqlite3.cpp:1593::rtabmap::DBDriverSqlite3::loadSignaturesQuery() count=2
[DEBUG] (2016-07-04 12:28:33) Camera.cpp:87::rtabmap::Camera::takeImage() slept=0.199999s vs target=0.200000s
[DEBUG] (2016-07-04 12:28:33) CameraRGB.cpp:477::rtabmap::CameraImages::captureImage()
[DEBUG] (2016-07-04 12:28:33) DBDriverSqlite3.cpp:1687::rtabmap::DBDriverSqlite3::loadSignaturesQuery() Creating 2 (map=0, pose=xyz=[null] rpy=[null])
[DEBUG] (2016-07-04 12:28:33) DBDriverSqlite3.cpp:1687::rtabmap::DBDriverSqlite3::loadSignaturesQuery() Creating 1 (map=0, pose=xyz=[null] rpy=[null])
[DEBUG] (2016-07-04 12:28:33) CameraRGB.cpp:557::rtabmap::CameraImages::captureImage() Loading image : D:/NewCollege/60.jpg
[DEBUG] (2016-07-04 12:28:33) DBDriverSqlite3.cpp:1714::rtabmap::DBDriverSqlite3::loadSignaturesQuery() Time=0.004899s
[DEBUG] (2016-07-04 12:28:33) DBDriverSqlite3.cpp:1817::rtabmap::DBDriverSqlite3::loadSignaturesQuery() Add 400 keypoints, 400 3d points and 400 descriptors to node 2
[DEBUG] (2016-07-04 12:28:33) DBDriverSqlite3.cpp:1817::rtabmap::DBDriverSqlite3::loadSignaturesQuery() Add 400 keypoints, 400 3d points and 400 descriptors to node 1
[DEBUG] (2016-07-04 12:28:33) DBDriverSqlite3.cpp:1829::rtabmap::DBDriverSqlite3::loadSignaturesQuery() Time=0.009483s
[DEBUG] (2016-07-04 12:28:33) Signature.cpp:118::rtabmap::Signature::addLink() Add link 1 to 2 (type=0)
[DEBUG] (2016-07-04 12:28:33) Signature.cpp:118::rtabmap::Signature::addLink() Add link 3 to 2 (type=0)
[DEBUG] (2016-07-04 12:28:33) DBDriverSqlite3.cpp:2424::rtabmap::DBDriverSqlite3::loadLinksQuery() time=0.002483s, node=2, links.size=2
[DEBUG] (2016-07-04 12:28:33) Signature.cpp:118::rtabmap::Signature::addLink() Add link 2 to 1 (type=0)
[DEBUG] (2016-07-04 12:28:33) DBDriverSqlite3.cpp:2424::rtabmap::DBDriverSqlite3::loadLinksQuery() time=0.002463s, node=1, links.size=1
[DEBUG] (2016-07-04 12:28:33) DBDriverSqlite3.cpp:1836::rtabmap::DBDriverSqlite3::loadSignaturesQuery() Time load links=0.007428s
[DEBUG] (2016-07-04 12:28:33) DBDriverSqlite3.cpp:1908::rtabmap::DBDriverSqlite3::loadSignaturesQuery() Loading calibration for 1 cameras (72 bytes)
[FATAL] (2016-07-04 12:28:33) CameraModel.cpp:76::rtabmap::CameraModel::CameraModel() Condition (fx > 0.0) not met! [fx=0.000000]
```
On retrieval, the code tries to load an empty calibration from the database, which causes the fatal error.

The commit above fixes the problem for the next release `0.11.8`.

Workarounds for the current release `0.11.7`:
1. Use `rtabmap-console.exe`. If you are benhmarking, it is the easiest way to setup the parameters for each dataset. See [Benchmark](https://github.com/introlab/rtabmap/wiki/Benchmark) page for more info.
2. Add a fake calibration file. In Preferences -> Source, click on `Create Calibration`, set `fx` and `fy` to `1`, then save the calibration to `calib.yaml` (default). Back to `Source` panel, set `Calibration file path` to `calib`.

lynnsky, 2016-07-05:
OK.Thanks for your help

closed 2016-07-05 by lynnsky