introlab/rtabmap #94: question about downloading dataset NCO and Rawseeds

lynnsky, 2016-07-03:
NCO dataset size is 1626,but this link http://www.robots.ox.ac.uk/NewCollegeData/index.php?n=Main.ZippedPanoramaImages have 8100 images.How can I get right dataset NCO.
I get BO dataset through http://www.rawseeds.org/rs/capture_sessions/view/13 , I downloaded Bovisa_2008-10-04-FRONTAL.tar.torrent .But I can't get bo image through this file.
[image]

matlabbe, 2016-07-04:
For NCO, we used only 1 image of 5 to have approximately 1 Hz input rate (~8100 images at 5 Hz). For Bovisa, try to reach the owner if you have difficulties to download their dataset.

cheers

lynnsky, 2016-07-06:
OK!Thanks for your help

closed 2016-07-06 by lynnsky

xzwang12345, 2021-03-31:
＠lynnsky Did you solve how to download Bovisa datasets ? Thanks