introlab/rtabmap #95: RGB-D Hand-Held Mapping With a Kinect2 ERROR

QinZiwen, 2016-07-08:
I am use Ubuntu 14.04 , ROS_indigo, and kinect2 drive by libfreenect2 and iai_kinect2.Typing some commands as following:
+++++++++++++++++++++++++++
 $roslaunch kinect2_bridge kinect2_bridge.launch publish_tf:=true
 $ rosrun tf static_transform_publisher 0 0 0 -1.5707963267948966 0 -1.5707963267948966 camera_link kinect2_link 100
 $ roslaunch rtabmap_ros rgbd_mapping.launch rtabmap_args:="--delete_db_on_start" rgb_topic:=/kinect2/qhd/image_color_rect depth_registered_topic:=/kinect2/qhd/image_depth_rect camera_info_topic:=/kinect2/qhd/camera_info
+++++++++++++++++++++++++++

But error show me:
+++++++++++++++++++++++++++
odometry: Could not get transform from camera_link to kinect2_rgb_optical_frame (stamp=1467986384.023608) after 0.200000 seconds ("wait_for_transform_duration"=0.200000)!
+++++++++++++++++++++++++++

What should I do? 
Thanks

matlabbe, 2016-07-12:
I cannot reproduce the problem by launching your commands. Can you show the TF tree?
```bash
$ rosrun tf view_frames
$ evince frames.pdf
```

QinZiwen, 2016-07-13:
Yes,I type the command:
$ rosrun tf view_frames
$ evince frames.pdf

The TF tree could show me as the picture
[frames.pdf](https://github.com/introlab/rtabmap/files/362752/frames.pdf)

QinZiwen, 2016-07-13:
Can I change the "wait for transform duration" for long, such as 0.3000. Where can I change it?

matlabbe, 2016-07-14:
Mmmm your TF tree is wrong. It is missing some frames. Make sure you start [kinect2_bridge.launch](https://github.com/code-iai/iai_kinect2/blob/master/kinect2_bridge/launch/kinect2_bridge.launch#L5) with `publish_tf:=true`. The frame `kinect2_rgb_optical_frame` should appear in the TF tree.

To change the duration, you can start [rgbd_mapping.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/rgbd_mapping.launch#L42) with `wait_for_transform:=0.3`.

cheers

QinZiwen, 2016-07-14:
Thank you very much @matlabbe ,  The program run nice!

closed 2016-11-09 by matlabbe