introlab/rtabmap #96: kinect mapping

lynnsky, 2016-07-10:
I have a  ASUS Xtion with openni2 on USB3.When I tested kinect mapping,I used parameters file `config.ini` from  https://github.com/introlab/rtabmap/wiki/Change-parameters ,I selected `Show in 3D map view`.But I didn,t get right mapping.It solved one image,the program stops working.How can I mapping well.
[image]

When I test with rgbd_mapping,execute `rtabmap-rgbd_mapping.exe 1` ,It seems Xtion only get one images, the result are as follows:
[image]

Can  rtabmap  implement a complete SLAM system,include odometry 、loop closure detection、robust graph optimization?

matlabbe, 2016-07-11:
Try steps in "Low performance of ASUS Xtion with OpenNI2 on USB3" section of [Installation](https://github.com/introlab/rtabmap/wiki/Installation#low-performance-of-asus-xtion-with-openni2-on-usb3) page.

Before 2014, it was only a loop closure detector, but now, RTAB-Map is a complete SLAM system. 

cheers

lynnsky, 2016-07-12:
I have uncommented these three lines in file `PS1080.ini`.But it has no effect.Is it related with using pcl1.7.2 with openni2.2 in vs2010？

matlabbe, 2016-07-12:
I have also problems with my Xtion on a computer (on Ubuntu) with only USB3. Are you building rtabmap from source? I don't think it is the code the problem, as on USB2, it would work flawlessly.

You could also try OpenNI2 alone (I think there is a openni2 viewer app somewhere, e.g., [SimpleViewer](https://github.com/occipital/OpenNI2/tree/master/Samples/SimpleViewer)) to see if you can have a decent frame rate (~30 Hz).

lynnsky, 2016-07-12:
I bulid rtabmap from source.I run SimpleViewer very quickly.
Can rtabmap  use openni1.5?Previously I installed openni1.5 and build pcl1.7.2 from source.I can run rtabmap debug version(vs2010) to test loop closure.But when I change openni1.5 to openni2.2,I build pcl1.7.2 from source agian,I can't run rtabmap debug version(vs2010)，only release version can run.
I was using loop closure detection before,so I didn't use openni before.Now I want to run SLAM use xtion or dataset,there is problem with xtion.And I still don't know how to set parameters and calibrate xtion parameter using your calibrate program?

matlabbe, 2016-07-12:
For Freenect, OpenNI and OpenNI2 drivers, the factory calibration is used automatically (no need to calibrate) and manual calibration, if done in rtabmap, will be ignored. The rtabmap's calibration tool is mainly useful for stereo cameras.

For some reasons, the [code](https://github.com/introlab/rtabmap/blob/cee9c65b063f6fcdd3eb7559ac8a54f6f199449c/corelib/src/CameraRGBD.cpp#L687) here fails to retrieve a frame from OpenNI2. You could change the "`if`" for a "`while`" (removing the next "`else`") to keep retrying to retrieve a frame instead of returning null data (which is causing the "Camera has reached the end of stream" error).

lynnsky, 2016-07-13:
So I don't need to calibrate xtion parameters in rtabmap.
I run rgbd_mapping a few times,sometimes get different result,but all end up with no more images.
[image]

I change the` if `to `while`,no image appear in 3D map,the result are as follows:

[image]
[image]

matlabbe, 2016-07-13:
I was more thinking about something like this:
```cpp
while(openni::OpenNI::waitForAnyStream(depthStream, 1, &readyStream, 2000) != openni::STATUS_OK ||
	   openni::OpenNI::waitForAnyStream(colorStream, 1, &readyStream, 2000) != openni::STATUS_OK)
{
	UWARN("No frames received since the last 2 seconds, end of stream is reached!");
}
//else
{
	openni::VideoFrameRef depthFrame, colorFrame;
	...
```

lynnsky, 2016-07-13:
Yes,I run rgbd_mapping.It works.But there is problem with rtabmap.exe.I used parameters file `config.ini`,the program stops working.The error is because of `rtabmap_core.dll`,Exception code 0xc0000005.
[image]
[image]

matlabbe, 2016-07-14:
Not sure about this new error, but I tested again on a Windows 10 machine with OpenNI2 and Xtion Pro Live on USB3. I reproduced the latency problem that makes the camera stop. I just uncomment `InputFormat=1` in `C:\Program Files\OpenNI2\Redist\OpenNI2\Drivers\PS1080.ini` and the latencies are gone. You can revert back the changes of the code above.

cheers

lynnsky, 2016-07-14:
Yes,it is OK.Thank you!I think this error occur is because of  compatibility between openni2.2 and pcl1.7.2. I use `pcl_viewer` ,when I close `pcl_viewer`,exception code 0xc0000005 also occur.
Can rtabmap use rgbd dataset as input to test  SLAM?

matlabbe, 2016-07-14:
Do you mean [TUM's RGB-D SLAM Dataset](http://vision.in.tum.de/data/datasets/rgbd-dataset)? Yes it can, with the standalone or with ROS. See this [post](http://official-rtab-map-forum.67519.x6.nabble.com/How-to-process-RGBD-SLAM-datasets-with-RTAB-Map-tp939p647.html) for more info.

I have also this error when quitting RTAB-Map after I started the OpenNI2 driver (if I quit RTAB-Map without starting the camera, there is no error). It may be a driver related bug.

cheers

lynnsky, 2016-07-19:
Thank you.This post is useful for me.But when I executed ` associate.py rgb.txt depth.txt`,I didn't  discover "rgb_sync" and "depth_sync" directories

matlabbe, 2016-07-20:
I updated the link of associate.py with the modified one I used to copy matching images to sync folders. Try this one: [associate.py](https://gist.github.com/matlabbe/484134a2d9da8ad425362c6669824798)

lynnsky, 2016-07-21:
I have two problems.
1、I discover the error `Exception code 0xc0000005` appear and rtabmap debug version cann't execute because I  built rtabmap with g2o.
So I build rtabmap without g2o.Again  the latency problem that makes the camera stop appear,but this time I uncomment `InputFormat=1` or change while to if,both didn't work.
[image]
[image]
2、I create "rgb_sync" and "depth_sync" directories successfully.Then I create the file "rgbddatasets.yaml"  and copy it to  RTAB-Map/camera_info.I open RTAB-Map, open Preferences dialog. Click on "`Restore Defaults`".The next step is the same as what you said.I select` show in 3D map view`  ,But what I acquire is 
[image]
If I want to run SLAM,should I need to build rtabmap with g2o?

matlabbe, 2016-07-21:
You don't need g2o to run SLAM. To avoid some runtime errors of library incompatibilities, if you build rtabmap in debug, you should have all dependencies built in debug mode too. However, I rarely build in debug, so there are may be some problems.

You should click on "**Reset all settings**" button in "General Settings (GUI)" panel, not the "Restore Defaults" that just restore to defaults the current panel. You may be still in loop closure mode as there is nothing on Odomerty and 3D map views.

cheers

lynnsky, 2016-07-22:
Thank you for your help.Now I can run SLAM system.
 I am thinking why only 19 loop closure detected in 2488 images?How can I evaluate loop closure using rgbd_dataset_freiburg3_long_office_household dataset?I know rtabmap can generate LogF.txt and LogI.txt ,and use showlogs to evaluate ROC,but lack of groundtruth image.
I  executed `evaluated_ate.py groundtruth.txt poses.txt`,there was only one number(`0.044561`) appear,not result  you post.Why only 88 lines in poses.txt not 2488 lines?
By the way, how can I know which method use to motion estimation and optimize.And motion estimation use feature or word to match?I want to understand the whole SLAM implementation.Sorry for so many questions.

matlabbe, 2016-07-22:
In the referred [post](http://official-rtab-map-forum.67519.x6.nabble.com/How-to-process-RGBD-SLAM-datasets-with-RTAB-Map-tp939p647.html) above, see point 5 to add more nodes in the graph (thus more poses to compare with the ground truth):

> 5 To have more points to compare with the ground truth, you can check "Create intermediate nodes..." under Advanced RTAB-Map settings panel.

When you start RTAB-Map like that, it is done in online mode. In particular, if you set camera frame rate to 30 Hz and that odometry can process them only at 10 Hz, then you will have only 2488/3=830 images processed by odometry. If you don't activate "intermediate nodes", you will have only nodes added to map at 1 Hz, so 2488/30=~83 images. If intermediate nodes are activated and buffer size for rtabmap is 0 (inf), you would have the same number as odometry can process (e.g., in this example 830 nodes). If you even reduce the frame rate of the camera to 5 Hz, then all images will be processed (as odometry can be processed faster than camera frame rate, you can set odometry buffer to 0 to make sure to process them all), and you will get 2488 poses in the graph.

You can use the [online evaluation tool](http://vision.in.tum.de/data/datasets/rgbd-dataset/online_evaluation) if you want to visualize the trajectory compared to ground truth. You can also add the ground truth file in Preferences->Source panel to visualize directly in RTAB-Map the ground truth (shown as gray trajectory).

Unfortunately, there is no documentation about how odometry is done in RTAB-Map, or the different strategies. You may find some answers across questions over the forum. Parameters are mostly in `Odometry` and `Visual Registration` panels:
 1. Under Odometry, you can choose between F2M and F2F odometry. This is slightly explained in this [post](http://official-rtab-map-forum.67519.x6.nabble.com/Ransac-Frame-to-Frame-Stability-tp225p227.html).
 2. Under Visual Registration, you can choose the corresponding approach, using features matching (matching the descriptor like BOW approaches) or Optical Flow (tracking features). 
 3. Actual registration using 3D->3D or 2D->3D estimation. 2D->3D minimizes the reprojection error (PnP) of the matched features. 3D->3D minimizes the distances between the corresponding 3D features.

cheers

lynnsky, 2016-07-22:
I appreciate your help. you're always so kind.

closed 2016-07-22 by lynnsky