introlab/rtabmap #97: Disable multi-arch for catkin

matlabbe, 2016-07-12:
There is still installation issues with multi-arch on fresh Kinectix system: http://official-rtab-map-forum.67519.x6.nabble.com/Problem-while-running-RTAB-Map-installed-from-ROS-tp1530.html

A way to fix it would be to disable `INCLUDE(GNUInstallDirs)` or [uncomment](https://github.com/introlab/rtabmap/blob/f03a1b50ef6236862cc168e446b7cd227d8df635/CMakeLists.txt#L7-L13)
```cmake
# Catkin doesn't support multiarch library path, 
# fix to "lib" if not set by user.
#IF(NOT DEFINED CMAKE_INSTALL_LIBDIR)
#   set(CMAKE_INSTALL_LIBDIR "lib")
#ENDIF(NOT DEFINED CMAKE_INSTALL_LIBDIR)
```

To detect catkin, this could be done like in [find-object](https://github.com/introlab/find-object/blob/881f70377ffd03b73b8f8c0d19958a0948b860f4/CMakeLists.txt#L3-L6):
```cmake
# Detect if it is called by catkin
IF(CATKIN_TOPLEVEL OR CATKIN_BUILD_BINARY_PACKAGE OR CATKIN_SKIP_TESTING OR CATKIN_ENABLE_TESTING)
  SET(CATKIN_BUILD TRUE)
ENDIF(CATKIN_TOPLEVEL OR CATKIN_BUILD_BINARY_PACKAGE OR CATKIN_SKIP_TESTING OR CATKIN_ENABLE_TESTING)
```

matlabbe, 2016-07-21:
Seems not working: http://build.ros.org/job/Kbin_uW64__rtabmap__ubuntu_wily_amd64__binary/14/console
```
14:56:00 -- Info :
14:56:00 --   Version : 0.11.8
14:56:00 --   CMAKE_INSTALL_PREFIX = /opt/ros/kinetic
14:56:00 --   CMAKE_BUILD_TYPE =     None
14:56:00 --   CMAKE_INSTALL_LIBDIR = lib/x86_64-linux-gnu
14:56:00 --   BUILD_APP =            ON
14:56:00 --   BUILD_TOOLS =          ON
...
```

willdzeng, 2016-07-21:
I fixed this by deleting all the files in lib/x84_64-linux-gnu.
Also this can be fixed by doing a make uninstall in the old build folder

matlabbe, 2016-07-21:
I did a fast test on a Virtual machine with 16.04/kinetic. 

When doing cmake when building from source, `CMAKE_INSTALL_LIBDIR = lib`. Not sure why on the ros buildfarm it is with `x86_64-linux-gnu`. 

I also tested kinetic ros binaries. The `rtabmap_ros` launch files can be started without problem (even if the libraries are installed in `/opt/ros/kinetic/lib/x86_64-linux-gnu`).

I'll set this bug as fixed as the `not found library` error is not reproductible.

More info: 
```
$ env | grep kinetic
ROS_ROOT=/opt/ros/kinetic/share/ros
ROS_PACKAGE_PATH=/opt/ros/kinetic/share:/opt/ros/kinetic/stacks
LD_LIBRARY_PATH=/opt/ros/kinetic/lib:/opt/ros/kinetic/lib/x86_64-linux-gnu
PATH=/opt/ros/kinetic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
ROS_DISTRO=kinetic
PYTHONPATH=/opt/ros/kinetic/lib/python2.7/dist-packages
PKG_CONFIG_PATH=/opt/ros/kinetic/lib/pkgconfig
CMAKE_PREFIX_PATH=/opt/ros/kinetic
ROS_ETC_DIR=/opt/ros/kinetic/etc/ros
```
`/opt/ros/kinetic/lib/x86_64-linux-gnu` is already added from sourcing `/opt/ros/kinetic/setup.bash`. 

From included `/opt/ros/kinetic/_setup_util.py`:
```python
# subfolder of workspace prepended to CMAKE_PREFIX_PATH
ENV_VAR_SUBFOLDERS = {
    'CMAKE_PREFIX_PATH': '',
    'LD_LIBRARY_PATH' if not IS_DARWIN else 'DYLD_LIBRARY_PATH': ['lib', os.path.join('lib', 'x86_64-linux-gnu')],
    'PATH': 'bin',
    'PKG_CONFIG_PATH': [os.path.join('lib', 'pkgconfig'), os.path.join('lib', 'x86_64-linux-gnu', 'pkgconfig')],
    'PYTHONPATH': 'lib/python2.7/dist-packages',
}

```

closed 2016-07-21 by matlabbe