introlab/rtabmap #99: Calibration of Kinect2

francescoli, 2016-07-19:
Hi, @matlabbe 
  I managed to calibrate my kinect2 and get 3 yaml files under the path ~/Document/RTAB-MAP/camera_info. Loading any of them in the rtabmap preference leads to a crash with a error log says rectifyImage() Cannot rectify image because the rectify amp is not initialized. While deleting the yaml files makes rtabmap run normally.
  Any step missed when generating or loading the calibration file?
  Is it necessary to calibrate to achieve more accurate tracking and reconstruciont?
Regards,

matlabbe, 2016-07-20:
With which tool did you calibrate the kinect? Can you copy/paste the yaml files here?

The libfreenect2 driver (for kinect v2) uses built-in calibration parameters by default. It is why it works without an extra calibration step. Manual calibration should still work though.

francescoli, 2016-07-21:
Hi, @matlabbe 
  I just use the built-in calibration tool of rtabmap, as shown in the figure below
[image]

_pose.yaml
> %YAML:1.0
camera_name: "013940441747"
rotation_matrix:
   rows: 3
   cols: 3
   data: [ 9.9834493204544494e-01, -1.5118841788647459e-02,
       -5.5487091130691141e-02, 1.4992670956560544e-02,
       9.9988398656131239e-01, -2.6894675765925411e-03,
       5.5521315517232495e-02, 1.8531166253325426e-03,
       9.9845578243706357e-01 ]
translation_matrix:
   rows: 3
   cols: 1
   data: [ -1.8089336332965018e-02, 2.7117288454258019e-03,
       -1.3125160656998980e-02 ]
essential_matrix:
   rows: 3
   cols: 3
   data: [ 3.4733996780644621e-04, 1.3128663111784688e-02,
       2.6722416520922072e-03, -1.2099093874157307e-02,
       2.3195887732359720e-04, 1.8789679447577652e-02,
       -2.9784482173749950e-03, -1.8046239527465750e-02,
       1.9911662911745217e-04 ]
fundamental_matrix:
   rows: 3
   cols: 3
   data: [ 3.1471253500962005e-07, 1.1937372908127477e-05,
       -1.6517893516142638e-03, -1.0984192284527165e-05,
       2.1132712939060373e-07, 9.0132861520866095e-03,
       2.8893297040205124e-03, -2.8945935314534831e-02, 1. ]

_left.yaml

> %YAML:1.0
camera_name: "013940441747_left"
image_width: 512
image_height: 424
camera_matrix:
   rows: 3
   cols: 3
   data: [ 3.7007007135050287e+02, 0., 2.4983609815225259e+02, 0.,
       3.6876949514826669e+02, 2.0684495999791883e+02, 0., 0., 1. ]
distortion_coefficients:
   rows: 1
   cols: 5
   data: [ 1.5820225951722386e-01, -4.4945572937895495e-01,
       1.7853919011533909e-03, -5.1251419163920280e-03,
       2.5753972191714675e-01 ]
distortion_model: plumb_bob
rectification_matrix:
   rows: 3
   cols: 3
   data: [ 1., 0., 0., 0., 1., 0., 0., 0., 1. ]
projection_matrix:
   rows: 3
   cols: 4
   data: [ 3.7007007135050287e+02, 0., 2.4983609815225259e+02, 0., 0.,
       3.6876949514826669e+02, 2.0684495999791883e+02, 0., 0., 0., 1.,
       1. ]

_right.yaml
> %YAML:1.0
camera_name: "013940441747_right"
image_width: 1920
image_height: 1080
camera_matrix:
   rows: 3
   cols: 3
   data: [ 1.0680238448825476e+03, 0., 9.4696263314563600e+02, 0.,
       1.0659208318156980e+03, 5.5257514699071612e+02, 0., 0., 1. ]
distortion_coefficients:
   rows: 1
   cols: 5
   data: [ 1.3608974022792117e-01, -2.5060557659017163e-01,
       3.2289342332718291e-03, -5.9116698657805328e-03,
       1.6419759601560829e-01 ]
distortion_model: plumb_bob
rectification_matrix:
   rows: 3
   cols: 3
   data: [ 1., 0., 0., 0., 1., 0., 0., 0., 1. ]
projection_matrix:
   rows: 3
   cols: 4
   data: [ 1.0680238448825476e+03, 0., 9.4696263314563600e+02,
       2.0294832851200397e+02, 0., 1.0659208318156980e+03,
       5.5257514699071612e+02, 4.3361337314706822e+02, 0., 0., 1., 1. ]

matlabbe, 2016-07-21:
This was indeed a bug. This is now fixed in https://github.com/introlab/rtabmap/commit/1a1cf5f672bb8d581ca16c45af1c59b2f52449a3

Thx for reporting it!

francescoli, 2016-07-22:
Nice, that works fine now, cheers!

closed 2016-07-22 by francescoli