introlab/rtabmap #103: Map was bending seriously using ZED as mapping source

ShineSong, 2016-08-03:
Hi,

I'm using RTABMap with ZED sensor and there are 2 weird issues.

1. Map was bending seriously when we use RTABMap,as you can see in the snapshot below, the horizontal position is pretty good. But there are a huge deformation exist on the Z-axis. According to our previous experience in airborne photogrammetry, this error may caused by image distortion, but we can not confirm it and we have no idea to correct it.

[image]
[image]
2.We noticed that you invoke ZED SDK with "self-calibration" parameter. We compared the result in project, intrinsic and distortion are deflect from official calibration file about nx10 pixels. And we use same *.SVO file to perform SLAM, the self-calibration result changed every time. I wonder know if I turn off the self-calibration, can result be better?

[image]
this one is from ZED official SDK

[image]
and this one is created by self-calibration in RTABMap project folder.

matlabbe, 2016-08-03:
Hi,

It seems there is a [wrong comment](https://github.com/introlab/rtabmap/blob/ca6cd19fb184f63486644e7b40d365a63b2484bf/corelib/src/CameraStereo.cpp#L844) in the code. The self-calibration [would be disabled](https://github.com/introlab/rtabmap/blob/ca6cd19fb184f63486644e7b40d365a63b2484bf/corelib/src/CameraStereo.cpp#L852).

How exactly did you export the calibration files?

Indeed, comparing with a manual calibration could be done. The `Zed sdk` driver in RTAB-Map standalone doesn't support manual calibration, it automatically uses the factory one. I am not sure if we can force a calibration. With the `Stereo Usb Camera` driver, you can use the calibration tool in Preferences->Source->Calibrate, but the depth image will be generated using OpenCV (on CPU with `cv::stereoBM()`) instead of the ZED SDK (on GPU). 

cheers

matlabbe, 2016-11-09:
From this commit https://github.com/introlab/rtabmap/commit/d102c43f9aeabbfa939d3190d4ef70e68d6cabd1, self-calibration can be disable or enabled

closed 2016-11-09 by matlabbe