introlab/rtabmap #106: camera initialization error

AndreV84, 2016-08-22:
I have a rgb-d camera connected which is not supported by openni, I think.
I run rtabmap RGB mode but the return is:
CameraRGB.cpp:787::init() Parameter "rectifyImages" is set, but no camera model is loaded or valid
The camera is /dev/video0
The Depth is /dev/video1
How to fix that?

matlabbe, 2016-08-22:
Hi,

This particular error is that rectification option is enabled, but no calibration file is provided. What is your camera? Using both streams from `/dev/video0` and `/dev/video1` at the same time will not work with RTAB-Map standalone app. You may have to use ROS version of RTAB-Map instead ([rtabmap_ros](http://wiki.ros.org/rtabmap_ros)), and create/find a node that can stream your RGB-D images in a format similar to [freenect_launch](http://wiki.ros.org/freenect_launch) or [openni_launch](http://wiki.ros.org/openni_launch). Some examples with other cameras on this [tutorial](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping).

cheers

AndreV84, 2016-08-22:
The camera is https://www.leopardimaging.com/uploads/LI-ESP870-STEREO-M031_datasheet.pdf . It has /dev/video0 YUV and /dev/video1 in-depth outputs as it is displayed by ls /dev/video* . What nodes?

willdzeng, 2016-08-22:
You may need to write a simple wrapper for your camera in ROS that publishing RGB and Depth images using OpencCv using  cv::VideoCapture

matlabbe, 2016-08-22:
Related post on ROS answers: http://answers.ros.org/question/237939/leopard-imaging-li-esp870-stereo-m031-with-ros/

If the camera is calibrated (I assume yes if it outputs depth images), the calibration parameters should be somewhere. As referred in the link above, it seems there is an SDK to get images, so maybe the calibration parameters can be obtained with it. 

I think the easier is to create a ROS driver for this camera.  As @willdzeng said, cv::VideoCapture could be used to get the streams. The color images should be converted to BGR or grayscale, and depth image should be converted to `unsigned short 16 bits` (millimeters) or `float 32 bits` (meters). See also [REP 118](http://www.ros.org/reps/rep-0118.html).

A compatible calibration file can be created manually in Preferences->Source->Create Calibration. If you are using ROS, you would publish a [sensor_msgs/CameraInfo](http://docs.ros.org/api/sensor_msgs/html/msg/CameraInfo.html) message directly along 2 [sensor_msgs/Image](http://docs.ros.org/api/sensor_msgs/html/msg/Image.html) messages.

If you don't know ROS, a driver could be created in RTAB-Map library directly. See [CameraRGB](https://github.com/introlab/rtabmap/blob/master/corelib/include/rtabmap/core/CameraRGB.h), [CameraRGBD](https://github.com/introlab/rtabmap/blob/master/corelib/include/rtabmap/core/CameraRGBD.h) and [CameraStereo](https://github.com/introlab/rtabmap/blob/master/corelib/include/rtabmap/core/CameraStereo.h) for some examples.

cheers

closed 2016-11-09 by matlabbe