introlab/rtabmap #112: Not able to run rtab_mapping c++ example program

kdkalvik, 2016-09-05:
I get the following error when I try to run the program

rgbd_mapping: /usr/local/include/eigen3/Eigen/src/SVD/JacobiSVD.h:632: const MatrixVType& Eigen::JacobiSVD<MatrixType, QRPreconditioner>::matrixV() const [with _MatrixType = Eigen::Matrix<double, 3, 3>; int QRPreconditioner = 2; Eigen::JacobiSVD<MatrixType, QRPreconditioner>::MatrixVType = Eigen::Matrix<double, 3, 3>; typename _MatrixType::Scalar = double]: Assertion `computeV() && "This JacobiSVD decomposition didn't compute V. Did you ask for it?"' failed.
Aborted

I changed the main.cpp to use the ZED camera driver. The program builds and compiles properly.

matlabbe, 2016-09-08:
Is it a ZED related issue? Did you try a Kinect for example? 

RTAB-Map doesn't do Eigen calls, maybe there is problem with the libraries installed (different versions used by different libraries). Make sure you are using the same OpenCV/Eigen version than the installed ZED SDK.

kdkalvik, 2016-09-09:
No that can't be it. I built opencv, pcl and rtabmap from source. And zed uses the same version of eigen.

matlabbe, 2016-09-09:
Just saw this [post](https://stackoverflow.com/questions/38696684/zed-sdk-crashes-when-used-with-pclintegralimagenormalestimation). It is maybe related to PCL using the point clouds created with the ZED.

I tried again on my Windows computer (ZED 1.1.0, OpenCV 3.1, PCL 1.8rc2, Eigen 3.2.8) and I cannot reproduce the error. What are your computer specs?

kdkalvik, 2016-09-09:
I am working on a Nvidia Jetson TX1.

matlabbe, 2016-09-09:
Unfortunately, I don't own a Jetson TX1. @willdzeng, do you have any experience with the Jetson and ZED?

kdkalvik, 2016-09-09:
I got it working. I installed the program was looking JacobiSVD.h in usr/local/eigen... but when I checked the location I didn't have eigen header files there.So, I installed eigen from source and that fixed the issue. Thank you.

closed 2016-09-09 by kdkalvik