introlab/rtabmap #114: Depth calibration

matlabbe, 2016-09-07:
It can be very useful, along the camera intrinsic calibration, that we could have a depth calibration too: https://github.com/AlexReimann/depth_calibration.

This could limit the distorsion in the point clouds created, thus getting more accurate depth values.

willdzeng, 2016-09-07:
Does this work with stereo camera also?

matlabbe, 2016-09-08:
With a good stereo calibration, there should not have such distortions (the noise in the point clouds is more related to the stereo disparity approach used). The approach above is specific for RGB-D cameras where an extra step after RGB/IR calibration is required to calibrate precisely the depth.

matlabbe, 2016-09-19:
Example of results:
[image]

matlabbe, 2016-09-19:
See this tutorial to know how to use it: https://github.com/introlab/rtabmap/wiki/Depth-Calibration-(CLAMS)

closed 2016-09-19 by matlabbe