introlab/rtabmap #116: rtabmap hangs when I select "open database"

fred087, 2016-09-09:
System Info : 
Arch Linux x86_64 up to date.

Here are the last messages : 

```
[ INFO] (2016-09-09 13:11:51.350) PreferencesDialog.cpp:1945::writeSettings() modified Vis/SubPixWinSize = 3->3

(rtabmap:16263): GLib-GObject-WARNING **: cannot register existing type 'GdkWindow'

(rtabmap:16263): GLib-CRITICAL **: g_once_init_leave: assertion 'result != 0' failed

(rtabmap:16263): Gdk-CRITICAL **: gdk_window_ensure_native: assertion 'GDK_IS_WINDOW (window)' failed

```

matlabbe, 2016-09-09:
Qt may have dificulty to create a native dialog for file selection. It is similar to this thread: http://official-rtab-map-forum.67519.x6.nabble.com/RTab-Map-closes-unexpectedly-when-I-m-trying-to-save-database-tp1149p1312.html

fred087, 2016-09-19:
I replaced all QFileDialog::getSaveFileName by absolute path and it worked.

closed 2016-09-19 by matlabbe