introlab/rtabmap #118: Running rtabmap stand alone results in pointer problems

axjensen, 2016-09-11:
I'm trying to get rtabmap stand alone app running on Ubuntu 14.04. I've tried the binaries and compiling from source. Binaries resulted in segfault (something with VTK smart pointers). Running rtabmap compiled from source, gives:
*** Error in `rtabmap': realloc(): invalid pointer: 0x00007efe7a937840 ***
Aborted (core dumped)

with GDB:

Starting program: /usr/local/bin/rtabmap 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
*** Error in `/usr/local/bin/rtabmap': realloc(): invalid pointer: 0x00007ffff6291840 ***

Program received signal SIGABRT, Aborted.
0x00007ffff5505c37 in __GI_raise (sig=sig@entry=6)
    at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56	../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.

I've installed VTK 5.8, OpenCV 2.4.8 (and 3.1). 

(I've also tried rtabmap_ros with similar problems)

willdzeng, 2016-09-11:
Looks like it's a linkage problem, did you installed rtabmap from apt get before? if it is, you need to remove them.
Try use ldconfig.

axjensen, 2016-09-12:
I removed rtabmap installed by apt and ran ldconfig before compiling from source.

axjensen, 2016-09-12:
The problem seems to be related to linking as @willdzeng pointed out. rtabmap was linked to the OpenCV version referenced by ROS.
I just had to remove the ROS from the path.
Thanks for helping.

closed 2016-09-12 by axjensen