introlab/rtabmap #120: cannot import poses to g20 format

Exception4U, 2016-09-27:
error on console while trying to export poses to g2o format is 

[ERROR] (2016-09-27 14:30:57.980) Graph.cpp:79::exportPoses() Cannot export in g2o format because RTAB-Map is not built with g2o support!

matlabbe, 2016-09-27:
That means rtabmap is not built with g2o. I assume you are using ubuntu binaries. To be able to export in g2o format, you would have to build rtabmap [from source](https://github.com/introlab/rtabmap_ros#build-from-source) and that g2o is also installed.

Exception4U, 2016-09-27:
Hi , I did installed g2o but my g2o was not build with g2o viewer, but
still is shows g2o not installed. so I changed macro in the code as
WITH_G2O of overide to one to make it work.

matlabbe, 2016-09-27:
In the main [CMakeLists.txt](https://github.com/introlab/rtabmap/blob/3530da935b03b3709e01d30f06ee370119a87120/CMakeLists.txt#L246), g2o is found with "`FIND_PACKAGE(G2O QUIET)`". You can remove the `QUIET` to see if CMake has detected g2o but not considering it. You can also look in [FindG2O.cmake](https://github.com/introlab/rtabmap/blob/3530da935b03b3709e01d30f06ee370119a87120/cmake_modules/FindG2O.cmake) file to see what should be installed so that g2o is considered found.

Exception4U, 2016-09-28:
okay thanks for the same ...

closed 2016-09-28 by Exception4U