introlab/rtabmap #121: Program crashed in 3-step Calibration Xtion Pro Live OPENNI2

antoniodepth, 2016-09-30:
I was doing a Calibration (OPENNI2) with a Xtion Pro Live. I saved the PS1080_rgb.yaml and PS1080_depth.yaml files, but in the third step, when I did the 3 sample... the program crashed (attached image).
[image]

matlabbe, 2016-09-30:
Did you build with Visual Studio? You may click Debug to get more info where it crashes. You can start RTAB-Map from a terminal, set logging mode to Debug then re-calibrate. You may have more info on the terminal when it crashes. For re-calibration, note that you can skip the 2 first parts to go directly on step 3 for convenience. 

Which OpenCV version did you compile too?

matlabbe, 2016-09-30:
Reproduced the error on Windows:
```bash
[DEBUG] (2016-09-30 13:41:10) CalibrationDialog.cpp:289::rtabmap::CalibrationDialog::processImages() Processing images
[ INFO] (2016-09-30 13:41:10) CalibrationDialog.cpp:451::rtabmap::CalibrationDialog::processImages() [0] Added board, total=1. (x=0.578789, y=0.649814, size=0.109273, skew=0.304882)
[ INFO] (2016-09-30 13:41:10) CalibrationDialog.cpp:451::rtabmap::CalibrationDialog::processImages() [1] Added board, total=1. (x=0.560412, y=0.622078, size=0.102182, skew=0.276749)
[ INFO] (2016-09-30 13:41:10) CalibrationDialog.cpp:550::rtabmap::CalibrationDialog::processImages() Add stereo image points (size=1)
[ INFO] (2016-09-30 13:41:10) CalibrationDialog.cpp:833::rtabmap::CalibrationDialog::stereoCalibration() stereo calibration (samples=1)...
OpenCV Error: Assertion failed (!fixedSize() || ((Mat*)obj)->size.operator()() == _sz) in cv::_OutputArray::create, file ..\..\..\modules\core\src\matrix.cpp, line 2190
```

matlabbe, 2016-09-30:
That [commit ](https://github.com/introlab/rtabmap/commit/b17e0dc26d74e99884278656883da242160d5cad)should fix the crash, an error message is shown instead. Note that the only way I reproduced the crash was by ignoring step 1 and 2, then go straight to step 3 without calibration files saved previously (distortion model was null, which caused the crash in [cv::stereoCalibrate()](https://github.com/introlab/rtabmap/blob/b17e0dc26d74e99884278656883da242160d5cad/guilib/src/CalibrationDialog.cpp#L885)).

closed 2016-09-30 by matlabbe