introlab/rtabmap #123: DBDriverSqlite3.cpp:3195::addStatisticsQuery() Condition (rc == SQLITE_DONE) not met! [DB error (0.11.11): UNIQUE constraint failed: Statistics.id] terminate called after throwing an instance of 'UException'

vfdev-5, 2016-10-07:
Hello, 

I observe a strange behavior with rtabmap *master branch* and rtabmap_ros. After mapping phase and creating a db, I restart rtabmap node with localization option enabled (`Mem/IncrementalMemory: false`). Then usually quiting with ctrl+C the node, again I restart the node with localization option and I get the following error. 

After the first launch of the rtabmap node with localization option enabled, there is however the message when exiting the node: `rtabmap: Saving database/long-term memory... (located at ~/maps/_map.db)` 

Comparing the databases (*clean db* after creation and a *modified db* before the second launch of rtabnode with localization option enabled), Statistics table of the second db contains more elements than the clean db. The error says that at the second launch it tries to write to the base on already existing rows. 

Observing the logs during the 1st, I see that the data with `id = 90` marked as bad signature has been written to the db :
```
[DEBUG] (2016-10-07 21:23:00.962) Memory.cpp:1568::cleanup()
[DEBUG] (2016-10-07 21:23:00.962) Memory.cpp:1576::cleanup() Bad signature! 90
[DEBUG] (2016-10-07 21:23:00.962) Memory.cpp:1775::moveToTrash() id=90
[DEBUG] (2016-10-07 21:23:00.962) Memory.cpp:3656::disableWordsRef() id=90
[ INFO] (2016-10-07 21:23:00.962) Rtabmap.cpp:2378::process() timeMemoryCleanup = 0.000043s... 1 signatures removed
[ INFO] (2016-10-07 21:23:00.962) Rtabmap.cpp:2391::process() Total time processing = 0.026209s...
[ INFO] (2016-10-07 21:23:00.962) Rtabmap.cpp:2485::process() Time limit reached processing = 0.000009...
[ INFO] (2016-10-07 21:23:00.962) Rtabmap.cpp:2514::process() Adding data 90 (rgb/left=1 depth/right=1)

[DEBUG] (2016-10-07 21:23:00.962) Rtabmap.cpp:2517::process()
[DEBUG] (2016-10-07 21:23:00.962) Rtabmap.cpp:2532::process() Get all node infos...
[DEBUG] (2016-10-07 21:23:00.962) Rtabmap.cpp:2557::process()
[DEBUG] (2016-10-07 21:23:00.962) DBDriverSqlite3.cpp:3166::addStatisticsQuery()
[DEBUG] (2016-10-07 21:23:00.962) DBDriverSqlite3.cpp:3204::addStatisticsQuery() Time=0.000206s
``` 
and similarly written ids = 91, 92, etc.  Thus, the *clean db* becomes the *modified db*. 

See the full logs attached: log1 (first launch in localization mode) and log2 (the second launch). I removed the info on camera, etc nodes.

[log2.txt](https://github.com/introlab/rtabmap/files/517046/log2.txt)
[log1.txt](https://github.com/introlab/rtabmap/files/517047/log1.txt)

Can you hint where is the problem ?

Thank you
Cheers

closed 2016-10-11 by matlabbe

matlabbe, 2016-10-11:
I reproduced the bug with the [demo_robot_mapping.launch](http://wiki.ros.org/rtabmap_ros#Robot_mapping) demo:
```bash
$ roslaunch rtabmap_ros demo_robot_mapping.launch
$ rosbag play --clock demo_mapping.bag
# wait 15 sec and kill everything

$ roslaunch rtabmap_ros demo_robot_mapping.launch localization:=true
$ rosbag play --clock demo_mapping.bag
# wait 15 sec and kill everything

$ roslaunch rtabmap_ros demo_robot_mapping.launch localization:=true
$ rosbag play --clock demo_mapping.bag
# This would crash on start with the error above
```
The commit above should fix the problem. The problem was that on the the first localization run, statistics with IDs after the last ID in database (on mapping mode) were added to database. On the second localization run, since no new nodes were added to database, IDs were restarted to last ID on mapping run, not the previous localization run, so same IDs were used to save the statistics. The fix disables saving statistics to database if we are in localization mode (the database should not change on localization).

Thx for reporting!

vfdev-5, 2016-10-11:
Right, thank you for the fix, now it works also in my case.
However, I have the following warning : 
```
[ WARN] Memory.cpp:602::update() The working memory is empty and the memory is not incremental (Mem/IncrementalMemory=False), no loop closure can be detected! Please set Mem/IncrementalMemory=true to increase the memory with new images or decrease the STM size (which is 1 including the new one added).
```
can you hint, please, why it is shown ?

Anyway, glad to help with your amazing project!

matlabbe, 2016-10-11:
That means the map is empty and you are in localization mode. The robot will never localize to anything. Make sure you have also this [fix](https://github.com/introlab/rtabmap_ros/commit/9a8538711ddcaf8895ab6dcdd54f2d92c9c6c674) from 2 hours ago for `rtabmap_ros`. That bug made rtabmap always deleting the database on start.

commit 94c25e20 (2016-10-11): fixed #123