introlab/rtabmap #124: Error in `rtabmap': realloc(): Aborted (core dumped)

willdzeng, 2016-10-11:
*** Error in `rtabmap': realloc(): invalid pointer: 0x00007f8e61f35840 ***
Aborted (core dumped)

I am having pointer issue with the new commits.

matlabbe, 2016-10-11:
Cannot reproduce the error here (Ubuntu 14.04).

willdzeng, 2016-10-11:
It's on start

willdzeng, 2016-10-11:
I am having the same issue with both my computer

matlabbe, 2016-10-12:
No updates, I tried building the source on two virtual machines too (Ubuntu 14.04/Indigo and Ubuntu 16.04/Kinetic) and it starts correctly.

willdzeng, 2016-10-12:
I will try to build it again tomorrow to see what's going on.

willdzeng, 2016-10-12:
Hey Mathieu,

Can you tell me your cmake print when you build rtabmap

matlabbe, 2016-10-12:
```bash
~/workspace/rtabmap/build$ cmake -DCMAKE_INSTALL_PREFIX=~/catkin_ws/devel ..
-- checking for module 'eigen3'
--   found eigen3, version 3.2.0
-- Found eigen: /usr/include/eigen3  
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Found openni: /usr/lib/libOpenNI.so  
-- Found libusb-1.0: /usr/include  
-- checking for module 'flann'
--   found flann, version 1.8.4
-- Found Flann: /usr/lib/x86_64-linux-gnu/libflann_cpp_s.a  
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Found qhull: /usr/lib/x86_64-linux-gnu/libqhull.so  
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Found PCL_COMMON: /usr/lib/libpcl_common.so  
-- Found PCL_OCTREE: /usr/lib/libpcl_octree.so  
-- Found PCL_IO: /usr/lib/libpcl_io.so  
-- Found PCL_KDTREE: /usr/lib/libpcl_kdtree.so  
-- Found PCL_SEARCH: /usr/lib/libpcl_search.so  
-- Found PCL_SAMPLE_CONSENSUS: /usr/lib/libpcl_sample_consensus.so  
-- Found PCL_FILTERS: /usr/lib/libpcl_filters.so  
-- Found PCL_FEATURES: /usr/lib/libpcl_features.so  
-- Found PCL_KEYPOINTS: /usr/lib/libpcl_keypoints.so  
-- Found PCL_GEOMETRY: /usr/include/pcl-1.7  
-- Found PCL_SEGMENTATION: /usr/lib/libpcl_segmentation.so  
-- Found PCL_VISUALIZATION: /usr/lib/libpcl_visualization.so  
-- Found PCL_OUTOFCORE: /usr/lib/libpcl_outofcore.so  
-- Found PCL_REGISTRATION: /usr/lib/libpcl_registration.so  
-- Found PCL_RECOGNITION: /usr/lib/libpcl_recognition.so  
-- Found PCL_SURFACE: /usr/lib/libpcl_surface.so  
-- Found PCL_PEOPLE: /usr/lib/libpcl_people.so  
-- Found PCL_TRACKING: /usr/lib/libpcl_tracking.so  
-- Found PCL_APPS: /usr/lib/libpcl_apps.so  
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Found OpenMP: -fopenmp  
-- Found OpenMP
-- Found OpenCV: /usr/local/include/opencv;/usr/local/include
-- Found PCL: /usr/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/vtk-5.8
-- Found ZLIB: /usr/include
-- Looking for Q_WS_X11
-- Looking for Q_WS_X11 - found
-- Looking for Q_WS_WIN
-- Looking for Q_WS_WIN - not found
-- Looking for Q_WS_QWS
-- Looking for Q_WS_QWS - not found
-- Looking for Q_WS_MAC
-- Looking for Q_WS_MAC - not found
-- Found Qt4: /usr/bin/qmake (found version "4.8.6") 
-- Found QVTK: /usr/lib/libQVTK.so  
-- Found Freenect: /usr/include
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- Found CSPARSE: /usr/include/suitesparse  
-- Found g2o: /usr/local/include;/usr/include/suitesparse;/usr/include/suitesparse
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/local/include/gtsam/3rdparty/Eigen/
-- Found octomap: /opt/ros/indigo/include
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   Version : 0.11.11
--   CMAKE_INSTALL_PREFIX = /home/mathieu/catkin_ws/devel
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -Wno-deprecated -fopenmp -std=c++11
--   With OpenCV 2 nonfree module (SIFT/SURF) = YES (License: Non commercial)
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = NO (libfreenect2 not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = YES (License: BSD)
--   With GTSAM                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = NO (cvsba not found)
--   With ZED                  = NO (ZED sdk not found)
--   With RealSense            = NO (librealsense not found)
--   With OCTOMAP              = YES (License: BSD)
--   With Qt4                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mathieu/workspace/rtabmap/build
```

willdzeng, 2016-10-12:
FYI:

-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Found CUDA: /usr/local/cuda (found suitable exact version "7.5") 
-- checking for module 'eigen3'
--   found eigen3, version 3.2.9
-- Found eigen: /usr/local/include/eigen3  
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Found openni: /usr/lib/libOpenNI.so  
-- checking for module 'openni2-dev'
--   package 'openni2-dev' not found
-- Found OpenNI2: /usr/lib/libOpenNI2.so  
-- Found libusb-1.0: /usr/include  
-- checking for module 'flann'
--   found flann, version 1.8.4
-- Found Flann: /usr/lib/x86_64-linux-gnu/libflann_cpp_s.a  
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni2-dev'
--   package 'openni2-dev' not found
-- Found qhull: /usr/lib/x86_64-linux-gnu/libqhull.so  
-- Found PCL_COMMON: /usr/local/lib/libpcl_common.so  
-- Found PCL_OCTREE: /usr/local/lib/libpcl_octree.so  
-- Found PCL_IO: /usr/local/lib/libpcl_io.so  
-- Found PCL_KDTREE: /usr/local/lib/libpcl_kdtree.so  
-- Found PCL_SEARCH: /usr/local/lib/libpcl_search.so  
-- Found PCL_SAMPLE_CONSENSUS: /usr/local/lib/libpcl_sample_consensus.so  
-- Found PCL_FILTERS: /usr/local/lib/libpcl_filters.so  
-- Found PCL_FEATURES: /usr/local/lib/libpcl_features.so  
-- Found PCL_KEYPOINTS: /usr/local/lib/libpcl_keypoints.so  
-- Found PCL_GEOMETRY: /usr/local/include/pcl-1.7  
-- Found PCL_SEGMENTATION: /usr/local/lib/libpcl_segmentation.so  
-- Found PCL_VISUALIZATION: /usr/local/lib/libpcl_visualization.so  
-- Found PCL_OUTOFCORE: /usr/local/lib/libpcl_outofcore.so  
-- Found PCL_SURFACE: /usr/local/lib/libpcl_surface.so  
-- Found PCL_REGISTRATION: /usr/local/lib/libpcl_registration.so  
-- Found PCL_TRACKING: /usr/local/lib/libpcl_tracking.so  
-- Found PCL_RECOGNITION: /usr/local/lib/libpcl_recognition.so  
-- Found PCL_PEOPLE: /usr/local/lib/libpcl_people.so  
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Found OpenMP: -fopenmp  
-- Found OpenMP
-- Found OpenCV: /usr/local/include/opencv;/usr/local/include
-- Found PCL: /usr/local/include/pcl-1.7;/usr/local/include/eigen3;/usr/include;/usr/include/ni;/usr/include/openni2;/usr/include/vtk-5.8
-- Found ZLIB: /usr/include
-- Looking for Q_WS_X11
-- Looking for Q_WS_X11 - found
-- Looking for Q_WS_WIN
-- Looking for Q_WS_WIN - not found
-- Looking for Q_WS_QWS
-- Looking for Q_WS_QWS - not found
-- Looking for Q_WS_MAC
-- Looking for Q_WS_MAC - not found
-- Found Qt4: /usr/bin/qmake-qt4 (found version "4.8.6") 
-- Found QVTK: /usr/lib/libQVTK.so  
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- Found CSPARSE: /usr/include/suitesparse  
-- Found g2o: /usr/local/include;/usr/include/suitesparse;/usr/include/suitesparse
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/local/include/gtsam/3rdparty/Eigen/
-- Found cvsba: /usr/local/include
-- Found ZED sdk: /usr/local/zed/include
-- Found CUDA: /usr/local/cuda (found version "7.5") 
-- Found CUDA: /usr/local/cuda/include
-- Found octomap: /opt/ros/indigo/include
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   Version : 0.11.11
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -Wno-deprecated -fopenmp -std=c++11
--   With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = YES (License: Non commercial)
--   With Freenect             = NO (libfreenect not found)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = NO (libfreenect2 not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = YES (License: BSD)
--   With GTSAM                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = YES (License: GPLv2)
--   With ZED                  = YES (With CUDA)
--   With RealSense            = NO (librealsense not found)
--   With OCTOMAP              = YES (License: BSD)
--   With Qt4                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/zeng/essential/rtabmap/build

matlabbe, 2016-10-12:
I think the ZED sdk is linked to a specified OpenCV binary version too. You may try with `WITH_ZED=OFF`to see if the problem still happen. CVSBA should be rebuilt too if it was built with another version of OpenCV (or set `WITH_CVSBA=OFF`). Having OpenCV linked to Qt5 while rtabmap is linking to Qt4 may cause problems. Normally rtabmap links to Qt5 first [but VTK>5 is required](https://github.com/introlab/rtabmap/blob/403bfc2b44771ac0d60d48ac9eeecac7113463c9/CMakeLists.txt#L197-L204) (which only installed by default on 16.04/Kinetic), so Qt4 is linked instead. To use Qt5, you may need to rebuild VTK6 (with qt support), then PCL for new VTK.

Maybe switching to Ubuntu 16.04 would make the use of OpenCV3 and Qt5 more easier. It looks like that ZED SDK is [now supporting](https://www.stereolabs.com/developers/release/1.1.1a/) Ubuntu 16.04.

cheers

willdzeng, 2016-10-12:
I turned off ZED support and it's working now!!
I am thinking maybe add all those ways of debugging "build" into a tutorial to help people.

That's really good to hear, now I can really switch to Kinect instead of Indigo, which will save a lot effort building opencv_bridge stuff. Thanks for notifying me.

closed 2016-10-12 by willdzeng