introlab/rtabmap #125: Option to automatically save to DB the latest node added to map

matlabbe, 2016-10-12:
This option could be useful if the system crashes for any reason. Right now, if the system crashes, we lose the working memory and the current word vocabulary, making the database not usable anymore.

mantouRobot, 2017-01-07:
u mean that rtabmap save database only in the normal termination?

matlabbe, 2017-01-07:
yes, rtabmap should ends normally so that all data in RAM are saved in the database. rtabmap keeps some stuff in RAM for computing efficiency. This option would add a computing overhead, but would assure that the database is not corrupted on system crash.

matlabbe, 2018-04-04:
Cannot refer to a specified commit, but current version it is done by default. rtabmap-recovery tool (or DatabaseViewer->Recovery) can be used to recover these databases. Note that when opening a corrupted database in RTAB-Map app, you will be asked directly to recover the database for convenience.

closed 2018-04-04 by matlabbe