introlab/rtabmap #126: Compilation fails on Archlinux

TomasBedrnik, 2016-10-13:
Hi,
I'm trying to compile rtabmap on Archlinux, but it fails:
`error: ‘class std::unordered_map<unsigned int, std::vector<unsigned int> >’ has no member named ‘serialize’`
I tried git and 0.11.8 version - both same probem. I have gcc 6.2.1 and boost 1.61.0.

[cmake-output.txt](https://github.com/introlab/rtabmap/files/526874/cmake.txt)
[make-output.txt](https://github.com/introlab/rtabmap/files/526873/make.txt)

I don't know if the problem is in my system or rtabmap has issues with some of new version of libraries in my system.

Thanks
Tomas

matlabbe, 2016-10-13:
The error is FLANN related, which uses `std::unordered_map` objects. Can you tell me if you are able to build standalone [flann-1.8.4-src](http://www.cs.ubc.ca/research/flann/uploads/FLANN/flann-1.8.4-src.zip)? 

In lsh_table.h, there is this [code](https://github.com/introlab/rtabmap/blob/a45aa5d8f7df4e41a27e90ed4285db8c45aba193/corelib/src/rtflann/util/lsh_table.h#L42-L47):
```cpp
// TODO as soon as we use C++0x, use the code in USE_UNORDERED_MAP
#if USE_UNORDERED_MAP
#include <unordered_map>
#else
#include <map>
#endif
```
When rtabmap is building, `USE_UNORDERED_MAP` is not defined, so I guess you could try to set `#define USE_UNORDERED_MAP 1` just before these lines:
```cpp
// TODO as soon as we use C++0x, use the code in USE_UNORDERED_MAP
#define USE_UNORDERED_MAP 1
#if USE_UNORDERED_MAP
#include <unordered_map>
#else
#include <map>
#endif
```

cheers

TomasBedrnik, 2016-10-13:
Thanks for fast reply.

This path is needed to compile flann: http://pkgs.fedoraproject.org/cgit/rpms/flann.git/plain/flann-1.8.4-gcc6.patch and some aditional changes for python bindings.

Unfortunately adding `#define USE_UNORDERED_MAP 1` didn't change anything.

matlabbe, 2016-10-13:
I tried with Fedora and gcc 6.2.1, and with default binaries available (e.g. pcl-devel). Minor this [issue](https://github.com/introlab/rtabmap/issues/127), it builds correctly with this config:
```
-- --------------------------------------------
-- Info :
--   Version : 0.11.11
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp
--   With OpenCV 2 nonfree module (SIFT/SURF) = NO (not found, License: BSD)
--   With Freenect             = NO (libfreenect not found)
--   With OpenNI2              = NO (OpenNI2 not found)
--   With Freenect2            = NO (libfreenect2 not found)
--   With dc1394               = NO (dc1394 not found)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = NO (WITH_G2O=OFF)
--   With GTSAM                = NO (GTSAM not found)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = NO (cvsba not found)
--   With ZED                  = NO (ZED sdk not found)
--   With RealSense            = NO (librealsense not found)
--   With OCTOMAP              = YES (License: BSD)
--   With Qt4                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
```

However, if I build with g2o, the `-std=c++11` flag is added to compilation instructions, and then the build fails with Openni errors like these:
```
/usr/include/ni/XnCppWrapper.h: In constructor ‘xn::Context::Context()’:
/usr/include/ni/XnCppWrapper.h:8658:40: error: class ‘xn::Context’ does not have any field named ‘m_bUsingDeprecatedAPI’
   inline Context() : m_pContext(NULL), m_bUsingDeprecatedAPI(FALSE), m_bAllocated(FALSE), m_hShuttingDownCallback(NULL) {}
                                        ^~~~~~~~~~~~~~~~~~~~~
/usr/include/ni/XnCppWrapper.h:8658:70: error: class ‘xn::Context’ does not have any field named ‘m_bAllocated’
   inline Context() : m_pContext(NULL), m_bUsingDeprecatedAPI(FALSE), m_bAllocated(FALSE), m_hShuttingDownCallback(NULL) {}
                                                                      ^~~~~~~~~~~~
/usr/include/ni/XnCppWrapper.h: In constructor ‘xn::Context::Context(XnContext*)’:
/usr/include/ni/XnCppWrapper.h:8665:59: error: class ‘xn::Context’ does not have any field named ‘m_bUsingDeprecatedAPI’
   inline Context(XnContext* pContext) : m_pContext(NULL), m_bUsingDeprecatedAPI(FALSE), m_bAllocated(FALSE), m_hShuttingDownCallback(NULL)
                                                           ^~~~~~~~~~~~~~~~~~~~~
/usr/include/ni/XnCppWrapper.h:8665:89: error: class ‘xn::Context’ does not have any field named ‘m_bAllocated’
   inline Context(XnContext* pContext) : m_pContext(NULL), m_bUsingDeprecatedAPI(FALSE), m_bAllocated(FALSE), m_hShuttingDownCallback(NULL)
                                                                                         ^~~~~~~~~~~~
/usr/include/ni/XnCppWrapper.h: In copy constructor ‘xn::Context::Context(const xn::Context&)’:
/usr/include/ni/XnCppWrapper.h:8676:60: error: class ‘xn::Context’ does not have any field named ‘m_bUsingDeprecatedAPI’
   inline Context(const Context& other) : m_pContext(NULL), m_bUsingDeprecatedAPI(FALSE), m_bAllocated(FALSE), m_hShuttingDownCallback(NULL)
                                                            ^~~~~~~~~~~~~~~~~~~~~
/usr/include/ni/XnCppWrapper.h:8676:90: error: class ‘xn::Context’ does not have any field named ‘m_bAllocated’
   inline Context(const Context& other) : m_pContext(NULL), m_bUsingDeprecatedAPI(FALSE), m_bAllocated(FALSE), m_hShuttingDownCallback(NULL)
                                                                                          ^~~~~~~~~~~~
/usr/include/ni/XnCppWrapper.h: In static member function ‘static void xn::Context::FreeLicensesList(XnLicense*)’:
/usr/include/ni/XnCppWrapper.h:8955:32: error: ‘xnFreeLicensesList’ was not declared in this scope
    xnFreeLicensesList(aLicenses);
                                ^
/usr/include/ni/XnCppWrapper.h: In member function ‘void xn::Context::UnregisterFromErrorStateChange(XnCallbackHandle)’:
/usr/include/ni/XnCppWrapper.h:9256:64: error: ‘xnUnregisterFromGlobalErrorStateChange’ was not declared in this scope
    xnUnregisterFromGlobalErrorStateChange(m_pContext, hCallback);
                                                                ^
/usr/include/ni/XnCppWrapper.h: In member function ‘void xn::Context::UnregisterFromNodeCreation(XnCallbackHandle)’:
/usr/include/ni/XnCppWrapper.h:9295:81: error: ‘xnUnregisterFromNodeCreation’ was not declared in this scope
    xnUnregisterFromNodeCreation(m_pContext, pCreationCookie->hUnderlyingCallback);
                                                                                 ^
/usr/include/ni/XnCppWrapper.h:9296:28: error: ‘xnOSFree’ was not declared in this scope
    xnOSFree(pCreationCookie);
                            ^
/usr/include/ni/XnCppWrapper.h: In member function ‘void xn::Context::UnregisterFromNodeDestruction(XnCallbackHandle)’:
[... 356 lines cut ...]
/usr/include/pcl-1.8/pcl/io/openni_camera/openni_image.h: In member function ‘void openni_wrapper::Image::fillRaw(unsigned char*) const’:
[... cut ...]
/usr/include/pcl-1.8/pcl/io/openni_camera/openni_image.h:193:23: error: ‘class xn::ImageMetaData’ has no member named ‘FrameID’
     return image_md_->FrameID ();
                       ^~~~~~~
/usr/include/pcl-1.8/pcl/io/openni_camera/openni_image.h: In member function ‘long unsigned int openni_wrapper::Image::getTimeStamp() const’:
/usr/include/pcl-1.8/pcl/io/openni_camera/openni_image.h:199:51: error: ‘class xn::ImageMetaData’ has no member named ‘Timestamp’
     return static_cast<unsigned long> (image_md_->Timestamp ());
```

Did you have these errors too? Did you build PCL from source (along with Openni)?

TomasBedrnik, 2016-10-14:
I have never seen this errors.
I have pcl-1.8.0 and openni-2.2beta2 from source (AUR builds). I disabled most dependencies, tried older gcc-5.4, older boost-1.54, recompiled pcl, disabled openni2... but errors with `serialize` remains same with some new ones.

TomasBedrnik, 2016-10-14:
Openni2 fails to build with gcc6 - needs to be build with clang. But `serialize` errors remains even without openni2.

matlabbe, 2016-10-14:
For an ugly solution, `serialize` related code can be commented, it is not used.

TomasBedrnik, 2016-10-14:
Thanks, that works.
If I figure out how to solve this problem without this ugly hack, I'll let you know.

TomasBedrnik, 2016-11-01:
I don't know what the problem was, but it is working now. Versions of gcc and boost are the same.

TomasBedrnik, 2016-11-19:
I found the probem. It is in current git version of opencv. With 3.1.0 it works OK, but probmem is this commit to opencv:https://github.com/PkLab/opencv/commit/012b20f4e9667fa0329eacc3879207a75bdaa097  associated to this issue: https://github.com/opencv/opencv/issues/7027

It basically adds `#include "opencv2/flann.hpp"` to include/opencv2/opencv.hpp
Opencv includes flann 1.6.10 and this #include probably overlaps system version of flann (last is 1.8.4).

I don't know if this should be solved in opencv (use newer flann) or rtabmap (some cmake magic?)

matlabbe, 2016-11-23:
RTAB-Map uses a slightly patched version of the latest flann library (1.8). The problem is similar to https://github.com/mariusmuja/flann/issues/214

OpenCV is using `#define USE_UNORDERED_MAP 1`, and since the define name is the same as the flann included in rtabmap, USE_UNORDERED_MAP was set to 1, making rtabmap compilation failing with `unordered_map serialization` errors. It seems that even the latest flann code ([this version tested](https://github.com/mariusmuja/flann/tree/06a49513138009d19a1f4e0ace67fbff13270c69)) cannot be built with USE_UNORDERED_MAP=1. The commit above fixes the problem.

Well, OpenCV is using USE_UNORDERED_MAP=1 for a long time, I think that the [commit](https://github.com/opencv/opencv/commit/e42750c63cd4933f654fdf87daf7f6e72496c04c) you referred is making the `lsh_table.h` header file included in rtabmap somehow, setting the define to 1.

cheers,
Mathieu

closed 2016-11-23 by matlabbe