introlab/rtabmap #129: Segmentation fault rtabmap, process exits -11 rtabmap_ros

chiragmajithia, 2016-10-16:
Hello,
I am new to linux and ros in general. I am trying to install rtabmap and rtabmap_ros on my laptop. I used the [source](https://github.com/introlab/rtabmap) to install rtabmap library and stand-alone application. Then I installed rtabmap_ros in a catkin workspace named slam and followed the instruction - I guess properly - to install rtabmap in the workspace/devel folder.
Everything compiled but when I ran the rtabmap_ros with Intel R200, the process died with exit code -11. After reading posts on the [forum](http://official-rtab-map-forum.67519.x6.nabble.com/Rtabmap-error-in-process-rtabmap-rgbd-odometry-1-ubuntu-on-raspberry-pi-td444.html), I tried to run rtabmap standalone application. I was able to set database and select sensor, but as soon as I clicked start, the application crashed with segmentation fault (core dumped).
Then, tried installing new vtk - as during cmake of rtabmap it showed this warning: 
```

> The imported target "vtkRenderingPythonTkWidgets" references the file
>    "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
> but this file does not exist.  Possible reasons include:
> * The file was deleted, renamed, or moved to another location.
> * An install or uninstall procedure did not complete successfully.
> * The installation package was faulty and contained
>    "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
> but not all the files it references.` 

Hence, I found out that this belongs to python-vtk and I installed [this](http://packages.ubuntu.com/search?suite=xenial&arch=amd64&mode=exactfilename&searchon=contents&keywords=libvtkRenderingPythonTkWidgets.so). 
After this installation, the cmake warning stopped and finally I get this when I do cmake ..

> `chirag@thunderbolt:~/vision/rtabmap/build$ cmake ..
> -- Found OpenMP
> -- Found OpenCV: /opt/ros/kinetic/include/opencv;/opt/ros/kinetic/include
> -- Found PCL: /usr/local/include/pcl-1.8;/usr/local/include/eigen3;/usr/include;/usr/include/ni;/usr/include/openni2;/opt/ensenso/development/c/include;/usr/local/include/davidSDK;/usr/local/include/vtk-7.0
> -- Found ZLIB: /usr/include
> -- Found OpenNI2: /usr/include/openni2
> -- Found DC1394: /usr/include/dc1394
> -- Could NOT find CSPARSE (missing:  CSPARSE_INCLUDE_DIR CSPARSE_LIBRARY) 
> -- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/local/include/gtsam/3rdparty/Eigen/
> -- Found cvsba: /usr/local/include
> -- Found RealSense: /opt/ros/kinetic/include
> -- Found octomap: /opt/ros/kinetic/include
> -- Found Pthreads
> -- --------------------------------------------
> -- Info :
> --   Version : 0.11.11
> --   CMAKE_INSTALL_PREFIX = /usr/local
> --   CMAKE_BUILD_TYPE =     Debug
> --   CMAKE_INSTALL_LIBDIR = lib
> --   BUILD_APP =            ON
> --   BUILD_TOOLS =          ON
> --   BUILD_EXAMPLES =       ON
> --   BUILD_SHARED_LIBS =    ON
> --   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++11
> --   With OpenCV 2 nonfree module (SIFT/SURF) = NO (not found, License: BSD)
> --   With Freenect             = NO (libfreenect not found)
> --   With OpenNI2              = YES (License: Apache v2)
> --   With Freenect2            = NO (libfreenect2 not found)
> --   With dc1394               = YES (License: LGPL)
> --   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
> --   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
> --   With g2o                  = NO (g2o not found)
> --   With GTSAM                = YES (License: BSD)
> --   With VERTIGO              = YES (License: GPLv3)
> --   With cvsba                = YES (License: GPLv2)
> --   With ZED                  = NO (ZED sdk not found)
> --   With RealSense            = YES (License: Apache-2)
> --   With OCTOMAP              = YES (License: BSD)
> --   With Qt5                  = YES (License: Open Source or Commercial)
> -- --------------------------------------------
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /home/chirag/vision/rtabmap/build
> 

```
But now when I compile rtabmap, I receive following error and I am not able to revert the changes.

```

> `[  7%] Linking CXX shared library ../../../bin/librtabmap_cored.so
> /usr/bin/ld: cannot find -lvtkftgl
> /usr/bin/ld: cannot find -lvtkIOFFMPEG
> /usr/bin/ld: cannot find -lvtkIOParallelNetCDF
> /usr/bin/ld: cannot find -lvtkParallelMPI
> /usr/bin/ld: cannot find -lvtkFiltersParallelGeometry
> /usr/bin/ld: cannot find -lvtkIOMPIImage
> /usr/bin/ld: cannot find -lvtkGUISupportQtWebkit
> /usr/bin/ld: cannot find -lvtkViewsQt
> /usr/bin/ld: cannot find -lvtkGUISupportQt
> /usr/bin/ld: cannot find -lvtkRenderingOpenGL
> /usr/bin/ld: cannot find -lvtkViewsGeovis
> /usr/bin/ld: cannot find -lvtkTestingIOSQL
> /usr/bin/ld: cannot find -lvtkIOMySQL
> /usr/bin/ld: cannot find -lvtkIOMPIParallel
> /usr/bin/ld: cannot find -lvtkIOXdmf2
> /usr/bin/ld: cannot find -lvtkxdmf2
> /usr/bin/ld: cannot find -lvtkRenderingFreeTypeOpenGL
> /usr/bin/ld: cannot find -lvtkRenderingGL2PS
> /usr/bin/ld: cannot find -lvtkRenderingContextOpenGL
> /usr/bin/ld: cannot find -lvtkTestingRendering
> /usr/bin/ld: cannot find -lvtkIOVPIC
> /usr/bin/ld: cannot find -lVPIC
> /usr/bin/ld: cannot find -lvtkIOPostgreSQL
> /usr/bin/ld: cannot find -lvtkFiltersReebGraph
> /usr/bin/ld: cannot find -lvtkRenderingVolumeOpenGL
> /usr/bin/ld: cannot find -lvtkRenderingMatplotlib
> /usr/bin/ld: cannot find -lvtkPythonInterpreter
> /usr/bin/ld: cannot find -lvtkWrappingPythonCore
> /usr/bin/ld: cannot find -lvtkWrappingTools
> /usr/bin/ld: cannot find -lvtkRenderingQt
> /usr/bin/ld: cannot find -lvtkIOParallelExodus
> /usr/bin/ld: cannot find -lvtkIOGDAL
> /usr/bin/ld: cannot find -lvtkGUISupportQtOpenGL
> /usr/bin/ld: cannot find -lvtkIOGeoJSON
> /usr/bin/ld: cannot find -lvtkFiltersPython
> /usr/bin/ld: cannot find -lvtkIOODBC
> /usr/bin/ld: cannot find -lvtkLocalExample
> /usr/bin/ld: cannot find -lvtkFiltersParallelStatistics
> /usr/bin/ld: cannot find -lvtkFiltersParallelFlowPaths
> /usr/bin/ld: cannot find -lvtkRenderingParallel
> /usr/bin/ld: cannot find -lvtkRenderingExternal
> /usr/bin/ld: cannot find -lvtkRenderingParallelLIC
> /usr/bin/ld: cannot find -lvtkRenderingLIC
> /usr/bin/ld: cannot find -lvtkInfovisBoostGraphAlgorithms
> /usr/bin/ld: cannot find -lvtkTestingGenericBridge
> /usr/bin/ld: cannot find -lvtkFiltersParallelMPI
> /usr/bin/ld: cannot find -lvtkParallelMPI4Py
> /usr/bin/ld: cannot find -lvtkRenderingFreeTypeFontConfig
> /usr/bin/ld: cannot find -lvtkGUISupportQtSQL
> /usr/bin/ld: cannot find -lvtkWrappingJava
> /usr/bin/ld: cannot find -lvtkIOParallelLSDyna
> collect2: error: ld returned 1 exit status
> corelib/src/CMakeFiles/rtabmap_core.dir/build.make:2152: recipe for target '../bin/librtabmap_cored.so.0.11.11' failed
> make[2]: *** [../bin/librtabmap_cored.so.0.11.11] Error 1
> CMakeFiles/Makefile2:267: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
> make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
> Makefile:149: recipe for target 'all' failed
> make: *** [all] Error 2

```
/`
Could anyone please guide me on how to get this running.
Thanks,
Chirag

willdzeng, 2016-10-16:
From Mathieu:"To use Qt5, you may need to rebuild VTK6 (with qt support), then PCL for new VTK."
#124

chiragmajithia, 2016-10-16:
I installed VTK and QT5 using [this instructions](http://www.vtk.org/Wiki/VTK/Building/Linux). I downloaded VTK6.3 release from [here](http://www.vtk.org/download/).
using Qt5 setup -- 

> `cd /path/to/VTK-Release-build
> cmake -DVTK_QT_VERSION:STRING=5 \
>       -DQT_QMAKE_EXECUTABLE:PATH=/path/to/qt5.2.1-install/5.2.1/gcc_64/bin/qmake \
>       -DVTK_Group_Qt:BOOL=ON \
>       -DCMAKE_PREFIX_PATH:PATH=/path/to/qt.5.2.1-install/5.2.1/gcc_64/lib/cmake  \
>       -DBUILD_SHARED_LIBS:BOOL=ON
>       /path/to/VTK`

willdzeng, 2016-10-16:
Which PCL are you using? it only support 1.7.3 I remember

chiragmajithia, 2016-10-16:
I have pcl-1.8,  but the package compiled successfully with it. If it compiles with pcl-1.8. should it not run with it?

chiragmajithia, 2016-10-17:
After reinstalling openni_tools, opencv2.4.3 and vtk with qt5 - following the same instructions from previous comment -- the rtabmap and rtabmap_ros compiles sucessfully. 
1. While running  
`cmake -DCMAKE_BUILD_TYPE=Release .. -DCMAKE_INSTALL_PREFIX=~/vision/SLAM/devel/`

> I get following output:
> chirag@thunderbolt:~/vision/rtabmap/build$ cmake -DCMAKE_BUILD_TYPE=Release .. -DCMAKE_INSTALL_PREFIX=~/vision/SLAM/devel/
> -- The imported target "vtkRenderingPythonTkWidgets" references the file
>    "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
> but this file does not exist.  Possible reasons include:
> * The file was deleted, renamed, or moved to another location.
> * An install or uninstall procedure did not complete successfully.
> * The installation package was faulty and contained
>    "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
> but not all the files it references.
> 
> -- Found OpenMP
> -- Found OpenCV: /opt/ros/kinetic/include/opencv-3.1.0-dev/opencv;/opt/ros/kinetic/include/opencv-3.1.0-dev
> -- Found PCL: /usr/local/include/pcl-1.8;/usr/local/include/eigen3;/usr/include;/usr/include/ni;/usr/include/openni2;/opt/ensenso/development/c/include;/usr/local/include/davidSDK;/usr/include/vtk-6.2;/usr/include/freetype2;/usr/include/x86_64-linux-gnu/freetype2;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include;/usr/lib/openmpi/include;/usr/lib/openmpi/include/openmpi;/usr/include/hdf5/openmpi;/usr/include/x86_64-linux-gnu;/usr/include/jsoncpp;/usr/include/libxml2;/usr/include/python2.7;/usr/include/tcl
> -- Found ZLIB: /usr/include
> -- Found Freenect: /usr/include
> -- Found OpenNI2: /usr/include/openni2
> -- Found DC1394: /usr/include/dc1394
> -- Could NOT find CSPARSE (missing:  CSPARSE_INCLUDE_DIR CSPARSE_LIBRARY) 
> -- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/local/include/gtsam/3rdparty/Eigen/
> -- Found cvsba: /usr/local/include
> -- Found RealSense: /opt/ros/kinetic/include
> -- Found octomap: /opt/ros/kinetic/include
> -- Found Pthreads
> -- --------------------------------------------
> -- Info :
> --   Version : 0.11.11
> --   CMAKE_INSTALL_PREFIX = /home/chirag/vision/SLAM/devel
> --   CMAKE_BUILD_TYPE =     Release
> --   CMAKE_INSTALL_LIBDIR = lib
> --   BUILD_APP =            ON
> --   BUILD_TOOLS =          ON
> --   BUILD_EXAMPLES =       ON
> --   BUILD_SHARED_LIBS =    ON
> --   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++11
> --   With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = YES (License: Non commercial)
> --   With Freenect             = YES (License: Apache v2 and/or GPLv2)
> --   With OpenNI2              = YES (License: Apache v2)
> --   With Freenect2            = NO (libfreenect2 not found)
> --   With dc1394               = YES (License: LGPL)
> --   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
> --   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
> --   With g2o                  = NO (g2o not found)
> --   With GTSAM                = YES (License: BSD)
> --   With VERTIGO              = YES (License: GPLv3)
> --   With cvsba                = YES (License: GPLv2)
> --   With ZED                  = NO (ZED sdk not found)
> --   With RealSense            = YES (License: Apache-2)
> --   With OCTOMAP              = YES (License: BSD)
> --   With Qt5                  = YES (License: Open Source or Commercial)
> -- --------------------------------------------
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /home/chirag/vision/rtabmap/build
> 

2. rtabmap (standalone) crashes with Intel R200 as soon as we start mapping.

willdzeng, 2016-10-17:
https://raw.githubusercontent.com/introlab/rtabmap_ros/master/README.md
"RTAB-Map works only with the PCL 1.7, which is the default version installed with ROS Hydro/Indigo/Jade/Kinetic (Fuerte and Groovy are not"

willdzeng, 2016-10-17:
@matlabbe maybe consider adding PCL1.7 Required in cmake?

matlabbe, 2016-10-17:
I updated the comment, it should work with PCL >=1.7.

matlabbe, 2016-10-17:
They may have updated the RealSense library, which would have break the r200 support in rtabmap. I'll check tomorrow.

chiragmajithia, 2016-10-17:
Thank you for your reply @matlabbe.
The file contains video of how rtabmap (standalone) is crashing - just in case it helps to figure out if there is something wrong with my installation.
[rtabmap.mp4.tar.gz](https://github.com/introlab/rtabmap/files/532332/rtabmap.mp4.tar.gz)

matlabbe, 2016-10-17:
Cannot reproduce the problem on Ubuntu 16.04/Kinetic after building `rtabmap`/`rtabmap_ros` from source with `ros-kinetic-librealsense` binary package.

standalone:
[image]

[From RGB-D hand-held tutorial with R200](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping):
[image]

Small observation: The registered depth image from `realsense_camera` has a lot of holes comparatively to the depth image generated in the standalone. There are maybe parameters to fill these holes on the `realsense_camera` side. Otherwise, you could create a nodelet to fill these holes, similarly to [what is done](https://github.com/introlab/rtabmap/blob/1853cfde9b5d7ea1697c5eb2fd4cf076b0f74fc5/corelib/src/CameraRGBD.cpp#L2145-L2149) in rtabmap standalone realsense driver.

cheers

matlabbe, 2016-10-17:
Note that I reproduced a similar error when building against the ZED SDK. To debug on your side, launch rtabmap with gdb:
```bash
$ gdb rtabmap
# type "run"
```
When rtabmap is opened, go in Preferences->Logging and set logger level to Debug. Start detection. When crashing, type `bt` (for backtrace) in the gdb console. Copy/paste here the whole terminal output.

On my side, if I build without ZED SDK (v1.1.1), the problem disappears.

chiragmajithia, 2016-10-17:
Hey,
Thank you for your attempts to debug the issue.
I am new to using ROS and gdb -- Following is the output you asked:
```
Thread 19 "rtabmap" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff8bfff700 (LWP 7927)]
Eigen::internal::dense_assignment_loop<Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<float, -1, 1, 0, -1, 1> >, Eigen::internal::evaluator<Eigen::Matrix<float, -1, 1, 0, -1, 1> >, Eigen::internal::assign_op<float>, 0>, 3, 0>::run (kernel=<synthetic pointer>)
    at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:386
386	      kernel.template assignPacket<dstAlignment, srcAlignment, PacketType>(index);
(gdb) bt
#0  Eigen::internal::dense_assignment_loop<Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<float, -1, 1, 0, -1, 1> >, Eigen::internal::evaluator<Eigen::Matrix<float, -1, 1, 0, -1, 1> >, Eigen::internal::assign_op<float>, 0>, 3, 0>::run (kernel=<synthetic pointer>)
    at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:386
#1  Eigen::internal::call_dense_assignment_loop<Eigen::Matrix<float, -1, 1, 0, -1, 1>, Eigen::Matrix<float, -1, 1, 0, -1, 1>, Eigen::internal::assign_op<float> > (func=..., src=..., dst=...)
    at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:653
#2  Eigen::internal::Assignment<Eigen::Matrix<float, -1, 1, 0, -1, 1>, Eigen::Matrix<float, -1, 1, 0, -1, 1>, Eigen::internal::assign_op<float>, Eigen::internal::Dense2Dense, float>::run (func=..., src=..., dst=...)
    at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:790
#3  Eigen::internal::call_assignment_no_alias<Eigen::Matrix<float, -1, 1, 0, -1, 1>, Eigen::Matrix<float, -1, 1, 0, -1, 1>, Eigen::internal::assign_op<float> >
    (dst=..., src=..., func=...)
    at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:747
#4  0x00007fffed6ff2dc in Eigen::internal::call_assignment<Eigen::Matrix<float, -1, 1, 0, -1, 1>, Eigen::Matrix<float, -1, 1, 0, -1, 1>, Eigen::internal::assign_op<float> >(Eigen::Matrix<float, -1, 1, 0, -1, 1>&, Eigen::Matrix<float, -1, 1, 0, -1, 1> const&, Eigen::internal::assign_op<float> const&, Eigen::internal::enable_if<!Eigen::internal::evaluator_assume_aliasing<Eigen::Matrix<float, -1, 1, 0, -1, 1>, Eigen::internal::evaluator_traits<Eigen::Matrix<float, -1, 1, 0, -1, ---Typ---Typ---Type <ret---Typ---Typ---Typ---Type <ret---Type <ret---Type <return> t---Type <return> t---Typ---Typ---Type <ret---Typ---Type <return> t---Typ---Typ---Typ---Typ---Typ---Typ---Typ---Typ---Typ---Typ---Typ---Typ---Typ---Typ---Typ---Typ---Type <return> to continue, or q <return> to quit---
1> >::Shape>::value, void*>::type) (func=..., src=..., dst=...)
    at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:712
#5  Eigen::internal::call_assignment<Eigen::Matrix<float, -1, 1, 0, -1, 1>, Eigen::Matrix<float, -1, 1, 0, -1, 1> > (
    src=..., dst=...) at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:693
#6  Eigen::PlainObjectBase<Eigen::Matrix<float, -1, 1, 0, -1, 1> >::_set<Eigen::Matrix<float, -1, 1, 0, -1, 1> > (
    other=..., this=0x7fff8bfefc78) at /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:682
#7  Eigen::Matrix<float, -1, 1, 0, -1, 1>::operator= (other=..., this=0x7fff8bfefc78)
    at /usr/include/eigen3/Eigen/src/Core/Matrix.h:208
#8  pcl::RandomSampleConsensus<pcl::PointXYZ>::computeModel (this=0x7fff8bfefc30)
    at /home/chirag/vision/src/pcl/sample_consensus/include/pcl/sample_consensus/impl/ransac.hpp:106
#9  0x00007ffff728a744 in rtabmap::util3d::transformFromXYZCorrespondences(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZ> const> const&, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZ> const> const&, double, int, int, double, std::vector<int, std::allocator<int> >*, double*) () from /home/chirag/vision/SLAM/devel/lib/librtabmap_core.so.0.11
#10 0x00007ffff72da7ca in rtabmap::util3d::estimateMotion3DTo3D(std::map<int, cv::Point3_<float>, std::less<int>, std::allocator<std::pair<int const, cv::Point3_<float> > > > const&, std::map<int, cv::Point3_<float>, std::less<int>, std::allocator<std::pair<int const, cv::Point3_<float> > > > const&, int, double, int, int, double*, std::vector<int, std::allocator<int> >*, std::vector<int, std::allocator<int> >*) () from /home/chirag/vision/SLAM/devel/lib/librtabmap_core.so.0.11
#11 0x00007ffff73553c1 in rtabmap::RegistrationVis::computeTransformationImpl(rtabmap::Signature&, rtabmap::Signature&, rtabmap::Transform, rtabmap::RegistrationInfo&) const () from /home/chirag/vision/SLAM/devel/lib/librtabmap_core.so.0.11
#12 0x00007ffff7340bd0 in rtabmap::Registration::computeTransformationMod(rtabmap::Signature&, rtabmap::Signature&, rtabmap::Transform, rtabmap::RegistrationInfo*) const () from /home/chirag/vision/SLAM/devel/lib/librtabmap_core.so.0.11
#13 0x00007ffff739b9b5 in rtabmap::OdometryF2M::computeTransform(rtabmap::SensorData&, rtabmap::Transform const&, rtabmap::OdometryInfo*) () from /home/chirag/vision/SLAM/devel/lib/librtabmap_core.so.0.11
#14 0x00007ffff73870f0 in rtabmap::Odometry::process(rtabmap::SensorData&, rtabmap::Transform const&, rtabmap::OdometryInfo*) () from /home/chirag/vision/SLAM/devel/lib/librtabmap_core.so.0.11
#15 0x00007ffff738b1e7 in rtabmap::Odometry::process(rtabmap::SensorData&, rtabmap::OdometryInfo*) ()
   from /home/chirag/vision/SLAM/devel/lib/librtabmap_core.so.0.11
#16 0x00007ffff73996b2 in rtabmap::OdometryThread::mainLoop() ()
[... cut ...]
   from /home/chirag/vision/SLAM/devel/lib/librtabmap_utilite.so.0.11
#18 0x00007ffff54d16fa in start_thread (arg=0x7fff8bfff700) at pthread_create.c:333
#19 0x00007ffff59f1b5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

```

Please do let me know if you need any other input from my end! I am going to try your previous suggestion - but I really doubt it would resolve the issue. 
I also have another system with me, I'll try and install rtabmap on it and see if the problem is reproduced in it.

Thank you again!
Regards,
Chirag

willdzeng, 2016-10-18:
Can you post your cmake print also, Chirag?

chiragmajithia, 2016-10-19:
Please find the cmake print -- 
> chirag@thunderbolt:~/vision/rtabmap/build$ cmake -DCMAKE_BUILD_TYPE=Release .. -DCMAKE_INSTALL_PREFIX=~/vision/SLAM/devel
> -- The C compiler identification is GNU 5.4.0
> -- The CXX compiler identification is GNU 5.4.0
> -- Check for working C compiler: /usr/bin/cc
> -- Check for working C compiler: /usr/bin/cc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: /usr/bin/c++
> -- Check for working CXX compiler: /usr/bin/c++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Checking for module 'eigen3'
> --   Found eigen3, version 3.2.10
> -- Found eigen: /usr/local/include/eigen3  
> -- Checking for module 'flann'
> --   Found flann, version 1.8.4
> -- Found Flann: /usr/lib/x86_64-linux-gnu/libflann_cpp_s.a  
> -- Checking for module 'libopenni'
> --   Found libopenni, version 1.5.4.0
> -- Found openni: /usr/lib/libOpenNI.so  
> -- The imported target "vtkRenderingPythonTkWidgets" references the file
>    "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
> but this file does not exist.  Possible reasons include:
> * The file was deleted, renamed, or moved to another location.
> * An install or uninstall procedure did not complete successfully.
> * The installation package was faulty and contained
>    "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
> but not all the files it references.
> 
> -- Found libusb-1.0: /usr/include  
> -- Found qhull: /usr/lib/x86_64-linux-gnu/libqhull.so  
> -- Found PCL_COMMON: /usr/local/lib/libpcl_common.so  
> -- Found PCL_KDTREE: /usr/local/lib/libpcl_kdtree.so  
> -- Found PCL_OCTREE: /usr/local/lib/libpcl_octree.so  
> -- Found PCL_SEARCH: /usr/local/lib/libpcl_search.so  
> -- Found PCL_SAMPLE_CONSENSUS: /usr/local/lib/libpcl_sample_consensus.so  
> -- Found PCL_FILTERS: /usr/local/lib/libpcl_filters.so  
> -- Found PCL_IO: /usr/local/lib/libpcl_io.so  
> -- Found PCL_2D: /usr/local/include/pcl-1.8  
> -- Found PCL_FEATURES: /usr/local/lib/libpcl_features.so  
> -- Found PCL_KEYPOINTS: /usr/local/lib/libpcl_keypoints.so  
> -- Found PCL_GEOMETRY: /usr/local/include/pcl-1.8  
> -- Found PCL_ML: /usr/local/lib/libpcl_ml.so  
> -- Found PCL_SEGMENTATION: /usr/local/lib/libpcl_segmentation.so  
> -- Found PCL_VISUALIZATION: /usr/local/lib/libpcl_visualization.so  
> -- Found PCL_OUTOFCORE: /usr/local/lib/libpcl_outofcore.so  
> -- Found PCL_STEREO: /usr/local/lib/libpcl_stereo.so  
> -- Found PCL_TRACKING: /usr/local/lib/libpcl_tracking.so  
> -- Found PCL_SURFACE: /usr/local/lib/libpcl_surface.so  
> -- Found PCL_PEOPLE: /usr/local/lib/libpcl_people.so  
> -- Found PCL_REGISTRATION: /usr/local/lib/libpcl_registration.so  
> -- Found PCL_RECOGNITION: /usr/local/lib/libpcl_recognition.so  
> -- Try OpenMP C flag = [-fopenmp]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Success
> -- Try OpenMP CXX flag = [-fopenmp]
> -- Performing Test OpenMP_FLAG_DETECTED
> -- Performing Test OpenMP_FLAG_DETECTED - Success
> -- Found OpenMP: -fopenmp  
> -- Found OpenMP
> -- Found OpenCV: /opt/ros/kinetic/include/opencv-3.1.0-dev/opencv;/opt/ros/kinetic/include/opencv-3.1.0-dev
> -- Found PCL: /usr/local/include/pcl-1.8;/usr/local/include/eigen3;/usr/include;/usr/include/ni;/usr/include/openni2;/opt/ensenso/development/c/include;/usr/local/include/davidSDK;/usr/include/vtk-6.2;/usr/include/freetype2;/usr/include/x86_64-linux-gnu/freetype2;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include;/usr/lib/openmpi/include;/usr/lib/openmpi/include/openmpi;/usr/include/hdf5/openmpi;/usr/include/x86_64-linux-gnu;/usr/include/jsoncpp;/usr/include/libxml2;/usr/include/python2.7;/usr/include/tcl
> -- Found ZLIB: /usr/include
> -- Found Freenect: /usr/include
> -- Found OpenNI2: /usr/include/openni2
[... 18 lines cut ...]
> --   BUILD_APP =            ON
> --   BUILD_TOOLS =          ON
> --   BUILD_EXAMPLES =       ON
> --   BUILD_SHARED_LIBS =    ON
> --   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++11
> --   With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = YES (License: Non commercial)
> --   With Freenect             = YES (License: Apache v2 and/or GPLv2)
> --   With OpenNI2              = YES (License: Apache v2)
> --   With Freenect2            = NO (libfreenect2 not found)
> --   With dc1394               = YES (License: LGPL)
> --   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
> --   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
> --   With g2o                  = YES (License: BSD)
> --   With GTSAM                = YES (License: BSD)
> --   With VERTIGO              = YES (License: GPLv3)
> --   With cvsba                = YES (License: GPLv2)
> --   With ZED                  = NO (ZED sdk not found)
> --   With RealSense            = YES (License: Apache-2)
> --   With OCTOMAP              = YES (License: BSD)
> --   With Qt5                  = YES (License: Open Source or Commercial)
> -- --------------------------------------------
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /home/chirag/vision/rtabmap/build
>

matlabbe, 2016-10-23:
Did you try with PCL binaries? (I think it is the 1.7.2 version)

chiragmajithia, 2016-10-28:
I had an exam so I was not able work on it, I ll let you know the results by tomorrow with PCL binaries.

evanbeard, 2016-11-30:
Having the same issue (i.e. a segfault at the same exact line above) when hitting "start" in rtabmap (without ros), both with the latest PCL libraries in APT (ubuntu 16.04) and PCL 1.7.2 built from source.

@matlabbe if you can't reproduce could you post the steps you use to build on Ubuntu 16.04?

matlabbe, 2016-11-30:
Which sensor are you using? You can try setting the logger level to Debug in Preferences->Logging to see some logs on terminal before it crashes. 

On my 16.04 machine, only libfreenect2 and GTSAM are built from source. For other dependencies, use apt-get. Look at my cmake status msg for packages installed from ros kinetic:
```bash
cmake ..
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- Could NOT find PCL_APPS (missing:  PCL_APPS_LIBRARY) 
-- Found OpenMP
-- Found OpenCV: /opt/ros/kinetic/include/opencv-3.1.0-dev/opencv;/opt/ros/kinetic/include/opencv-3.1.0-dev
-- Found PCL: /usr/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/openni2;/usr/include/vtk-6.2;/usr/include/x86_64-linux-gnu;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include;/usr/lib/openmpi/include;/usr/lib/openmpi/include/openmpi;/usr/include/jsoncpp;/usr/include/freetype2;/usr/include/x86_64-linux-gnu/freetype2;/usr/include/hdf5/openmpi;/usr/include/python2.7;/usr/include/tcl;/usr/include/libxml2
-- Found ZLIB: /usr/include
-- Found Freenect: /opt/ros/kinetic/include/libfreenect
-- Found freenect2: /usr/local/include
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- Found g2o: /opt/ros/kinetic/include;/usr/include/suitesparse;/usr/include/suitesparse
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/local/include/gtsam/3rdparty/Eigen/
-- Found RealSense: /opt/ros/kinetic/include
-- Found octomap: /opt/ros/kinetic/include
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   Version : 0.11.12
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++11
--   With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = YES (License: Non commercial)
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = YES (License: Apache v2 and/or GPLv2)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = YES (License: BSD)
--   With GTSAM                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = NO (WITH_CVSBA=OFF)
--   With ZED                  = NO (WITH_ZED=OFF)
--   With RealSense            = YES (License: Apache-2)
--   With OCTOMAP              = YES (License: BSD)
--   With Qt5                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mathieu/workspace/rtabmap/build
```

evanbeard, 2016-11-30:
R200 sensor - same error sequence as in the video earlier in this thread. I just set logging to debug, here is the last bit:

[DEBUG] (2016-11-30 15:38:25.105) RegistrationVis.cpp:193::computeTransformationImpl() Vis/CorFlowMaxLevel=3
[DEBUG] (2016-11-30 15:38:25.105) RegistrationVis.cpp:194::computeTransformationImpl() guess=xyz=[null] rpy=[null]
[DEBUG] (2016-11-30 15:38:25.105) RegistrationVis.cpp:202::computeTransformationImpl() Input(-1): from=940 words, 940 3D words, 940 words descriptors,  0 kpts, 0 descriptors
[DEBUG] (2016-11-30 15:38:25.105) RegistrationVis.cpp:210::computeTransformationImpl() Input(3): to=0 words, 0 3D words, 0 words descriptors, 0 kpts, 0 descriptors
[DEBUG] (2016-11-30 15:38:25.105) RegistrationVis.cpp:229::computeTransformationImpl() 
[DEBUG] (2016-11-30 15:38:25.105) RegistrationVis.cpp:446::computeTransformationImpl() 
[DEBUG] (2016-11-30 15:38:25.107) MainWindow.cpp:2820::createAndAddCloudToMap() 
[DEBUG] (2016-11-30 15:38:25.107) MainWindow.cpp:2249::updateMapCloud() 
[DEBUG] (2016-11-30 15:38:25.107) MainWindow.cpp:2298::updateMapCloud() labels.size()=1
[DEBUG] (2016-11-30 15:38:25.107) MainWindow.cpp:2324::updateMapCloud() 
[DEBUG] (2016-11-30 15:38:25.107) MainWindow.cpp:2383::updateMapCloud() 
[DEBUG] (2016-11-30 15:38:25.107) MainWindow.cpp:2387::updateMapCloud() 
[DEBUG] (2016-11-30 15:38:25.107) CloudViewer.cpp:782::removeOctomap() 
[DEBUG] (2016-11-30 15:38:25.108) MainWindow.cpp:2476::updateMapCloud() 
[DEBUG] (2016-11-30 15:38:25.108) CloudViewer.cpp:390::updateCloudPose() Updating pose featuresOdom to xyz=0.000000,0.000000,0.000000 rpy=0.000000,-0.000000,0.000000
[DEBUG] (2016-11-30 15:38:25.108) MainWindow.cpp:2482::updateMapCloud() 
[DEBUG] (2016-11-30 15:38:25.108) MainWindow.cpp:2484::updateMapCloud() 
[DEBUG] (2016-11-30 15:38:25.108) MainWindow.cpp:1740::processStats() time= 7 ms
[DEBUG] (2016-11-30 15:38:25.108) util2d.cpp:1142::computeRoi() roi ratios = 0.000000, 0.000000, 0.000000, 0.000000
[DEBUG] (2016-11-30 15:38:25.108) util2d.cpp:1143::computeRoi() roi = 0, 0, 640, 480
[DEBUG] (2016-11-30 15:38:25.108) util2d.cpp:1168::computeRoi() roi = 0, 0, 640, 480
[DEBUG] (2016-11-30 15:38:25.108) MainWindow.cpp:1867::processStats() time= 0 ms
[DEBUG] (2016-11-30 15:38:25.108) MainWindow.cpp:1891::processStats() 
[DEBUG] (2016-11-30 15:38:25.108) MainWindow.cpp:1907::processStats() 
[ INFO] (2016-11-30 15:38:25.108) MainWindow.cpp:1919::processStats() Updating GUI time = 0.011000s
[DEBUG] (2016-11-30 15:38:25.109) MainWindow.cpp:877::processOdometry() 
[DEBUG] (2016-11-30 15:38:25.109) MainWindow.cpp:901::processOdometry() odom lost
[DEBUG] (2016-11-30 15:38:25.115) Features2d.cpp:476::generateKeypoints() Keypoints extraction time = 0.006779 s, keypoints extracted = 1000 (mask empty=0)
[DEBUG] (2016-11-30 15:38:25.115) RegistrationVis.cpp:486::computeTransformationImpl() kptsFrom=940
[DEBUG] (2016-11-30 15:38:25.115) RegistrationVis.cpp:487::computeTransformationImpl() kptsTo=1000
[DEBUG] (2016-11-30 15:38:25.117) Features2d.cpp:517::generateDescriptors() Descriptors extracted = 927, remaining kpts=927
[DEBUG] (2016-11-30 15:38:25.118) RegistrationVis.cpp:643::computeTransformationImpl() descriptorsFrom=940
[DEBUG] (2016-11-30 15:38:25.118) RegistrationVis.cpp:644::computeTransformationImpl() descriptorsTo=927
[DEBUG] (2016-11-30 15:38:25.118) RegistrationVis.cpp:889::computeTransformationImpl() 
[DEBUG] (2016-11-30 15:38:25.118) VWDictionary.cpp:302::update() 
[DEBUG] (2016-11-30 15:38:25.118) VWDictionary.cpp:502::update() Dictionary updated! (size=0 added=0 removed=0)
[DEBUG] (2016-11-30 15:38:25.118) VWDictionary.cpp:510::update() 
[DEBUG] (2016-11-30 15:38:25.118) VWDictionary.cpp:302::update() 
[DEBUG] (2016-11-30 15:38:25.118) VWDictionary.cpp:317::update() Incremental FLANN: Removing 0 words...
[DEBUG] (2016-11-30 15:38:25.118) VWDictionary.cpp:326::update() Incremental FLANN: Removing 0 words... done!
[DEBUG] (2016-11-30 15:38:25.118) VWDictionary.cpp:330::update() Incremental FLANN: Inserting 940 words...
[DEBUG] (2016-11-30 15:38:25.118) VWDictionary.cpp:357::update() Building FLANN index...
[DEBUG] (2016-11-30 15:38:25.118) VWDictionary.cpp:375::update() Building FLANN index... done!
[DEBUG] (2016-11-30 15:38:25.121) VWDictionary.cpp:389::update() Incremental FLANN: Inserting 940 words... done!
[DEBUG] (2016-11-30 15:38:25.121) VWDictionary.cpp:502::update() Dictionary updated! (size=0 added=940 removed=0)
[DEBUG] (2016-11-30 15:38:25.121) VWDictionary.cpp:510::update() 
[DEBUG] (2016-11-30 15:38:25.121) VWDictionary.cpp:588::addNewWords() id=2 descriptors=927
[DEBUG] (2016-11-30 15:38:25.121) VWDictionary.cpp:681::addNewWords() newPts.total()=927 
[DEBUG] (2016-11-30 15:38:25.121) Camera.cpp:100::takeImage() Time capturing image = 0.016465s
[DEBUG] (2016-11-30 15:38:25.121) CameraThread.cpp:128::mainLoop() 
[DEBUG] (2016-11-30 15:38:25.121) OdometryThread.cpp:136::addData() Data buffer is full, the oldest data is removed to add the new one.
[DEBUG] (2016-11-30 15:38:25.127) VWDictionary.cpp:746::addNewWords() Time to find nn = 0.006234 s
[DEBUG] (2016-11-30 15:38:25.128) VWDictionary.cpp:867::addNewWords() naive search and add ref/words time = 0.000595 s
[DEBUG] (2016-11-30 15:38:25.128) VWDictionary.cpp:869::addNewWords() 773 new words added...
[DEBUG] (2016-11-30 15:38:25.128) VWDictionary.cpp:871::addNewWords() 154 duplicated words added (from current image = 0)...
[... cut ...]
[DEBUG] (2016-11-30 15:38:25.128) VWDictionary.cpp:515::clear() 
[DEBUG] (2016-11-30 15:38:25.129) VWDictionary.cpp:515::clear() 
[DEBUG] (2016-11-30 15:38:25.130) RegistrationVis.cpp:1127::computeTransformationImpl() 
[DEBUG] (2016-11-30 15:38:25.130) util3d_motion_estimation.cpp:223::estimateMotion3DTo3D() Unique correspondences = 99
[DEBUG] (2016-11-30 15:38:25.130) util3d_registration.cpp:83::transformFromXYZCorrespondences() iterations=100 inlierThreshold=0.100000
Segmentation fault (core dumped)

evanbeard, 2016-11-30:
Error is coming from Eigen and I had to install the latest version of that from source to avoid this message when running make (even after installing eigen from apt)

``/usr/local/include/pcl-1.7/pcl/impl/point_types.hpp:46:22: fatal error: Eigen/Core: No such file or directory

maybe the issue is with the latest version of Eigen (3.3)

matlabbe, 2016-11-30:
In your log, the application is crashing in `util3d_registration.cpp:83::transformFromXYZCorrespondences()`, which is exactly the same problem as https://github.com/introlab/rtabmap/issues/130 which seems indeed related to different Eigen versions used by RTAB-Map's dependencies.