introlab/rtabmap #130: Segementation fault in transformFromXYZCorrespondences() when building with ZED SDK support

matlabbe, 2016-10-17:
gdb output (with rtabmap built in Debug, Ubuntu 16.04/Kinectic, ZED 1.1.1 with libraries linked to Kinectic OpenCV3.1 libraries):
```bash
[DEBUG] (2016-10-17 17:07:11.869) RegistrationVis.cpp:1097::computeTransformationImpl() 
[DEBUG] (2016-10-17 17:07:11.875) util3d_motion_estimation.cpp:225::estimateMotion3DTo3D() Unique correspondences = 462
[DEBUG] (2016-10-17 17:07:11.875) util3d_registration.cpp:83::transformFromXYZCorrespondences() iterations=100 inlierThreshold=0.100000

Thread 19 "rtabmap" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff7dffb700 (LWP 12315)]
0x00007ffff69d94b9 in _mm256_load_ps (__P=0x7fff6c2004b0) at /usr/lib/gcc/x86_64-linux-gnu/5/include/avxintrin.h:848
848	  return *(__m256 *)__P;
(gdb) bt
#0  0x00007ffff69d94b9 in _mm256_load_ps (__P=0x7fff6c2004b0) at /usr/lib/gcc/x86_64-linux-gnu/5/include/avxintrin.h:848
#1  Eigen::internal::pload<float __vector(8)>(Eigen::internal::unpacket_traits<float __vector(8)>::type const*) (from=0x7fff6c2004b0)
    at /usr/include/eigen3/Eigen/src/Core/arch/AVX/PacketMath.h:206
#2  0x00007ffff6aa8349 in Eigen::internal::ploadt<float __vector(8), 32>(Eigen::internal::unpacket_traits<float __vector(8)>::type const*) (from=0x7fff6c2004b0)
    at /usr/include/eigen3/Eigen/src/Core/GenericPacketMath.h:475
#3  Eigen::internal::evaluator<Eigen::PlainObjectBase<Eigen::Matrix<float, -1, 1, 0, -1, 1> > >::packet<32, float __vector(8)>(long) const (this=0x7fff7dff30e0, 
    index=0) at /usr/include/eigen3/Eigen/src/Core/CoreEvaluators.h:189
#4  0x00007ffff6aa4a12 in Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<float, -1, 1, 0, -1, 1> >, Eigen::internal::evaluator<Eigen::Matrix<float, -1, 1, 0, -1, 1> >, Eigen::internal::assign_op<float>, 0>::assignPacket<32, 32, float __vector(8)>(long) (this=0x7fff7dff30f0, 
    index=0) at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:598
#5  0x00007ffff6aa0782 in Eigen::internal::dense_assignment_loop<Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<float, -1, 1, 0, -1, 1> >, Eigen::internal::evaluator<Eigen::Matrix<float, -1, 1, 0, -1, 1> >, Eigen::internal::assign_op<float>, 0>, 3, 0>::run (kernel=...)
    at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:386
#6  0x00007ffff6a9cefa in Eigen::internal::call_dense_assignment_loop<Eigen::Matrix<float, -1, 1, 0, -1, 1>, Eigen::Matrix<float, -1, 1, 0, -1, 1>, Eigen::internal::assign_op<float> > (dst=..., src=..., func=...) at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:653
#7  0x00007ffff6a98867 in Eigen::internal::Assignment<Eigen::Matrix<float, -1, 1, 0, -1, 1>, Eigen::Matrix<float, -1, 1, 0, -1, 1>, Eigen::internal::assign_op<float>, Eigen::internal::Dense2Dense, float>::run (dst=..., src=..., func=...) at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:790
#8  0x00007ffff6a93870 in Eigen::internal::call_assignment_no_alias<Eigen::Matrix<float, -1, 1, 0, -1, 1>, Eigen::Matrix<float, -1, 1, 0, -1, 1>, Eigen::internal::assign_op<float> > (dst=..., src=..., func=...) at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:747
#9  0x00007ffff6a8e2e1 in Eigen::internal::call_assignment<Eigen::Matrix<float, -1, 1, 0, -1, 1>, Eigen::Matrix<float, -1, 1, 0, -1, 1>, Eigen::internal::assign_op<float> >(Eigen::Matrix<float, -1, 1, 0, -1, 1>&, Eigen::Matrix<float, -1, 1, 0, -1, 1> const&, Eigen::internal::assign_op<float> const&, Eigen::internal::enable_if<!Eigen::internal::evaluator_assume_aliasing<Eigen::Matrix<float, -1, 1, 0, -1, 1>, Eigen::internal::evaluator_traits<Eigen::Matrix<float, -1, 1, 0, -1, 1> >::Shape>::value, void*>::type) (dst=..., src=..., func=...) at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:712
#10 0x00007ffff6a87904 in Eigen::internal::call_assignment<Eigen::Matrix<float, -1, 1, 0, -1, 1>, Eigen::Matrix<float, -1, 1, 0, -1, 1> > (dst=..., src=...)
    at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:693
#11 0x00007ffff6a8167f in Eigen::PlainObjectBase<Eigen::Matrix<float, -1, 1, 0, -1, 1> >::_set<Eigen::Matrix<float, -1, 1, 0, -1, 1> > (this=0x7fff7dff3340, 
    other=...) at /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:682
#12 0x00007ffff6a7cc45 in Eigen::Matrix<float, -1, 1, 0, -1, 1>::operator= (this=0x7fff7dff3340, other=...) at /usr/include/eigen3/Eigen/src/Core/Matrix.h:208
#13 0x00007ffff6a7c041 in pcl::SampleConsensus<pcl::PointXYZ>::getModelCoefficients (this=0x7fff7dff3540, model_coefficients=...)
    at /usr/include/pcl-1.7/pcl/sample_consensus/sac.h:306
#14 0x00007ffff6a77fdb in rtabmap::util3d::transformFromXYZCorrespondences (cloud1=..., cloud2=..., inlierThreshold=0.10000000149011612, iterations=100, 
    refineIterations=5, refineSigma=3, inliersOut=0x7fff7dff4100, varianceOut=0x7fff7dff5af0)
    at /home/mathieu/workspace/rtabmap/corelib/src/util3d_registration.cpp:111
#15 0x00007ffff6b287e9 in rtabmap::util3d::estimateMotion3DTo3D (words3A=std::map with 1870 elements = {...}, words3B=std::map with 852 elements = {...}, 
    minInliers=20, inliersDistance=0.10000000149011612, iterations=100, refineIterations=5, varianceOut=0x7fff7dff5af0, matchesOut=0x7fff7dff4710, 
    inliersOut=0x7fff7dff46e0) at /home/mathieu/workspace/rtabmap/corelib/src/util3d_motion_estimation.cpp:256
#16 0x00007ffff6b6a0a7 in rtabmap::RegistrationVis::computeTransformationImpl (this=0xed6010, fromSignature=..., toSignature=..., guess=..., info=...)
    at /home/mathieu/workspace/rtabmap/corelib/src/RegistrationVis.cpp:1104
#17 0x00007ffff6b5b9a0 in rtabmap::Registration::computeTransformationMod (this=0xed6010, from=..., to=..., guess=..., infoOut=0x7fff7dff6c00)
    at /home/mathieu/workspace/rtabmap/corelib/src/Registration.cpp:194
#18 0x00007ffff6c1a1f2 in rtabmap::OdometryF2M::computeTransform (this=0xef4600, data=..., guess=..., info=0x7fff7dff9d30)
    at /home/mathieu/workspace/rtabmap/corelib/src/OdometryF2M.cpp:218
#19 0x00007ffff6bb3dbf in rtabmap::Odometry::process (this=0xef4600, data=..., guessIn=..., info=0x7fff7dff9d30)
    at /home/mathieu/workspace/rtabmap/corelib/src/Odometry.cpp:348
#20 0x00007ffff6bb280a in rtabmap::Odometry::process (this=0xef4600, data=..., info=0x7fff7dff9d30)
    at /home/mathieu/workspace/rtabmap/corelib/src/Odometry.cpp:198
#21 0x00007ffff6c17a8d in rtabmap::OdometryThread::mainLoop (this=0xefd240) at /home/mathieu/workspace/rtabmap/corelib/src/OdometryThread.cpp:97
#22 0x00007ffff5c97a29 in UThread::ThreadMain (this=0xefd240) at /home/mathieu/workspace/rtabmap/utilite/src/UThread.cpp:283
#23 0x00007ffff5c97e97 in UThreadC<void>::ThreadMainHandler (Param=0x7fffffffc730)
    at /home/mathieu/workspace/rtabmap/utilite/src/../include/rtabmap/utilite/Posix/UThreadC.h:392
#24 0x00007ffff43846fa in start_thread (arg=0x7fff7dffb700) at pthread_create.c:333
#25 0x00007ffff48a4b5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

CMake config:
```bash
$ cmake ..
-- Checking for module 'libopenni2'
--   No package 'libopenni2' found
-- Could NOT find OpenNI2 (missing:  OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
[... 33 lines cut ...]
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++11
--   With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = YES (License: Non commercial)
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI2              = NO (OpenNI2 not found)
--   With Freenect2            = NO (libfreenect2 not found)
--   With dc1394               = NO (dc1394 not found)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = NO (g2o not found)
--   With GTSAM                = NO (GTSAM not found)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = NO (cvsba not found)
--   With ZED                  = YES (With CUDA)
--   With RealSense            = YES (License: Apache-2)
--   With OCTOMAP              = YES (License: BSD)
--   With Qt5                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mathieu/workspace/rtabmap/build
```

matlabbe, 2016-10-25:
After some enabling/disabling dependencies, I finally got all working with this config (ZED 1.1.1/CUDA8):
```bash
$ cmake ..
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- Could NOT find PCL_APPS (missing:  PCL_APPS_LIBRARY) 
-- Found OpenMP
-- Found OpenCV: /opt/ros/kinetic/include/opencv-3.1.0-dev/opencv;/opt/ros/kinetic/include/opencv-3.1.0-dev
-- Found PCL: /usr/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/openni2;/usr/include/vtk-6.2;/usr/include/x86_64-linux-gnu;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include;/usr/lib/openmpi/include;/usr/lib/openmpi/include/openmpi;/usr/include/jsoncpp;/usr/include/freetype2;/usr/include/x86_64-linux-gnu/freetype2;/usr/include/hdf5/openmpi;/usr/include/python2.7;/usr/include/tcl;/usr/include/libxml2
-- Found ZLIB: /usr/include
-- Found Freenect: /opt/ros/kinetic/include/libfreenect
-- Found freenect2: /usr/local/include
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- Found g2o: /opt/ros/kinetic/include;/usr/include/suitesparse;/usr/include/suitesparse
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/local/include/gtsam/3rdparty/Eigen/
-- Found cvsba: /usr/local/include
build with AVX support... detected from cpuid
-- Found ZED sdk: /usr/local/zed/include
-- Found CUDA: /usr/local/cuda/include
-- Found RealSense: /opt/ros/kinetic/include
-- Found octomap: /opt/ros/kinetic/include
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   Version : 0.11.11
--   CMAKE_INSTALL_PREFIX = /home/mathieu/catkin_ws/devel
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++11
--   With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = YES (License: Non commercial)
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = YES (License: Apache v2 and/or GPLv2)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = YES (License: BSD)
--   With GTSAM                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = YES (License: GPLv2)
--   With ZED                  = YES (With CUDA)
--   With RealSense            = YES (License: Apache-2)
--   With OCTOMAP              = YES (License: BSD)
--   With Qt5                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mathieu/workspace/rtabmap/build
```

Not sure what was the problem, I will close it for now.

closed 2016-10-25 by matlabbe

indianajohn, 2017-02-06:
I've been fighting with this for several hours now and have finally narrowed it down to Eigen v3.3+. The problem seems to be related specifically to AVX support, added in that version. I compiled with Eigen 3.3.2 and experienced the problem, and by recompiling PCL and RTAB-Map with Eigen 3.2.10 I was able to get past the problem. You can read more about it in this Eigen bug ticket:

http://eigen.tuxfamily.org/bz/show_bug.cgi?id=1146

Another reference in the PCL forum: http://www.pcl-users.org/Segmentation-Fault-on-Keypoint-Extraction-td4042316.html

Also referencing #129 here so that guy might get a notification about this.

One more thing. The Eigen ticket was produced using GCC, but I am using Clang on MacOSX.

matlabbe, 2017-02-16:
Thx for the hint about Eigen. I had to reinstall on a new machine and I got the same error. I've found that it was because of GTSAM that the problem was fixed the last time. GTSAM installs Eigen in its installation directory. When rtabmap adds include directories and if it is built with GTSAM, the Eigen folder of GTSAM is used before the one installed on the machine (which would be a different Eigen version). 

Here is the include order:
`
Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Rtabmap.cpp.o
cd /home/mathieu/workspace/rtabmap/build/corelib/src && /usr/bin/c++   -DDISABLE_OPENNI2 -DDISABLE_PCAP -DDISABLE_PNG -DEIGEN_USE_NEW_STDVECTOR -DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET -DFLANN_STATIC -DGTSAM_IMPORT_STATIC -DPCL_OMP -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -DQT_WIDGETS_LIB -Dqh_QHpointer -Drtabmap_core_EXPORTS -DvtkFiltersFlowPaths_AUTOINIT="1(vtkFiltersParallelFlowPaths)" -DvtkIOExodus_AUTOINIT="1(vtkIOParallelExodus)" -DvtkIOGeometry_AUTOINIT="1(vtkIOMPIParallel)" -DvtkIOImage_AUTOINIT="1(vtkIOMPIImage)" -DvtkIOSQL_AUTOINIT="2(vtkIOMySQL,vtkIOPostgreSQL)" -DvtkRenderingContext2D_AUTOINIT="1(vtkRenderingContextOpenGL)" -DvtkRenderingCore_AUTOINIT="4(vtkInteractionStyle,vtkRenderingFreeType,vtkRenderingFreeTypeOpenGL,vtkRenderingOpenGL)" -DvtkRenderingFreeType_AUTOINIT="2(vtkRenderingFreeTypeFontConfig,vtkRenderingMatplotlib)" -DvtkRenderingLIC_AUTOINIT="1(vtkRenderingParallelLIC)" -DvtkRenderingVolume_AUTOINIT="1(vtkRenderingVolumeOpenGL)" -I/usr/include/vtk-6.2 -I/usr/include/jsoncpp -I/usr/include/freetype2 -I/usr/include/x86_64-linux-gnu/freetype2 -I/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent -I/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include -I/usr/lib/openmpi/include -I/usr/lib/openmpi/include/openmpi -I/usr/include/hdf5/openmpi -I/usr/include/x86_64-linux-gnu -I/usr/include/libxml2 -I/usr/include/python2.7 -I/usr/include/tcl -I/usr/local/lib/cmake/GTSAM/../../../include -I/usr/local/include/gtsam/3rdparty/Eigen -I/home/mathieu/workspace/rtabmap/utilite/include -I/home/mathieu/workspace/rtabmap/corelib/src/../include -I/home/mathieu/workspace/rtabmap/corelib/src -I/home/mathieu/workspace/rtabmap/build/corelib/src -isystem /opt/ros/kinetic/include/opencv-3.1.0-dev/opencv -isystem /opt/ros/kinetic/include/opencv-3.1.0-dev -I/usr/include/pcl-1.7 -I/usr/include/eigen3 -I/usr/include/ni -I/opt/ros/kinetic/include/libfreenect -I/usr/local/include -I/opt/ros/kinetic/include -I/usr/include/suitesparse -I/usr/local/zed/include -I/usr/local/cuda/include -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64  -fmessage-length=0  -fopenmp -std=c++11 -O3 -DNDEBUG -fPIC   -Wall -Wno-unknown-pragmas -mavx -fPIC -o CMakeFiles/rtabmap_core.dir/Rtabmap.cpp.o -c /home/mathieu/workspace/rtabmap/corelib/src/Rtabmap.cpp
`

Note the **-I/usr/local/include/gtsam/3rdparty/Eigen**.