introlab/rtabmap #132: With libfreenect2: undefined reference to OpenCL when linking rtabmap

matlabbe, 2016-10-21:
```bash
Linking CXX executable ../../../bin/rtabmap
/usr/local/lib/libfreenect2.so: undefined reference to `clReleaseMemObject@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clEnqueueWriteBuffer@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clReleaseDevice@OPENCL_1.2'
/usr/local/lib/libfreenect2.so: undefined reference to `clRetainKernel@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clGetPlatformInfo@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clCreateProgramWithSource@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clCreateCommandQueue@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clWaitForEvents@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clEnqueueReadBuffer@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clRetainEvent@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clCreateBuffer@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clBuildProgram@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clGetDeviceInfo@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clReleaseContext@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clGetPlatformIDs@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clReleaseCommandQueue@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clSetKernelArg@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clCreateContext@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clReleaseProgram@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clEnqueueUnmapMemObject@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clCreateKernel@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clRetainContext@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clEnqueueNDRangeKernel@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clEnqueueMapBuffer@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clRetainProgram@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clGetProgramBuildInfo@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clRetainMemObject@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clReleaseKernel@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clReleaseEvent@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clGetDeviceIDs@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clRetainDevice@OPENCL_1.2'
/usr/local/lib/libfreenect2.so: undefined reference to `clRetainCommandQueue@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clGetExtensionFunctionAddress@OPENCL_1.0'
```

closed 2016-10-21 by matlabbe

Algomorph, 2017-01-27:
@matlabbe , could you tell me how you fixed this?

matlabbe, 2017-01-28:
In the referred [commit](https://github.com/introlab/rtabmap/commit/4017db1d15815b1961f9cb5bae8f2f6391acf34a), cmake searches explicitly for OpenCL and add the libraries for the linker.

Algomorph, 2017-01-29:
@matlabbe , thanks!

commit 4017db1d (2016-10-21): fixed #132