introlab/rtabmap #134: Issues with GainCompensator.cpp during compile

jacksonkr, 2016-10-27:
Here's a new one:

ubuntu 14.04 amd 64
3.19.0-25-generic

```
$ make
[  6%] Built target rtabmap_utilite
[  6%] Built target res_tool
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/GainCompensator.cpp.o
/home/jackson/github/rtabmap/corelib/src/GainCompensator.cpp: In function ‘void rtabmap::feedImpl(const std::map<int, typename pcl::PointCloud<PointT>::Ptr>&, const std::map<int, boost::shared_ptr<std::vector<int> > >&, const std::multimap<int, rtabmap::Link>&, float, double, double, double, cv::Mat_<double>&, std::map<int, int>&)’:
/home/jackson/github/rtabmap/corelib/src/GainCompensator.cpp:194:63: error: no matching function for call to ‘transformPoint(Eigen::Matrix<float, 3, 1>&, Eigen::Matrix<float, 3, 1>&, Eigen::Affine3f&)’
      pcl::transformPoint(bbMinMaxTo.first, bbMinMaxTo.first, t);
                                                               ^
/home/jackson/github/rtabmap/corelib/src/GainCompensator.cpp:194:63: note: candidates are:
In file included from /usr/include/pcl-1.7/pcl/common/transforms.h:418:0,
                 from /home/jackson/github/rtabmap/corelib/src/GainCompensator.cpp:36:
/usr/include/pcl-1.7/pcl/common/impl/transforms.hpp:295:1: note: template<class PointT, class Scalar> PointT pcl::transformPoint(const PointT&, const Eigen::Transform<Scalar, 3, 2>&)
 pcl::transformPoint (const PointT &point, 
 ^
/usr/include/pcl-1.7/pcl/common/impl/transforms.hpp:295:1: note:   template argument deduction/substitution failed:
/home/jackson/github/rtabmap/corelib/src/GainCompensator.cpp:194:63: note:   ‘Eigen::Matrix<float, 3, 1>’ is not derived from ‘const Eigen::Transform<Scalar, 3, 2>’
      pcl::transformPoint(bbMinMaxTo.first, bbMinMaxTo.first, t);
                                                               ^
In file included from /home/jackson/github/rtabmap/corelib/src/GainCompensator.cpp:36:0:
/usr/include/pcl-1.7/pcl/common/transforms.h:393:3: note: template<class PointT> PointT pcl::transformPoint(const PointT&, const Affine3f&)
   transformPoint (const PointT &point, 
   ^
/usr/include/pcl-1.7/pcl/common/transforms.h:393:3: note:   template argument deduction/substitution failed:
/home/jackson/github/rtabmap/corelib/src/GainCompensator.cpp:194:63: note:   cannot convert ‘bbMinMaxTo.first’ (type ‘Eigen::Matrix<float, 3, 1>’) to type ‘const Affine3f& {aka const Eigen::Transform<float, 3, 2>&}’
      pcl::transformPoint(bbMinMaxTo.first, bbMinMaxTo.first, t);
                                                               ^
/home/jackson/github/rtabmap/corelib/src/GainCompensator.cpp:195:65: error: no matching function for call to ‘transformPoint(Eigen::Matrix<float, 3, 1>&, Eigen::Matrix<float, 3, 1>&, Eigen::Affine3f&)’
      pcl::transformPoint(bbMinMaxTo.second, bbMinMaxTo.second, t);
                                                                 ^
/home/jackson/github/rtabmap/corelib/src/GainCompensator.cpp:195:65: note: candidates are:
In file included from /usr/include/pcl-1.7/pcl/common/transforms.h:418:0,
                 from /home/jackson/github/rtabmap/corelib/src/GainCompensator.cpp:36:
/usr/include/pcl-1.7/pcl/common/impl/transforms.hpp:295:1: note: template<class PointT, class Scalar> PointT pcl::transformPoint(const PointT&, const Eigen::Transform<Scalar, 3, 2>&)
 pcl::transformPoint (const PointT &point, 
 ^
/usr/include/pcl-1.7/pcl/common/impl/transforms.hpp:295:1: note:   template argument deduction/substitution failed:
/home/jackson/github/rtabmap/corelib/src/GainCompensator.cpp:195:65: note:   ‘Eigen::Matrix<float, 3, 1>’ is not derived from ‘const Eigen::Transform<Scalar, 3, 2>’
      pcl::transformPoint(bbMinMaxTo.second, bbMinMaxTo.second, t);
                                                                 ^
In file included from /home/jackson/github/rtabmap/corelib/src/GainCompensator.cpp:36:0:
/usr/include/pcl-1.7/pcl/common/transforms.h:393:3: note: template<class PointT> PointT pcl::transformPoint(const PointT&, const Affine3f&)
   transformPoint (const PointT &point, 
   ^
/usr/include/pcl-1.7/pcl/common/transforms.h:393:3: note:   template argument deduction/substitution failed:
/home/jackson/github/rtabmap/corelib/src/GainCompensator.cpp:195:65: note:   cannot convert ‘bbMinMaxTo.second’ (type ‘Eigen::Matrix<float, 3, 1>’) to type ‘const Affine3f& {aka const Eigen::Transform<float, 3, 2>&}’
      pcl::transformPoint(bbMinMaxTo.second, bbMinMaxTo.second, t);
                                                                 ^
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/GainCompensator.cpp.o] Error 1
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [all] Error 2
```

I'm going to try a kernel update to resolve this one.

jacksonkr, 2016-10-28:
No luck..

ubuntu 14.04.5
4.4.0-45-generic

```
$ make
Scanning dependencies of target rtabmap_utilite
[  1%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UEventsManager.cpp.o
[  1%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UEventsHandler.cpp.o
[  2%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UEventsSender.cpp.o
[  2%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UFile.cpp.o
[  3%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UDirectory.cpp.o
[  3%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UConversion.cpp.o
[  4%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/ULogger.cpp.o
[  4%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UThread.cpp.o
[  5%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UTimer.cpp.o
[  5%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UProcessInfo.cpp.o
[  6%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UVariant.cpp.o
Linking CXX shared library ../../../bin/librtabmap_utilite.so
[  6%] Built target rtabmap_utilite
Scanning dependencies of target res_tool
[  6%] Building CXX object utilite/resource_generator/CMakeFiles/res_tool.dir/main.cpp.o
Linking CXX executable ../../../bin/rtabmap-res_tool
[  6%] Built target res_tool
[  6%] [Creating resources]
 Using namespace=rtabmap
 Using target directory=/home/jackson/github/rtabmap/build/corelib/src
Input file "/home/jackson/github/rtabmap/corelib/src/resources/DatabaseSchema.sql" size = 4208 bytes
Output file "/home/jackson/github/rtabmap/build/corelib/src/DatabaseSchema_sql.h" size = 8637 bytes
Scanning dependencies of target rtabmap_core
[  7%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Rtabmap.cpp.o
[  7%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/RtabmapThread.cpp.o
[  8%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Statistics.cpp.o
[  8%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Memory.cpp.o
[  9%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/DBDriver.cpp.o
[  9%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/DBDriverSqlite3.cpp.o
[ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/DBReader.cpp.o
[ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Camera.cpp.o
[ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/CameraThread.cpp.o
[ 11%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/CameraRGB.cpp.o
[ 11%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/CameraRGBD.cpp.o
[ 12%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/CameraStereo.cpp.o
[ 12%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/CameraModel.cpp.o
[ 13%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/EpipolarGeometry.cpp.o
[ 13%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/VisualWord.cpp.o
[ 14%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/VWDictionary.cpp.o
[ 14%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/BayesFilter.cpp.o
[ 15%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Parameters.cpp.o
[ 15%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Signature.cpp.o
[ 16%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Features2d.cpp.o
[ 16%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Transform.cpp.o
[ 17%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/GeodeticCoords.cpp.o
[ 17%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/util2d.cpp.o
[ 18%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/util3d.cpp.o
[ 18%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/util3d_filtering.cpp.o
[ 19%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/util3d_mapping.cpp.o
[ 19%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/util3d_transforms.cpp.o
[ 20%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/util3d_registration.cpp.o
[ 20%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/util3d_surface.cpp.o
[ 21%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/util3d_features.cpp.o
[ 21%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/util3d_correspondences.cpp.o
[ 22%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/util3d_motion_estimation.cpp.o
[ 22%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/SensorData.cpp.o
[ 23%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Graph.cpp.o
[ 23%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Compression.cpp.o
[ 24%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Link.cpp.o
[ 24%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Optimizer.cpp.o
[ 24%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/OptimizerTORO.cpp.o
[ 25%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/OptimizerG2O.cpp.o
[... 37 lines cut ...]
/home/jackson/github/rtabmap/corelib/src/GainCompensator.cpp:195:65: error: no matching function for call to ‘transformPoint(Eigen::Matrix<float, 3, 1>&, Eigen::Matrix<float, 3, 1>&, Eigen::Affine3f&)’
      pcl::transformPoint(bbMinMaxTo.second, bbMinMaxTo.second, t);
                                                                 ^
[... cut ...]
/home/jackson/github/rtabmap/corelib/src/GainCompensator.cpp:195:65: note:   cannot convert ‘bbMinMaxTo.second’ (type ‘Eigen::Matrix<float, 3, 1>’) to type ‘const Affine3f& {aka const Eigen::Transform<float, 3, 2>&}’
      pcl::transformPoint(bbMinMaxTo.second, bbMinMaxTo.second, t);
                                                                 ^
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/GainCompensator.cpp.o] Error 1
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [all] Error 2
```

jacksonkr, 2016-10-28:
From the looks of the [last commit](https://github.com/introlab/rtabmap/commit/9123889876d97a4cdd01a753e6c6921c8d3adc9c) there were some changes to the GainCompensatior integration. I suspect that's where this is coming from. I may dive in if I get time.

jacksonkr, 2016-10-28:
Temporary solution, rollback:

`git checkout -b 49cb35175f080600501b7242cc804272ebd4407d old`

Leaving this open so you're aware of the issue.

matlabbe, 2016-10-30:
https://travis-ci.org/introlab/rtabmap/builds/170942755

matlabbe, 2016-10-30:
should be fixed by https://github.com/introlab/rtabmap/commit/4bcb31e532c16816d59bb1de73bfc6b1fba138d6

closed 2016-10-30 by matlabbe